Azure/meta-iotedge #12: iotedge list - module runtime error occured

Shadpwness, 2019-02-21:
Hello,

upon running **iotedge list** I get the following:
```
A module runtime error occurred
        caused by: Could not list modules
        caused by: an error occurred trying to connect: Connection refused (os error 111)

```

systemctl status iotedge:

```
[[0;1;31m?[[0m iotedge.service - Azure IoT Edge daemon
   Loaded: loaded (/lib/systemd/system/iotedge.service; enabled; vendor preset: enabled)
   Active: [[0;1;31mfailed[[0m (Result: exit-code) since Thu 2019-02-21 13:58:39 UTC; 27min ago
     Docs: man:iotedged(8)
  Process: 751 ExecStart=/usr/bin/iotedged -c /etc/iotedge/config.yaml [[0;1;31m(code=exited, status=1/FAILURE)[[0m
 Main PID: 751 (code=exited, status=1/FAILURE)
      CPU: 183ms
```

**Additional context**
```
Linux 4.4.52  armv7l GNU/Linux

dpkg --print-architecture
armel
```

The build itself completed successfully, the only thing that's different from the readme.md is that I require the poky version of meta-virt and meta-oe instead of the listed sumo, could that be the issue? I've started building a new image with just the layers listed there with their respective versions already to see if that would work out, but it's still in progress.
Tried installing manually as well on the device, but for some reason, it's identified as armel instead of armhf, which isn't supported.
Tried restarting the service, also, the .yaml conf file is set, however, I think it should start even without that if I'm not mistaken.

If any more information is needed, let me know.

Regards,
Levente

myagley, 2019-02-22:
It looks like the `iotedged` is exiting. Can you run `journalctl -u iotedge` to see if there is any more information in the `iotedged` logs?

Shadpwness, 2019-02-25:
journalctl -u iotedge returns 

`No journal files were found.
-- No entries --`

Shadpwness, 2019-02-25:
When trying to start iotedged I get the following:
`<6>2019-02-25T09:32:28Z [INFO] - Starting Azure IoT Edge Security Daemon
<6>2019-02-25T09:32:28Z [INFO] - Version - 1.0.6.1 (3fa6cbef8b7fc3c55a49a622735eb1021b8a5963)
<6>2019-02-25T09:32:28Z [INFO] - Using default configuration
<3>2019-02-25T09:32:28Z [ERR!] - The daemon could not start up successfully: Edge device information is required.
Please update the config.yaml and provide the IoTHub connection information.`

It was a bit unclear what config.yaml should include, I've configured it with manual provisioning, set the device_connection_string (hostname=..;deviceid=..;sharedacceskey=.. format, is that right?) and hostname.

Shadpwness, 2019-03-04:
Managed to fix it by:

```
touch /var/lib/iotedge.mgmt.sock and workload.sock + giving permissions
modifying config.yaml (both connect and listen) to the .sock files from /var/lib
```

after this, running any command followed by --host unix:///var/lib/iotedge/mgmt.sock worked.
Managed to deploy as well, if anyone has issues with that, make sure that your docker has acces to internet, you can try the following for eg.:
```
pkill docker
iptables -t nat -F
ifconfig docker0 down
brctl delbr docker0
docker -d
```

closed 2019-03-04 by Shadpwness

andyrivMSFT, 2019-06-21:
In my case I needed to comment out the Edge device hostename section.  This will work for testing but not sure of the impact yet when I attempt to connect to this device with leaf device :(

ilyas-it83, 2020-06-21:
@myagley, is this applicable for the current version of IotEdged.. I don't see the resolutions suggested by @andyrivMSFT or @Shadpwness actually works

ilyas-it83, 2020-06-21:
I also suggest that such information and resolutions maintained in the docs of IoTEdge for a quickly knowing the known issues and fixes.