Azure/meta-iotedge #15: meta-iotedge not compatible with poky thud branch, but doesn't work with sumo either.

mgrischy, 2019-04-12:
**Describe the bug**
When I give the command "bitbake core-image-base", I get the error
    error: meta-iotedge is not compatible with the core layer which only supports these series: thud (layer is compatible with sumo rocko)
If I change my poky branch to sumo and then give the command "bitbake core-image-base", I get the errors:
    error: layer 'meta-iotedge' depends on layer 'rust-layer', but this layer in not enabled in your configuration
    error: layer 'meta-iotedge' depends on layer 'virtualization-layer', but this layer in not enabled in your configuration

What is the best way to move forward from this situation?

Other concerns/questions:
Pardon my ignorance; I'm a newbie to yocto, poky, iotedge, etc...

I'm confused as to where the meta-iotedge directory should be placed with 
respect to the poky directory.  If I place it outside of poky, then the
"bitbake-layers add-layer meta-iotedge" command doesn't work.  If I place it
inside poky the "bitbake-layers add-layer meta-iotedge" still doesn't work.
It's only if I cd to poky/build and modify the command to be "bitbake-layers 
add-layer ../meta-iotedge" that the command works.

I can get the meta-iotedge code to build if I manually run 
"env METARUST_REV='refs/remotes/origin/master' sh scripts/fetch.sh"
and also manually run scripts/build.sh.  This results in the meta-iotedge 
directory itself including poky and others as subdirectories.  This seems a
bit strange.  Is this intended?

**To Reproduce**
Here are the steps I followed:

    first, install yocto on my dev machine (following https://www.yoctoproject.org/docs/2.6.1/brief-yoctoprojectqs/brief-yoctoprojectqs.html)...
    browse to http://downloads.yoctoproject.org/releases/yocto/yocto-2.6.1/buildtools/
    download x86_64-buildtools-nativesadk-standalone-2.6.1.sh
    cd /mnt/sdc1
    mkdir yocto
    cd yocto
    mkdir buildtools
    cd buildtools
    mv ~/Downloads/x86_64-buildtools-nativesadk-standalone-2.6.1.sh .
    chmod a+x x86_64-buildtools-nativesadk-standalone-2.6.1.sh
    sh x86_64-buildtools-nativesadk-standalone-2.6.1.sh
      at prompt, enter install dir: (I used /opt/poky/2.6.1)
      instructions at end say:
          Each time you wish to use the SDK in a new shell session, you need to source the environment setup script e.g.
          ./opt/poky/2.6.1/environment-setup-x86_64-pokysdk-linux
    source /opt/poky/2.6.1/environment-setup-x86_64-pokysdk-linux
    sudo apt-get install diffstat

    now, bring down poky and configure it...
    cd /mnt/sdc1/yocto
    git clone git://git.yoctoproject.org/poky
    cd poky
    git fetch --tags
    git tag  (displays all of the tags available)
    git checkout tags/yocto-2.6.1 -b my-yocto-2.6.1
    source oe-init-build-env  (auto cd's to build dir)
    edit conf/local.conf and make changes:
        MACHINE ?= "beaglebone-yocto"         #mcg: uncommented this line
        PACKAGE_CLASSES ?= "package_deb"	  #mcg: replaced line with this line
        EXTRA_IMAGE_FEATURES += "tools-debug" #mcg: added this line

    now, add meta-iotedge layer and build...
    cd /mnt/sdc1/yocto/poky
    git clone https://github.com/Azure/meta-iotedge.git
    bitbake-layers add-layer meta-iotedge
        unable to find bblayers.conf
    cd build
    bitbake-layers add-layer meta-iotedge
        specified layer dir doesn't exist: /mnt/sdc1/yocto/poky/build/meta-iotedge
    bitbake-layers add-layer ../meta-iotedge   (that worked)
    bitbake core-image-base
        error: meta-iotedge is not compatible with the core layer which only supports these series: thud (layer is compatible with sumo rocko)

    now, configure poky to use the sumo branch and try the build again...
    cd .. (ie, up to yocto/poky)
    git branch -a (to see all branches)
    git checkout -b sumo remotes/origin/sumo  (my local sumo is now set to track remote branch)
    mv build build-thud
    source oe-init-build-env  (makes new build dir and auto cd's to it)
    edit conf/local.conf and make changes:
        MACHINE ?= "beaglebone-yocto"         #mcg: uncommented this line
        PACKAGE_CLASSES ?= "package_deb"	  #mcg: replaced line with this line
        EXTRA_IMAGE_FEATURES += "tools-debug" #mcg: added this line
    bitbake-layers add-layer ../meta-iotedge
        error: layer 'meta-iotedge' depends on layer 'rust-layer', but this layer in not enabled in your configuration
        error: layer 'meta-iotedge' depends on layer 'virtualization-layer', but this layer in not enabled in your configuration
    so edited conf/bblayers.conf manually and added to the end of BBLAYERS
          /mnt/sdc1/yocto/poky/meta-iotedge \
    bitbake core-image-base
        error: layer 'meta-iotedge' depends on layer 'rust-layer', but this layer in not enabled in your configuration
        error: layer 'meta-iotedge' depends on layer 'virtualization-layer', but this layer in not enabled in your configuration

**Expected behavior**
I expect it to build without errors.

**Logs**
If applicable, add logs to help explain your problem.

**Build System (please complete the following information):**
 - OS: Debian 8 (jessie)
 - Version: 3.16.51-3

**Target System (please complete the following information):**
 - OS: linux kernel based on kernel.org
 - Version 4.1.51
 - architecture: ARMv8 using 64-bits for the kernel and 32-bits for user space programs

**Additional context**
Add any other context about the problem here.

myagley, 2019-04-26:
Thank you for your patience. I am just seeing these issues now.

Normally all of the layers should be placed at the same level. The build scripts git clone the repos into the poky folder. This isn't normal but was required to get some CI build to work. You probably don't want to follow this pattern.

Typically you will create a directory and clone each layer into this directory. All of the layers should be on the same level, including poky. The iotedge layer requires `meta-rust`, `meta-virtualization`, and `openembedded-core` to build. You will need to clone these as well. The git uris and included in `meta-iotedge`'s README as well as the branch that needs to be checked out.

mgrischy, 2019-04-29:
Thank you for the additional instructions.  I tried following your suggestions and built poky with iotedge as an added layer but didn't see any  iotedge executables after the build.  Previously, I had found them in build/tmp/deploy/deb/cortexa8hf-neon.  Also, I think the README.md file calls out git://github.com/openembedded/openembedded-core.git but maybe you really meant git://git.openembedded.org/meta-openembedded.git?  And I had to manually add the iotedge layers to bblayers.conf because I ran into problems with bitbake-layers (see below).  Any advice on how I can move forward is appreciated.

Here is what I did to build it:
    cd /mnt/sdc1
    mkdir 3yocto
    cd 3yocto
    git clone https://github.com/Azure/meta-iotedge.git     (branch in use is master)
    git clone https://github.com/meta-rust/meta-rust.git    (branch in use is master)
    git clone git://git.yoctoproject.org/meta-virtualization
    cd meta-virtualization
    git branch -a  (to see all branches)
    git checkout -b sumo remotes/origin/sumo  (my local sumo branch is now set to track remote branch)
    cd ..
    git clone git://github.com/openembedded/openembedded-core.git
    cd openembedded-core
    git branch -a  (to see all branches)
    git checkout -b sumo remotes/origin/sumo  (my local sumo branch is now set to track remote branch)
    cd ..
    git clone git://git.yoctoproject.org/poky
    cd poky
    git branch -a  (to see all branches)
    git checkout -b sumo remotes/origin/sumo  (my local sumo is now set to track remote branch)
    cd ../poky
    source oe-init-build-env  (auto cd's to build dir)
    source /opt/poky/2.6.1/environment-setup-x86_64-pokysdk-linux  (where build tools were installed previously)
    edit conf/local.conf and make changes:
        MACHINE ?= "beaglebone-yocto"              #mcg: uncommented this line
        PACKAGE_CLASSES ?= "package_deb"		   #mcg: replaced other line with this line
        EXTRA_IMAGE_FEATURES += "tools-debug"          #mcg: added this line
    cd /mnt/sdc1/3yocto/poky/build
    bitbake-layers add-layer ../../meta-iotedge
        ERROR: Layer 'meta-iotedge' depends on layer 'rust-layer','virtualization-layer', but this layer is not enabled in your configuration
    bitbake-layers add-layer ../../openembedded-core
        Specified layer directory /mnt/sdc1/3yocto/openembedded-core doesn't contain a conf/layer.conf file
    bitbake-layers add-layer ../../meta-rust
        ERROR: Layer 'rust-layer' depends on layer 'openembedded-layer', but this layer is not enabled in your configuration
    so I just edited poky/build/conf/bblayers.conf manually and this is what it looks like now (based on meta-iotedge/conf/bblayers.conf.sample)
        BBLAYERS ?= " \
          /mnt/sdc1/3yocto/meta-iotedge \
          /mnt/sdc1/3yocto/poky/meta \
          /mnt/sdc1/3yocto/poky/meta-poky \
          /mnt/sdc1/3yocto/meta-rust \
          /mnt/sdc1/3yocto/poky/meta-yocto-bsp \
          /mnt/sdc1/3yocto/poky/meta-openembedded/meta-filesystems \
          /mnt/sdc1/3yocto/poky/meta-openembedded/meta-oe \
          /mnt/sdc1/3yocto/poky/meta-openembedded/meta-networking \
          /mnt/sdc1/3yocto/poky/meta-openembedded/meta-python \
          /mnt/sdc1/3yocto/meta-virtualization \
          "

        BBLAYERS_NON_REMOVABLE ?= " \
          /mnt/sdc1/3yocto/poky/meta \
          /mnt/sdc1/3yocto/poky/meta-yocto \
          "
    cd /mnt/sdc1/3yocto/poky/build
    bitbake core-image-base
        ERROR: Layer directory '/mnt/sdc1/3yocto/poky/meta-openembedded/meta-filesystems' does not exist!
    cd ../..
    git clone git://git.openembedded.org/meta-openembedded.git  (this contains meta-networking)
    cd meta-openembedded
    git checkout -b sumo remotes/origin/sumo
    cd ../poky/build/conf
    edited poky/build/conf/bblayers.conf manually to fix references to meta-openembedded
        BBLAYERS ?= " \
          /mnt/sdc1/3yocto/meta-iotedge \
          /mnt/sdc1/3yocto/poky/meta \
          /mnt/sdc1/3yocto/poky/meta-poky \
          /mnt/sdc1/3yocto/meta-rust \
          /mnt/sdc1/3yocto/poky/meta-yocto-bsp \
          /mnt/sdc1/3yocto/meta-openembedded/meta-filesystems \
          /mnt/sdc1/3yocto/meta-openembedded/meta-oe \
          /mnt/sdc1/3yocto/meta-openembedded/meta-networking \
          /mnt/sdc1/3yocto/meta-openembedded/meta-python \
          /mnt/sdc1/3yocto/meta-virtualization \
          "
    cd /mnt/sdc1/3yocto/poky/build
    During previous build efforts, I got a warning about meta-virtualization not being enabled in my DISTRO_FEATURES, so to avoid that I edited conf/local.conf and added
        DISTRO_FEATURES_append += " virtualization"
    as is shown in meta-iotedge/conf/local.conf.sample
    bitbake core-image-base
        The build completed after a few hours
    I then looked for the iotedge executables, but they are not in the usual place (build/tmp/deploy/deb/cortexa8hf-neon).
    So, I assume the build didn't build the iotedge stuff.

emilm, 2019-05-15:
I have just tried compiling with thud, and the build goes through, but no artifacts or packages or anything seems to be produced. Where should I check? Thanks! :)

myagley, 2019-05-15:
If you are running `bitbake core-image-base` you should be able to find the image in `build/tmp/deploy/images/{machine}/`. I believe by default bitbake will clean up all of the intermediate artifacts.

emilm, 2019-05-15:
Thanks for replying. I am specifically looking for iot edge files. The image is indeed produced, but where do I look to confirm that all the IoT Edge files are there?

wlenzeder, 2019-08-29:
Don't know if this is still relevant ... but you could verify that you are adding iotedge to your image in the image configuration file: `<layer-name>\recipes-core\images\<image-name>.bb`

```
IMAGE_INSTALL += "docker iotedge-daemon iotedge-cli"
```

Then it should be included when doing:

```
$ bitbake <image-name>
```

emilm, 2019-12-05:
> Don't know if this is still relevant ... but you could verify that you are adding iotedge to your image in the image configuration file: `<layer-name>\recipes-core\images\<image-name>.bb`
>
> ``` [...]

Thank, yes I think it was something like that. I was inexperienced in yocto, and assumed it magically got added when I added the layer. :)

geoffrey-vl, 2020-03-17:
so, solved?

micahl, 2021-07-09:
Stale issue.  Closing.

closed 2021-07-09 by micahl