Azure/meta-iotedge #16: LAYERSERIES_COMPAT is incompatible with required rustc compiler version

mgrischy, 2019-04-19:
**Describe the bug**
I am building iotedge for use with an embedded target that is an existing product in the market.  This target uses specific versions of various shared libraries such as openssl and the C libraries, and these specific versions cannot be changed.  For example the C libraries on the target are all version 2.21.  However, the iotedge yocto layer builds against C libraries that are version 2.27.  The iotedge executables generated during the build process won't run on the target platform because of the library version mismatch.  Is there a way I can control the shared library versions that are used during the build?

**To Reproduce**

**Expected behavior**

**Logs**

**Build System (please complete the following information):**
 - OS: debian
 - Version 8 (jessie)

**Target System (please complete the following information):**
 - OS: custom linux kernel based on kernel.org
 - Version 4.1.51

**Additional context**
Add any other context about the problem here.

myagley, 2019-04-26:
IoT Edge builds against what ever version of the libraries are included in the Yocto distribution.

Are you building `libc` as part of the Yocto build?

Yocto assumes that the entire system is being built via Yocto. This includes the linux kernel, libc, and the rest of the root filesystem. This is how we can guarantee that all of the components work together.

mgrischy, 2019-04-26:
Thanks for your response. 
Sorry for being ignorant, but how do I tell whether I'm building libc as part of the Yocto build?  There seem to be files under build/tmp/ that indicate that libc is being built:  For example, build/tmp/beaglebone-yocto/<big-number>.log has various entries such as: "NOTE: recipe glibc-2.27-r0: task do_compile: Succeeded".

Also, how do I control which version of libc is built/used in the yocto build?

micahl, 2023-01-20:
Re: identifying which version of libc you're building
You can try `git grep PV -- glibc-common.inc`

closed 2023-01-20 by micahl