Azure/meta-iotedge #17: Rust module seems incompatible with systemd (filesystem)

PrimaElectro, 2019-05-06:
i'm compiling a yocto project  (sumo based) with systemd option. Adding the Rust layer, Iget the following error:

> ERROR: cargo-native-0.25.0-r0 do_compile: Function failed: do_compile
> 
>   ERROR: Task (virtual:native:/home/user/pe-yocto-4.1/build/../git/meta-rust/recipes-devtools/cargo/cargo_0.25.0.bb:do_compile)

probably this module is not compatible with the filesystem (systemd), has anyone had the same problem ?

myagley, 2019-05-07:
Thanks for the issue report. Systemd is compatible with these layers. The sumo images we run all use `systemd` as init.

It looks like `cargo` is failing to build. Is it possible to get the logs for this build step. It also looks like this version of cargo is fairly old. I would try pulling down the latest commits for `meta-rust`. Note, that we require `meta-rust` to use the `master` branch, not the `sumo` branch.

geoffrey-vl, 2020-03-17:
It's been nearly one year, is this still relevant?

micahl, 2021-07-09:
Closing.  Stale issue.

closed 2021-07-09 by micahl