Azure/meta-iotedge #22: Update to 1.0.8

mschwan-phytec, 2019-08-02:
Hi!

Are there any plans on updating this layer to version 1.0.8? I tried doing this myself and wanted to open a pull request but I encountered several issues:

* The download for the .tar.gz of version 1.0.8 does not exist: https://github.com/Azure/azure-iotedge/releases/download/1.0.8/iotedge-1.0.8.tar.gz Which got me wondering why there is a separate repository containing releases for `iotedge`? Is it because 1.0.8 is not GA at the moment?
* Using the tagged commit from the upstream branch `release/1.0.8` does not work for some reason. Yocto claims that the commit in this branch does not exist, though it is available. Even specifying the master branch with the commit ID (https://github.com/Azure/iotedge/commit/f5186c42ebb721893c5a497c789236803de42fb7 or https://github.com/Azure/iotedge/commit/208b2204fd30e856d00b280112422130c104b9f0) does not work.

Can you provide any information on this?

wlenzeder, 2019-08-19:
Hi, when will this layer be updated to 1.0.8 - I've seen that 1.0.9-rc1 has been released, will 1.0.8 be skipped for meta-iotedge?
There are also Yocto build issues with version 1.0.7 and the current rust/cargo versions in the master branch of meta-rust.

wlenzeder, 2019-08-28:
Hi, thanks for pushing the update!

I tried building the new version for sumo, but unfortunately I'm having issues during compilation:

```
[ 69%] Built target hsm_client_tpm_ut_exe
Makefile:162: recipe for target 'all' failed

--- stderr
CMake Warning:
  Manually-specified variables were not used by the project:

    use_emulator

/nStack/devLicense/share/yocto/build/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/lib/libssl.so: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
make[2]: *** [libiothsm.so.1.0.8] Error 1
make[1]: *** [CMakeFiles/iothsm.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
thread 'main' panicked at '
command did not execute successfully, got: exit code: 2
```

The full compile log file is attached: [log.do_compile.zip](https://github.com/Azure/meta-iotedge/files/3550537/log.do_compile.zip)

Could you please have a look?

Thanks,
Wolfgang

mschwan-phytec, 2019-08-28:
Can confirm that sumo does not build, but thud does.

myagley, 2019-09-03:
Are you on the `sumo` branch`?

wlenzeder, 2019-09-06:
I'm on the most recent `sumo` branches:

```
Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
...
TUNE_FEATURES        = "arm armv7ve vfp thumb neon callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta
meta-poky            = "sumo:5ddf7fff992b065ee512878d2fe65f3e35d818cf"
meta-oe              = "sumo:8760facba1bceb299b3613b8955621ddaa3d4c3f"
...
meta-virtualization  = "sumo:ed2038c935777d1336c17989d454f4e9c95fea7f"
meta-rust            = "master:11aed43748d7dd194dc6c06a35a16104400d1090"
meta-iotedge         = "sumo:734dc718da4fe3ac39b6049f543223698897433f"
```

ksaye, 2019-10-25:
I get the same error:

ksaye@el1000:~/rpi$ bitbake core-image-base

Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5.3"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "sumo:5ddf7fff992b065ee512878d2fe65f3e35d818cf"
meta-oe
meta-networking
meta-python
meta-filesystems     = "sumo:8760facba1bceb299b3613b8955621ddaa3d4c3f"
meta-raspberrypi     = "sumo:2d40b000021bc8a9ef7f329ed0ad410f8d227b97"
meta-iotedge         = "sumo:734dc718da4fe3ac39b6049f543223698897433f"
meta-rust            = "HEAD:9487b089ea4779c2b494b17b9254219226efa539"
meta-virtualization  = "sumo:ed2038c935777d1336c17989d454f4e9c95fea7f"

I get the error shown below:

| [ 64%] Linking C shared library libiothsm.so
| CMakeFiles/iothsm.dir/build.make:543: recipe for target 'libiothsm.so.1.0.8' failed
| CMakeFiles/Makefile2:68: recipe for target 'CMakeFiles/iothsm.dir/all' failed
| [ 65%] Linking C executable edge_openssl_enc_ut_exe
| [ 65%] Built target edge_openssl_enc_ut_exe
| [ 66%] Linking C executable edge_hsm_key_intf_sas_ut_exe
| [ 66%] Built target edge_hsm_key_intf_sas_ut_exe
| [ 66%] Linking C executable edge_hsm_tpm_ut_exe
| [ 66%] Built target edge_hsm_tpm_ut_exe
| [ 66%] Linking C executable edge_hsm_crypto_ut_exe
| [ 66%] Built target edge_hsm_crypto_ut_exe
| [ 67%] Linking C executable edge_hsm_x509_ut_exe
| [ 67%] Built target edge_hsm_x509_ut_exe
| [ 68%] Linking C executable edge_openssl_pki_ut_exe
| [ 68%] Built target edge_openssl_pki_ut_exe
| [ 69%] Linking C executable hsm_client_tpm_ut_exe
| [ 69%] Built target hsm_client_tpm_ut_exe
| Makefile:162: recipe for target 'all' failed
|
| --- stderr
| fatal: not a git repository (or any of the parent directories): .git
| CMake Warning:
|   Manually-specified variables were not used by the project:
|
|     use_emulator
|
|
| /home/ksaye/rpi/rpi-build/tmp/work/aarch64-poky-linux/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/lib/libssl.so: error adding symbols: File in wrong format
| collect2: error: ld returned 1 exit status
| make[2]: *** [libiothsm.so.1.0.8] Error 1
| make[1]: *** [CMakeFiles/iothsm.dir/all] Error 2
| make[1]: *** Waiting for unfinished jobs....
| make: *** [all] Error 2
| thread 'main' panicked at '
| command did not execute successfully, got: exit code: 2

Looking in the 'rpi-build/tmp/work/aarch64-poky-linux/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/lib' I do not see libssl.so.1.0.8, which I guess is the problem.

ksaye@el1000:~/rpi/rpi-build/tmp/work/aarch64-poky-linux/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/lib$ ls -all libssl*.*
-rw-r--r-- 440 ksaye ksaye 730666 Feb  3  2019 libssl.a
lrwxrwxrwx   1 ksaye ksaye     15 Oct 25 18:45 libssl.so -> libssl.so.1.0.2
-rwxr-xr-x 440 ksaye ksaye 437408 Feb  3  2019 libssl.so.1.0.2
ksaye@el1000:~/rpi/rpi-build/tmp/work/aarch64-poky-linux/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/lib$

emilm, 2019-12-19:
Did anyone solve this? I tried to add 1.0.9 RC3 in my own meta-layer for now but I run into the same error. Has to be an issue with libssl10 vs libssl or something

emilm, 2019-12-20:
iotedge-cli does not have the necessary OpenSSL  dirs specified to staging dir 

https://github.com/Azure/meta-iotedge/blob/sumo/recipes-core/iotedge-cli/iotedge-cli.inc

Needs:

export OPENSSL_DIR = "${STAGING_EXECPREFIXDIR}"
export OpenSSLDir = "${STAGING_EXECPREFIXDIR}"
export OPENSSL_ROOT_DIR = "${STAGING_EXECPREFIXDIR}"

wlenzeder, 2020-04-09:
I can confirm that build for Yocto Sumo is fixed by #35

vaishnavils97, 2020-11-11:
I am facing same issue while building a yocto image for raspberry pi Model B. I am referring this link for the build "https://kevinsaye.wordpress.com/2019/06/21/how-to-build-a-yocto-image-for-the-raspberry-pi-running-azure-iot-edge/"  .
Have cloned meta-iotedge sumo branch and got the below error

/home/YoctoLinux/rpi/rpi-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/lib/libssl.so: file not recognized: File format not recognized
| collect2: error: ld returned 1 exit status
| make[2]: *** [libiothsm.so.1.0.8] Error 1
| make[1]: *** [CMakeFiles/iothsm.dir/all] Error 2
| make[1]: *** Waiting for unfinished jobs....
| make: *** [all] Error 2
| thread 'main' panicked at '
| command did not execute successfully, got: exit code: 2
|
| build script failed, must exit now', /home/YoctoLinux/rpi/rpi-build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/cmake-0.1.30/src/lib.rs:643:5

Please give inputs to solve the issue

vaishnavils97, 2020-11-12:
I have updated the file < https://github.com/Azure/meta-iotedge/blob/sumo/recipes-core/iotedge-cli/iotedge-cli.inc>

export OPENSSL_DIR = "${STAGING_EXECPREFIXDIR}"
export OpenSSLDir = "${STAGING_EXECPREFIXDIR}"
export OPENSSL_ROOT_DIR = "${STAGING_EXECPREFIXDIR}"

and perhaps export LIBIOTHSM_NOBUILD="On"

But not resolving issue. Please let me know if any solution available

XIIIVI, 2020-11-27:
Hi all,

I have also the same issue when compiling the tool tpm_device_provision of the azure-iot-sdk-c in Yocto.
I tried to add the following lines to the recipe without any success
DEPENDS += "openssl virtual/libiothsm"
RDEPENDS_${PN} += "virtual/libiothsm"

export OPENSSL_DIR = "${STAGING_EXECPREFIXDIR}"
export LIBIOTHSM_NOBUILD="On"
INHIBIT_PACKAGE_STRIP = "1"

The last line comes from this [post](https://forums.xilinx.com/t5/Embedded-Linux/File-Format-not-Recognized-when-packaging-user-app/td-p/919901).

I'm using the branch "dunfell".
I've spent over a week on this issue and I'm desesperately interesting in any clue to solve it ;)

Have a nice WE

XIIIVI, 2020-11-28:
After an ultimate try, my recipe managed to compile.
The settings are the following
```
DEPENDS += "openssl"

export OPENSSL_DIR = "${STAGING_EXECPREFIXDIR}"
export OpenSSLDir = "${STAGING_EXECPREFIXDIR}"
export OPENSSL_ROOT_DIR = "${STAGING_EXECPREFIXDIR}"
export LIBIOTHSM_NOBUILD="On"
```

Have a nice WE

micahl, 2021-07-09:
Cool, thanks for the update, @XIIIVI!  Closing

closed 2021-07-09 by micahl