Azure/meta-iotedge #26: Problem with docker and iptables

hb-peer, 2019-09-04:
**Describe the bug**
When the docker daemon should be started, it doesn't and fails. The output gives me some errors, it stops with the error "failed to start daemon: Error initializing network controller: Error creating default "bridge" network: Failed to program NAT chain: Failed to inject DOCKER in PREROUTING chain: iptables failed: iptables --wait -t nat -A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER: iptables: No chain/target/match by that name."

**To Reproduce**
Start dockerd from the commandline

**Expected behavior**
I expected it to run and configure the iptables as needed.

**Logs**
If applicable, add logs to help explain your problem.

**Build System (please complete the following information):**
 - OS: Ubuntu 18.04 64 bit
 - Version 18.04

**Target System (please complete the following information):**
 - OS: Yocto warrior generic_X64
 - Version Linux version 5.0.3-yocto-standard (oe-user@oe-host) (gcc version 8.3.0 (GCC)) #1 SMP PREEMPT Wed Aug 28 07:45:14 UTC 2019

**Additional context**
When I start docker with --iptables=false docker will run, but I have connections issues from a container to the internet.
I tampered a bit with iptables and I could make a container get a connection to the internet over the built in bridge.
But iot edge uses the azure-iot-edge bridge apparently and I could not get a container in the network connect to the internet. At least, thats wheat iotedge check says:

Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port
√ host can connect to and perform TLS handshake with IoT Hub HTTPS port
√ host can connect to and perform TLS handshake with IoT Hub MQTT port
√ container on the default network can connect to IoT Hub AMQP port
√ container on the default network can connect to IoT Hub HTTPS port
√ container on the default network can connect to IoT Hub MQTT port
× container on the IoT Edge module network can connect to IoT Hub AMQP port
    Container on the azure-iot-edge network could not connect to HbTestHubS1.azure-devices.net:5671
× container on the IoT Edge module network can connect to IoT Hub HTTPS port
    Container on the azure-iot-edge network could not connect to HbTestHubS1.azure-devices.net:443
× container on the IoT Edge module network can connect to IoT Hub MQTT port
    Container on the azure-iot-edge network could not connect to HbTestHubS1.azure-devices.net:8883
× Edge Hub can bind to ports on host
    Could not check current state of Edge Hub container

One or more checks raised errors. Re-run with --verbose for more details.

I think that is the reason why iotedge does not work on the target.

On docker there is only the edgeAgent running, the edgeHub does not run. Output of /var/log/iotedge/iotedge.log:
<6>2019-09-04T09:07:17Z [INFO] - Checking edge runtime status
<6>2019-09-04T09:07:17Z [INFO] - Edge runtime is running.
<6>2019-09-04T09:07:17Z [INFO] - [mgmt] - - - [2019-09-04 09:07:17.231004070 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 530 "-" "-" pid(7586)
<6>2019-09-04T09:07:22Z [INFO] - [mgmt] - - - [2019-09-04 09:07:22.239242885 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 530 "-" "-" pid(7586)

Is there a way to easily change iptables entries so I can run dockerd --iptables=false or is there a way to get iptables to work with docker correctly?

Or should I use a different Yocto? I used the warrior-brach from the iotedge-layer btw.

Thanks for your help!

Best regards,

P. Ketterle

hb-peer, 2019-09-04:
Oh btw: I used warrior because it was the current build and there was a branch for it for the iot-edge-layer. If you know of a Yocto-Version which works out of the box with the iot-edge-layer, I can try to switch there. The only mein requirement I have is that I also need Java8 on the host.

Thanks!

myagley, 2019-09-06:
I will follow up on the Docker error. The most tested Yocto version at this point is Sumo, as this is what more of the devices we are running are based on.

hb-peer, 2019-09-19:
Hello!

thanks for your reply. I tried sumo but my gcc is too new. I tried thud with the same results as before with warrior. So, probably not a branch-related issue.

I found out that I cannot install meta-virtualization with docker at all, so I followed up on this. Turns out, I was missing the kernel-modules. Added those, now docker works.

So, one problem solved, next one comes up. Maybe you can help me with this.
The startup script doesn't work right away, so I start iotedge by commandline. I can change it later to work right away, I believe. I just start `iotedged -c <path-to-cfg-file>`

I had to change config.yaml and set the listen sockets to the same as the command sockets.

The result is a bit like what I had before, but I can get one step further. The command iotedge check tells me now that a container on the iot edge module network can connect to an azure iot hub:
`
Connectivity checks
-------------------
√ host can connect to and perform TLS handshake with IoT Hub AMQP port - OK
√ host can connect to and perform TLS handshake with IoT Hub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with IoT Hub MQTT port - OK
√ container on the default network can connect to IoT Hub AMQP port - OK
√ container on the default network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the default network can connect to IoT Hub MQTT port - OK
√ container on the IoT Edge module network can connect to IoT Hub AMQP port - OK
√ container on the IoT Edge module network can connect to IoT Hub HTTPS / WebSockets port - OK
√ container on the IoT Edge module network can connect to IoT Hub MQTT port - OK
× Edge Hub can bind to ports on host - Error
    Could not check current state of Edge Hub container
        caused by: docker returned exit code: 1, stderr = Error: No such object: edgeHub

17 check(s) succeeded.
5 check(s) raised warnings.
1 check(s) raised errors.
`

Now, a test with the edge hub fails. The reason is probably that no edgeHub is runing in docker (docker ps):

`CONTAINER ID        IMAGE                                      COMMAND                   CREATED             STATUS              PORTS               NAMES
d6c5cf73cb34        mcr.microsoft.com/azureiotedge-agent:1.0   "/bin/sh -c 'echo \"$…"   About an hour ago   Up About an hour                        edgeAgent
`

iotedge logs edgeModule shows:

`2019-09-19 12:06:50  Starting Edge Agent
2019-09-19 12:06:51.046 +00:00 Edge Agent Main()
<6> 2019-09-19 12:06:51.246 +00:00 [INF] - Initializing Edge Agent.
<6> 2019-09-19 12:06:51.476 +00:00 [INF] - Version - 1.0.8.1.24554792 (8a9cfedfc6c7455c080dcee6ee21978069024d2f)
<6> 2019-09-19 12:06:51.476 +00:00 [INF] - 
        █████╗ ███████╗██╗   ██╗██████╗ ███████╗
       ██╔══██╗╚══███╔╝██║   ██║██╔══██╗██╔════╝
       ███████║  ███╔╝ ██║   ██║██████╔╝█████╗
       ██╔══██║ ███╔╝  ██║   ██║██╔══██╗██╔══╝
       ██║  ██║███████╗╚██████╔╝██║  ██║███████╗
       ╚═╝  ╚═╝╚══════╝ ╚═════╝ ╚═╝  ╚═╝╚══════╝

 ██╗ ██████╗ ████████╗    ███████╗██████╗  ██████╗ ███████╗
 ██║██╔═══██╗╚══██╔══╝    ██╔════╝██╔══██╗██╔════╝ ██╔════╝
 ██║██║   ██║   ██║       █████╗  ██║  ██║██║  ███╗█████╗
 ██║██║   ██║   ██║       ██╔══╝  ██║  ██║██║   ██║██╔══╝
 ██║╚██████╔╝   ██║       ███████╗██████╔╝╚██████╔╝███████╗
 ╚═╝ ╚═════╝    ╚═╝       ╚══════╝╚═════╝  ╚═════╝ ╚══════╝

<6> 2019-09-19 12:06:51.559 +00:00 [INF] - Experimental features configuration: {"Enabled":false,"DisableCloudSubscriptions":false,"EnableUploadLogs":false}
<6> 2019-09-19 12:06:52.030 +00:00 [INF] - Started operation refresh twin config
<6> 2019-09-19 12:06:52.080 +00:00 [INF] - Edge agent attempting to connect to IoT Hub via Amqp_Tcp_Only...
<6> 2019-09-19 12:06:52.899 +00:00 [INF] - Created persistent store at /tmp/edgeAgent
<6> 2019-09-19 12:06:54.045 +00:00 [INF] - Edge agent connected to IoT Hub via Amqp_Tcp_Only.
<6> 2019-09-19 12:06:54.313 +00:00 [INF] - Initialized new module client with subscriptions enabled
<6> 2019-09-19 12:06:54.487 +00:00 [INF] - Obtained Edge agent twin from IoTHub with desired properties version 1 and reported properties version 9.
<6> 2019-09-19 12:06:54.492 +00:00 [INF] - Deployment config in edge agent's desired properties is empty.
<3> 2019-09-19 12:06:54.495 +00:00 [ERR] - Error refreshing edge agent configuration from twin.
Microsoft.Azure.Devices.Edge.Agent.Core.ConfigSources.ConfigEmptyException: This device has an empty configuration for the edge agent. Please set a deployment manifest.
[... 12 lines cut ...]
For completeness the output of iotedged seems somewhat normal, regularily checks for the edgeAgant semm to be sucessfull, this repeats:

`<6>2019-09-19T12:59:21Z [INFO] - Checking edge runtime status
<6>2019-09-19T12:59:21Z [INFO] - Edge runtime is running.
<6>2019-09-19T12:59:25Z [INFO] - [mgmt] - - - [2019-09-19 12:59:25.029481783 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 449 "-" "-" auth_id(-)
<6>2019-09-19T12:59:30Z [INFO] - [mgmt] - - - [2019-09-19 12:59:30.032768757 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 449 "-" "-" auth_id(-)
<6>2019-09-19T12:59:35Z [INFO] - [mgmt] - - - [2019-09-19 12:59:35.036530655 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 449 "-" "-" auth_id(-)
<6>2019-09-19T12:59:40Z [INFO] - [mgmt] - - - [2019-09-19 12:59:40.040375606 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 449 "-" "-" auth_id(-)
[... cut ...]
<6>2019-09-19T13:00:10Z [INFO] - [mgmt] - - - [2019-09-19 13:00:10.069445177 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 449 "-" "-" auth_id(-)
<6>2019-09-19T13:00:15Z [INFO] - [mgmt] - - - [2019-09-19 13:00:15.074546081 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 449 "-" "-" auth_id(-)
<6>2019-09-19T13:00:20Z [INFO] - [mgmt] - - - [2019-09-19 13:00:20.080424388 UTC] "GET /modules?api-version=2019-01-30 HTTP/1.1" 200 OK 449 "-" "-" auth_id(-)
`

Do you have any idea what is still missing?

Thanks for your help!

Best regards,

Peer

hb-peer, 2019-09-20:
Hello!

A follow up on this so anyone else who may have the same problems can read this:

The empty configuration was apparently really an empty configuration. In the Azure portal I had to click "set modules" and click next to create a configuration. I never had to do that before, don't know why.
Then edgeHub was pulled and started but immediatly stopped. Logs showed that permissions for /varRun/iotedge/managment.sock was not sufficient. I changed the rights and restarted edgeHub. Then it kept on working.

I could add my own iot edge module and it is running now.

I have to fiddle a bit with the startup script, but I should get it working. There are a lot of loose ends to tie up when making Yocto iotedge-capable :)

Thanks for your help!

Best regards,

Peer

ksaye, 2020-01-07:
> kernel-modules

> Hello!
> thanks for your reply. I tried sumo but my gcc is too new. I tried thud with the same results as before with warrior. So, probably not a branch-related issue.
> I found out that I cannot [...]

Peer, how did you include kernel-modules and exclude meta-virtualization?  when I exclude meta-virtualization, the build throws an error saying that layer is required by iotedge

hb-peer, 2020-01-08:
Hello!

Maybe I was wording it the wrong way. I did not exclude meta-virtualization, because I need it for Docker, which is needed for iot edge.
I just added the kernel modules. After that, docker worked.

Best regards,

Peer

at Microsoft.Azure.Devices.Edge.Agent.IoTHub.EdgeAgentConnection.b__29_0(Twin twin) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.IoTHub/EdgeAgentConnection.cs:line 176
at Microsoft.Azure.Devices.Edge.Agent.IoTHub.EdgeAgentConnection.b__29_0(Twin twin) in /home/vsts/work/1/s/edge-agent/src/Microsoft.Azure.Devices.Edge.Agent.IoTHub/EdgeAgentConnection.cs:line 176

ksaye, 2020-01-08:
@hb-peer, can you share the change you made in your build files?

I added:

IMAGE_INSTALL_append += " kernel-modules " to local.conf (entire script here: https://github.com/ksaye/IoTDemonstrations/blob/master/batch/run.sh) which compiled without error, but did not resolve the issue.  

I am still getting the following if I run dockerd -D:

"pickfirstBalancer: HandleSubConnStateChange: 0xc0007c6b80, CONNECTING  module=grpc
failed to start daemon: Error initializing network controller: Error creating default "bridge" network: Failed to program NAT chain: Failed to inject DOCKER in PREROUTING chain: iptables failed: iptables --wait -t nat -A PREROUTING -m addrtype --dst-type LOCAL -j DOCKER: iptables: No chain/target/match by that name."

Running this script: https://github.com/moby/moby/blob/master/contrib/check-config.sh, I see the following output:

root@intel-corei7-64:~# ./check-config.sh
info: reading kernel config from /proc/config.gz ...

Generally Necessary:
- cgroup hierarchy: properly mounted [/sys/fs/cgroup]
- CONFIG_NAMESPACES: enabled
- CONFIG_NET_NS: enabled
- CONFIG_PID_NS: enabled
- CONFIG_IPC_NS: enabled
- CONFIG_UTS_NS: enabled
- CONFIG_CGROUPS: enabled
- CONFIG_CGROUP_CPUACCT: enabled
- CONFIG_CGROUP_DEVICE: enabled
- CONFIG_CGROUP_FREEZER: enabled
- CONFIG_CGROUP_SCHED: enabled
- CONFIG_CPUSETS: enabled
- CONFIG_MEMCG: enabled
- CONFIG_KEYS: enabled
- CONFIG_VETH: missing
- CONFIG_BRIDGE: enabled (as module)
- CONFIG_BRIDGE_NETFILTER: enabled (as module)
- CONFIG_NF_NAT_IPV4: enabled (as module)
- CONFIG_IP_NF_FILTER: enabled (as module)
- CONFIG_IP_NF_TARGET_MASQUERADE: enabled (as module)
- CONFIG_NETFILTER_XT_MATCH_ADDRTYPE: missing
- CONFIG_NETFILTER_XT_MATCH_CONNTRACK: enabled (as module)
- CONFIG_NETFILTER_XT_MATCH_IPVS: missing
- CONFIG_IP_NF_NAT: enabled (as module)
- CONFIG_NF_NAT: enabled (as module)
- CONFIG_NF_NAT_NEEDED: enabled
- CONFIG_POSIX_MQUEUE: enabled

Optional Features:
- CONFIG_USER_NS: enabled
- CONFIG_SECCOMP: enabled
- CONFIG_CGROUP_PIDS: enabled
- CONFIG_MEMCG_SWAP: enabled
- CONFIG_MEMCG_SWAP_ENABLED: enabled
    (cgroup swap accounting is currently enabled)
- CONFIG_LEGACY_VSYSCALL_NONE: enabled
    (containers using eglibc <= 2.13 will not work. Switch to
     "CONFIG_VSYSCALL_[NATIVE|EMULATE]" or use "vsyscall=[native|emulate]"
     on kernel command line. Note that this will disable ASLR for the,
     VDSO which may assist in exploiting security vulnerabilities.)
- CONFIG_BLK_CGROUP: enabled
- CONFIG_BLK_DEV_THROTTLING: missing
- CONFIG_IOSCHED_CFQ: enabled
- CONFIG_CFQ_GROUP_IOSCHED: missing
- CONFIG_CGROUP_PERF: enabled
- CONFIG_CGROUP_HUGETLB: missing
- CONFIG_NET_CLS_CGROUP: enabled
- CONFIG_CGROUP_NET_PRIO: enabled
- CONFIG_CFS_BANDWIDTH: missing
- CONFIG_FAIR_GROUP_SCHED: enabled
- CONFIG_RT_GROUP_SCHED: enabled
- CONFIG_IP_NF_TARGET_REDIRECT: enabled (as module)
- CONFIG_IP_VS: missing
- CONFIG_IP_VS_NFCT: missing
- CONFIG_IP_VS_PROTO_TCP: missing
- CONFIG_IP_VS_PROTO_UDP: missing
- CONFIG_IP_VS_RR: missing
- CONFIG_EXT4_FS: enabled
- CONFIG_EXT4_FS_POSIX_ACL: enabled
- CONFIG_EXT4_FS_SECURITY: enabled
- Network Drivers:
  - "overlay":
    - CONFIG_VXLAN: missing
    - CONFIG_BRIDGE_VLAN_FILTERING: missing
      Optional (for encrypted networks):
      - CONFIG_CRYPTO: enabled
      - CONFIG_CRYPTO_AEAD: enabled
      - CONFIG_CRYPTO_GCM: enabled (as module)
      - CONFIG_CRYPTO_SEQIV: enabled
      - CONFIG_CRYPTO_GHASH: enabled (as module)
      - CONFIG_XFRM: enabled
      - CONFIG_XFRM_USER: enabled (as module)
      - CONFIG_XFRM_ALGO: enabled
      - CONFIG_INET_ESP: enabled (as module)
      - CONFIG_INET_XFRM_MODE_TRANSPORT: enabled
  - "ipvlan":
    - CONFIG_IPVLAN: missing
  - "macvlan":
    - CONFIG_MACVLAN: missing
    - CONFIG_DUMMY: enabled (as module)
  - "ftp,tftp client in container":
    - CONFIG_NF_NAT_FTP: enabled (as module)
    - CONFIG_NF_CONNTRACK_FTP: enabled (as module)
    - CONFIG_NF_NAT_TFTP: enabled (as module)
    - CONFIG_NF_CONNTRACK_TFTP: enabled (as module)
- Storage Drivers:
  - "aufs":
    - CONFIG_AUFS_FS: missing
  - "btrfs":
    - CONFIG_BTRFS_FS: enabled
    - CONFIG_BTRFS_FS_POSIX_ACL: enabled
  - "devicemapper":
    - CONFIG_BLK_DEV_DM: enabled
    - CONFIG_DM_THIN_PROVISIONING: missing
  - "overlay":
    - CONFIG_OVERLAY_FS: missing
  - "zfs":
    - /dev/zfs: missing
    - zfs command: missing
    - zpool command: missing

Limits:
- /proc/sys/kernel/keys/root_maxkeys: 1000000

which leaves me to believe that there are a few kernel modules not enabled in the warrior build of yocto.

Were you able to get docker running on warrior?

ksaye, 2020-01-08:
@hb-peer , I just rebuild using the machine type 'genericx86-64' with the yocto build of warrior and included the 'kernel-modules' and iotedge is running just fine.

Sadly, the genericx86-64 does not include my tpm drivers, so any suggestion on how to modify the kernel features for 'intel-corei7-64' would be great.  :)

hb-peer, 2020-01-09:
Hello!

Nice to hear that it is working with a fresh rebuilt of generic x86 64. This is what I used.
I am sorry, I have no experience regarding TPM drivers.

Best regards,

Peer

@hb-peer<https://smex-ctp.trendmicro.com:443/wis/clicktime/v1/query?url=https%3a%2f%2fgithub.com%2fhb%2dpeer&umid=a68b3a29-42cb-4e4d-9e51-d1f261b28e70&auth=720035bb746f82c04e83e800a4cd33f474cc4b20-b84468f8f5997c182a17e88699343b772127bd6f> , I just rebuild using the machine type 'genericx86-64' with the yocto build of warrior and included the 'kernel-modules' and iotedge is running just fine.

Sadly, the genericx86-64 does not include my tpm drivers, so any suggestion on how to modify the kernel features for 'intel-corei7-64' would be great. :)

ksaye, 2020-01-10:
In working with the Intel meta layer, the issue with 'intel-corei7-64' is that meta-virtualization is hard coded to include the docker kernel modules to only the genericx86-64 machine type.   Running the following 2 commands resolve the issue.

1. cp meta-virtualization/recipes-kernel/linux/linux-yocto_4.19.bbappend meta-virtualization/recipes-kernel/linux/linux-intel_4.19.bbappend 
2. cp meta-virtualization/recipes-kernel/linux/linux-yocto_virtualization.inc meta-virtualization/recipes-kernel/linux/linux-intel_virtualization.inc

This copies the needed kernel modifications to include the meta-intel.  More info here: https://bugzilla.yoctoproject.org/show_bug.cgi?id=13727#c2

micahl, 2021-07-09:
Closing this issue

closed 2021-07-09 by micahl