Azure/meta-iotedge #29: Build error with master/sumo brand of meta-rust

LinhTran1, 2019-10-11:
I tried to build meta-iotedge for raspberry however it did not succeed. I also tried to switch between branch of meta-iotedge and meta-rust as follows:
master meta-iotedge + master meta-rust
sumo meta-iotedge + sumo meta-rust

Does folk here know how to to overcome this issue?

ksaye, 2019-10-25:
@LinhTran1 , As you see, Yocto is very version specific.  I recently had the same issue and perhaps you are seeing a version issue.  Try the commands: https://kevinsaye.wordpress.com/2019/06/21/how-to-build-a-yocto-image-for-the-raspberry-pi-running-azure-iot-edge/ (note the git checkout 9487b089ea4779c2b494b17b9254219226efa539) and if this works we might could merge this issue with https://github.com/Azure/meta-iotedge/issues/30

geoffrey-vl, 2020-03-17:
This is most probable solved?

wlenzeder, 2020-04-09:
I can confirm that building with current meta-rust master does not work.
In our case, the build is successful with an older rev 84710b341be53cc2af13ecc8ae87773cf84687bf (2019-04-24).

micahl, 2022-12-03:
Closing.  The latest 1.4 recipes are in the Dunfell branch.  We're not planning to port them to older branches, but are open to reviewing a pull request.

closed 2022-12-03 by micahl