Azure/meta-iotedge #30: ERROR: Task (virtual:native:/mnt/rpi//sources/meta-rust/recipes-devtools/cargo/cargo_0.25.0.bb:do_compile) failed with exit code '1'

ksaye, 2019-10-23:
**Describe the bug**
While it has worked in the past, recently attempting to compile the Yocto layers for a Raspberry Pi seem to fail, and it appears to be related to cargo.

**To Reproduce**
Steps to reproduce the behavior:

sudo apt update && sudo apt-get install -y gawk wget git-core diffstat unzip texinfo gcc-multilib \
     build-essential chrpath socat cpio python python3 python3-pip python3-pexpect \
     xz-utils debianutils iputils-ping python3-git python3-jinja2 libegl1-mesa libsdl1.2-dev \
     xterm locales

sudo mkdir -p /mnt/rpi/sources
sudo chown -R kevinsay:kevinsay /mnt/rpi
cd /mnt/rpi/sources
sudo locale-gen en_US.UTF-8

git clone -b sumo https://github.com/Azure/meta-iotedge.git
git clone -b sumo git://git.yoctoproject.org/meta-virtualization
git clone -b sumo git://github.com/meta-rust/meta-rust.git
git clone -b sumo git://git.yoctoproject.org/poky
git clone -b sumo git://git.openembedded.org/meta-openembedded
git clone -b sumo git://git.yoctoproject.org/meta-raspberrypi

cd /mnt/rpi
source sources/poky/oe-init-build-env rpi-build

cd /mnt/rpi/rpi-build/
echo '# Machine Selection' > conf/local.conf
echo 'MACHINE ?= "raspberrypi3-64"' >> conf/local.conf
echo 'DISTRO ?= "poky"' >> conf/local.conf
echo 'PACKAGE_CLASSES ?= "package_rpm"' >> conf/local.conf
echo 'EXTRA_IMAGE_FEATURES ?= "debug-tweaks ssh-server-dropbear tools-debug tools-sdk"' >> conf/local.conf
echo 'USER_CLASSES ?= "buildstats image-mklibs"' >> conf/local.conf
echo 'PATCHRESOLVE = "noop"' >> conf/local.conf
echo 'BB_DISKMON_DIRS = "\' >> conf/local.conf
echo '    STOPTASKS,${TMPDIR},1G,100K \' >> conf/local.conf
echo '    STOPTASKS,${DL_DIR},1G,100K \' >> conf/local.conf
echo '    STOPTASKS,${SSTATE_DIR},1G,100K \' >> conf/local.conf
echo '    STOPTASKS,/tmp,100M,100K \' >> conf/local.conf
echo '    ABORT,${TMPDIR},100M,1K \' >> conf/local.conf
echo '    ABORT,${DL_DIR},100M,1K \' >> conf/local.conf
echo '    ABORT,${SSTATE_DIR},100M,1K \' >> conf/local.conf
echo '    ABORT,/tmp,10M,1K"' >> conf/local.conf
echo 'SSTATE_MIRRORS = "\' >> conf/local.conf
echo '     file://.* http://sstate.yoctoproject.org/dev/PATH;downloadfilename=PATH \n \' >> conf/local.conf
echo '     file://.* http://sstate.yoctoproject.org/2.4.4/PATH;downloadfilename=PATH \n \' >> conf/local.conf
echo '     file://.* http://sstate.yoctoproject.org/2.5.2/PATH;downloadfilename=PATH \n \' >> conf/local.conf
echo '     file://.* http://sstate.yoctoproject.org/2.5.3/PATH;downloadfilename=PATH \n \' >> conf/local.conf
echo '     file://.* http://sstate.yoctoproject.org/2.6.2/PATH;downloadfilename=PATH \n \' >> conf/local.conf
echo '     file://.* http://sstate.yoctoproject.org/2.7/PATH;downloadfilename=PATH \n \' >> conf/local.conf
echo '     "' >> conf/local.conf
echo 'PACKAGECONFIG_append_pn-qemu-native = " sdl"' >> conf/local.conf
echo 'PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"' >> conf/local.conf
echo 'CONF_VERSION = "1"' >> conf/local.conf
echo 'IMAGE_INSTALL_append += "\' >> conf/local.conf
echo '                         iotedge-daemon \' >> conf/local.conf
echo '                         iotedge-cli \' >> conf/local.conf
echo '                         docker \' >> conf/local.conf
echo '                         docker-contrib \' >> conf/local.conf
echo '                         ca-certificates \' >> conf/local.conf
echo '                         connman \' >> conf/local.conf
echo '                         connman-client \' >> conf/local.conf
echo '                         "' >> conf/local.conf
echo 'DISTRO_FEATURES_append += " systemd wifi virtualization"' >> conf/local.conf
echo 'VIRTUAL-RUNTIME_init_manager = "systemd"' >> conf/local.conf
echo 'DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"' >> conf/local.conf
echo 'VIRTUAL-RUNTIME_initscripts = ""' >> conf/local.conf
echo '# we need space for images' >> conf/local.conf
echo 'IMAGE_ROOTFS_EXTRA_SPACE = "10485760"' >> conf/local.conf
echo '# The extra space takes us above 4gb, so' >> conf/local.conf
[... 263 lines cut ...]
| Caused by:
|   process didn't exit successfully: `rustc --crate-name rustc_demangle /mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/registry/src/github.com-1ecc6299db9ec823/rustc-demangle-0.1.16/src/lib.rs --crate-type lib --emit=dep-info,link -C opt-level=3 -C metadata=de64cc5bec0e8255 -C extra-filename=-de64cc5bec0e8255 --out-dir /mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/x86_64-linux/release/deps --target x86_64-linux -C linker=/mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-ccld -L dependency=/mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/x86_64-linux/release/deps -L dependency=/mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/target/release/deps --cap-lints allow -L /mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/lib/rust -Zremap-path-prefix-from=/mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0 -Zremap-path-prefix-to=/usr/src/debug/cargo-native/0.25.0-r0` (exit code: 101)
| warning: build failed, waiting for other jobs to finish...
| error: build failed
| WARNING: exit code 101 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /mnt/rpi/rpi-build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.120401)
ERROR: Task (virtual:native:/mnt/rpi//sources/meta-rust/recipes-devtools/cargo/cargo_0.25.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4620 tasks of which 1426 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/mnt/rpi//sources/meta-rust/recipes-devtools/cargo/cargo_0.25.0.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
kevinsay@yoctobuild:/mnt/rpi$

**Build System (please complete the following information):**
 - OS: Ubuntu 18.04 LTS
 - Version 5.0.0-1020-azure #21~18.04.1-Ubuntu

**Target System (please complete the following information):**
 - OS: Raspberry Pi
 - Version 3

**Additional context**
Add any other context about the problem here.

ksaye, 2019-10-23:
Perhaps related to: https://github.com/meta-rust/meta-rust/issues/220

ksaye, 2019-10-25:
Update:  It looks like meta-rust has updated their rust version to a version that is incompatible.  

Using the following, I was able to 'get' a compatible version that worked.

```
git clone https://github.com/Azure/meta-iotedge.git
git clone -b sumo git://git.yoctoproject.org/meta-virtualization
git clone -b master git://github.com/meta-rust/meta-rust.git
cd meta-rust
git checkout 9487b089ea4779c2b494b17b9254219226efa539
cd ..
git clone -b sumo git://git.yoctoproject.org/poky
git clone -b sumo git://git.openembedded.org/meta-openembedded
git clone -b sumo git://git.yoctoproject.org/meta-intel
git clone -b sumo git://git.yoctoproject.org/meta-raspberrypi
```

geoffrey-vl, 2020-03-17:
Issue is solved using the correct version of meta-rust, please close

vaishnavils97, 2020-11-17:
> Issue is solved using the correct version of meta-rust, please close

Please suggest which all branches of meta-rust & iotedge  should have to use for a proper yocto
 build .

closed 2021-01-26 by ksaye