Azure/meta-iotedge #31: Support for Yocto 3.0 (Zeus)

wlenzeder, 2019-11-29:
Do you have an estimation when Yocto 3.0 (Zeus) will be supported by meta-iotedge?

Siddius, 2020-04-08:
#36 have a look here ;)

vjdaemp, 2020-04-15:
@Siddius Great news, I will try it.

vjdaemp, 2020-04-17:
Update: The +v8 related error was coming from meta-rust (master). I reverted to an older version of meta-rust and am now receiving a different issue related to meta-iotedge:

`
ERROR: Nothing PROVIDES 'docker' (but /home/vjdaemp/Jetson/Build/jetson-nano-devkit/meta-iotedge/recipes-core/iotedge-daemon/iotedge-daemon_1.0.9.bb DEPENDS on or otherwise requires it). Close matches:
  docker-ce
  runc-docker
  docker-moby
  docker-ce RPROVIDES docker
NOTE: Runtime target 'iotedge-daemon' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['iotedge-daemon', 'docker']
ERROR: Required build target 'core-image-base' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-base', 'iotedge-daemon', 'docker']
`
I tried removing `docker` from `DEPENDS` in `iotedge-daemon.inc` and then encountered a different error:

Running groupadd commands...
Running useradd commands...
useradd: group 'docker' does not exist
WARNING: exit code 1 from a shell command.

```
ERROR: Logfile of failure stored in: /home/vjdaemp/Jetson/Build/jetson-nano-devkit/poky/build/tmp/work/aarch64-poky-linux/iotedge-daemon/1.0.9-r0/temp/log.do_prepare_recipe_sysroot.2724
ERROR: Task (/home/vjdaemp/Jetson/Build/jetson-nano-devkit/meta-iotedge/recipes-core/iotedge-daemon/iotedge-daemon_1.0.9.bb:do_prepare_recipe_sysroot) failed with exit code '1'

```

Do you know how to resolve these errors, or can you recommend which version  of meta-rust you used to build successfully? I am trying to build meta-iotedge for zeus.

By the way, I am new to yocto so I may be missing something.

Thanks!
Vincent

Siddius, 2020-04-22:
@vjdaemp yeah this error with docker you get because iotegde relies on docker. Add _docker-ce_  to your installed packages. in local.conf for example:
`IMAGE_INSTALL_append = "  docker-ce" `

I was using master or rust-meta commit a012a1027defe28495f06ed522a7a82bdd59a610

mirzabrc, 2020-07-15:
@Siddius Thanks for your work so far! I encounter the following Errors:

```
No recipes available for:
  /sources/meta-iotedge/recipes-devtools/cargo/cargo_1.41.0.bbappend
```

and

```
ERROR: Nothing PROVIDES 'docker' (but sources/meta-iotedge/recipes-core/iotedge-daemon/iotedge-daemon_1.0.9.bb DEPENDS on or otherwise requires it). Close matches:
  docker-ce
  runc-docker
  docker-moby
  docker-ce RPROVIDES docker

ERROR: Nothing RPROVIDES 'iotedge-daemon-dev' (but /sources/meta-iotedge/recipes-core/iotedge-daemon/iotedge-daemon_1.0.9.bb RDEPENDS on or otherwise requires it)
No eligible RPROVIDERs exist for 'iotedge-daemon-dev'

```
I added IMAGE_INSTALL_append = " docker-ce"  to my local.conf and edited the first line of iotedge-daemon.inc
from
`DEPENDS += "openssl docker virtual/libiothsm"`
to
`DEPENDS += "openssl docker-ce virtual/libiothsm"`
which seemes to fix it?

Can you help me out here?

Siddius, 2020-07-15:
@mirzabrc i am sorry you are totally right, changed `virtual/docker`

have you checked out https://github.com/meta-rust/meta-rust ?

Siddius, 2020-07-15:
@mirzabrc layers.conf ? layers are configured in bblayers.conf right ?

mirzabrc, 2020-07-15:
@Siddius you're absolutely right, i got confused there for a minute.
meta-rust is checked out and added in bblayers.conf
Would there be another workaround without /recipes-devtools/.. ?

Siddius, 2020-07-16:
@mirzabrc it should be fine if you work with the master branch of  meta-rust.
If you use the commit hash i mentioned earlier, `cargo_1.43.0.bb` is not present. so you can just delete `cargo_1.43.0.bbappend`

mirzabrc, 2020-07-17:
@Siddius When using meta-rust master I get the same +v8 related error as @vjdaemp
I will revert to meta-rust commit a012a1027defe28495f06ed522a7a82bdd59a610 and give an update later. I removed cargo_1.43.0.bbappend.

Also virtual/docker does not work for me, I had to change it to docker-ce.
local.conf:
`IMAGE_INSTALL_append = " docker-ce iotedge-cli iotedge-daemon libiothsm-std"`
iotedge-daemon.inc:
```
DEPENDS += "openssl docker-ce virtual/libiothsm"
RDEPENDS_${PN} += "virtual/libiothsm docker-ce"
```

shr-project, 2021-02-08:
The "'+v8' is not a recognized feature for this target (ignoring feature)" message is just an warning, not the actual error, see:
https://github.com/meta-rust/meta-rust/pull/293
which fixes this warning and
https://github.com/Azure/meta-iotedge/pull/58
which fixes compatibility with newer rust.

vaishnavils97, 2021-11-08:
@shr-project @Siddius  I am building Yocto image for jetson tx2 and got stuck with this error.
TE: Runtime target 'iotedge-daemon' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['iotedge-daemon']
ERROR: Required build target 'core-image-minimal' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-minimal', 'iotedge-daemon']

The commit id of the repositories are 
Submodule path 'repos/meta-mender': checked out 'e47297c6aee747c331663d8acb933e85d3d55812'
Submodule path 'repos/meta-mender-community': checked out '2f52cf4948bc964f1e5f3f4193b7cf33c1b13037'
Submodule path 'repos/meta-openembedded': checked out '814eec96c2a29172da57a425a3609f8b6fcc6afe'
Submodule path 'repos/meta-tegra': checked out '227c8ba5bd69a62ce86420b7833befcbb57bc33f'
Submodule path 'repos/meta-tegra-community': checked out '4539e0ae1f8145d8c01c190bad650d5adce0c620'
Submodule path 'repos/meta-virtualization': checked out 'f0d8a55c220fdec46927c106568657b725e17126'
Submodule path 'repos/poky': checked out '0810ac6b926cd901f0619e95f367efc79d4c3159'

meta-iotedge :  '6840cf2afb51b4f2c535a05d5635736652f664f7'.
meta- rust : 3b783652cc9e0e0ff95fdf151eab3e178b5f1667

please help me to fix this bug

shr-project, 2021-11-08:
@vaishnavils97 I'm not interested in this layer anymore and removed it from all our builds, see https://github.com/Azure/meta-iotedge/issues/38 and the pending PRs, I recommend to stop using it as well.

vaishnavils97, 2021-11-08:
@shr-project  , so should I have to switch to some other branch instead of dunfell. Can you please suggest a solution?

MZDN, 2022-05-16:
any update on this issue ?

micahl, 2022-12-03:
Closing.  The latest 1.4 recipes are in the Dunfell branch.  We're not planning to port them to older branches, but are open to reviewing a pull request.

closed 2022-12-03 by micahl

manojrampur, 2023-05-16:
We are currently in the process of building a Yocto image for the Raspberry Pi 3 device. Our objective is to incorporate the defender-iot-micro-agent-edge into the Yocto build. To achieve this, we have upgraded the meta-iot edge layer to version 1.4.9. However, even after flashing the image, we are unable to use or observe the presence of the defender-iot-micro-agent-edge on the target device. We are seeking guidance on how to include the defender-iot-micro-agent-edge in the Yocto build effectively, ensuring it is reflected and accessible on the target device. The commit IDs for the relevant layers are as follows:

meta-iotedge: db4541e7c9edb718682275ec1816ac2ed163e8a7
meta-rust: 5656b5ad104bbbbe1b711e3bbe35c88f0d9c9a05