Azure/meta-iotedge #34: Sumo compile error: libssl.so: file not recognized: File format not recognized

geoffrey-vl, 2020-03-17:
**Describe the bug**
I'm trying to get the Sumo branch to build on Yocto Rocko, since it is closest to Rocko and best maintained. During compiling it fails to find `libssl` for what I understand: `libssl.so: file not recognized: File format not recognized`.

Its probable related to https://github.com/Azure/meta-iotedge/issues/22#issuecomment-546474794

**To Reproduce**
Steps to reproduce the behavior:
1. add meta-vertualization
2. add meta-rust (version 9487b089ea4779c2b494b17b9254219226efa539)
3. add meta-iotedge
4. bitbake iotedge-cli

**Expected behavior**
Should compile and package

**Logs**
```
| #Start building HSM dev-mode library
| detected home dir change, cleaning out entire build directory
| running: "cmake" "/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/hsm-sys/azure-iot-hsm-c" "-Duse_openssl=ON" "-DCMAKE_BUILD_TYPE=Release" "-Drun_unittests=ON" "-Duse_default_uuid=ON" "-Duse_http=OFF" "-Dskip_samples=ON" "-Drun_valgrind=OFF" "-Duse_emulator=OFF" "-DBUILD_SHARED=ON" "-DUSE_TEST_TPM_INTERFACE_IN_MEM=OFF" "-DCMAKE_INSTALL_PREFIX=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/build/hsm-sys-fe99d18cce916c08/out" "-DCMAKE_C_FLAGS= -ffunction-sections -fdata-sections -fPIC -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0=/usr/src/debug/iotedge-cli/1.0.8-r0 -fdebug-prefix-map=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot=" "-DCMAKE_C_COMPILER=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-cc" "-DCMAKE_CXX_FLAGS= -ffunction-sections -fdata-sections -fPIC -pipe -feliminate-unused-debug-types -fdebug-prefix-map=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0=/usr/src/debug/iotedge-cli/1.0.8-r0 -fdebug-prefix-map=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot-native= -fdebug-prefix-map=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot= -fvisibility-inlines-hidden" "-DCMAKE_CXX_COMPILER=/home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-cxx"
| -- The C compiler identification is GNU 7.3.0
| -- The CXX compiler identification is GNU 7.3.0
| -- Check for working C compiler: /home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-cc
| -- Check for working C compiler: /home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-cc -- works
| -- Detecting C compiler ABI info
| -- Detecting C compiler ABI info - done
| -- Detecting C compile features
| -- Detecting C compile features - done
| -- Check for working CXX compiler: /home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-cxx
| -- Check for working CXX compiler: /home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-cxx -- works
| -- Detecting CXX compiler ABI info
| -- Detecting CXX compiler ABI info - done
| -- Detecting CXX compile features
| -- Detecting CXX compile features - done
| -- Found OpenSSL: /home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/lib/libcrypto.so (found version "1.0.2n")
| -- Looking for include file stdint.h
| -- Looking for include file stdint.h - found
| -- Looking for include file stdbool.h
| -- Looking for include file stdbool.h - found
| -- target architecture: ARM
| -- Performing Test CXX_FLAG_CXX11
| -- Performing Test CXX_FLAG_CXX11 - Success
| -- target architecture: ARM
| -- Configuring done
| -- Generating done
| -- Build files have been written to: /home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/build/hsm-sys-fe99d18cce916c08/out/build
| running: "cmake" "--build" "." "--target" "install" "--config" "Release" "--"
| Scanning dependencies of target aziotsharedutil
| [  0%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/azure_base32.c.o
| [  0%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/azure_base64.c.o
| [  1%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/buffer.c.o
| [  1%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/constbuffer_array.c.o
| [  2%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/connection_string_parser.c.o
| [  2%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/constbuffer.c.o
| [  3%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/consolelogger.c.o
| Scanning dependencies of target ctest
| Scanning dependencies of target testrunnerswitcher
| [  3%] Building C object tests/c-shared/testtools/ctest/CMakeFiles/ctest.dir/src/ctest.c.o
| [  4%] Building C object tests/c-shared/testtools/testrunner/CMakeFiles/testrunnerswitcher.dir/src/testmutex.c.o
| [  4%] Building C object tests/c-shared/testtools/testrunner/CMakeFiles/testrunnerswitcher.dir/src/ctrs_sprintf.c.o
| [  4%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/crt_abstractions.c.o
| [  5%] Linking C static library libtestrunnerswitcher.a
| [  5%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/constmap.c.o
| [  5%] Built target testrunnerswitcher
| [  6%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/doublylinkedlist.c.o
| Scanning dependencies of target umock_c
| [  6%] Building C object deps/c-shared/deps/umock-c/CMakeFiles/umock_c.dir/src/umock_c.c.o
| [  6%] Building C object deps/c-shared/CMakeFiles/aziotsharedutil.dir/src/gballoc.c.o
| [  7%] Linking C static library libctest.a
| [  7%] Built target ctest
[... 226 lines cut ...]
|              at src/libstd/rt.rs:48
|   15: main
|   16: __libc_start_main
|   17: _start
|
| WARNING: exit code 101 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/jenkins/build/build-x11-pico-imx7/tmp/work/cortexa7hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/temp/log.do_compile.78571)
ERROR: Task (/home/jenkins/build/sources/meta-iotedge/recipes-core/iotedge-cli/iotedge-cli_1.0.8.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 916 tasks of which 832 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/jenkins/build/sources/meta-iotedge/recipes-core/iotedge-cli/iotedge-cli_1.0.8.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```

**Build System (please complete the following information):**
 - OS: Ubuntu 16.04 docker container
 - Host: Ubuntu 18.04

**Target System (please complete the following information):**
 - OS: Yocto
 - Version 2.4 Rocko (linux 4.9)

**Additional context**
Image is working including Docker without the azure iot-edge components

geoffrey-vl, 2020-03-17:
fixed by #35

wlenzeder, 2020-04-09:
I can confirm that build for Yocto Sumo is fixed by #35

closed 2022-12-03 by micahl