Azure/meta-iotedge #37: thud not compatible with meta-rust HEAD?

gvkalra, 2020-05-18:
**Describe the bug**
Unable to compile with meta-rust HEAD (be88d857a6ba9134abb795b3a34d3a839196335f)

**To Reproduce**
1. Add the layers
2. Compile

**Expected behavior**
`iotedge-cli` should compile normally.

**Logs**
```
$ bitbake iotedge-cli
[...]
[...]
[...]
ERROR: iotedge-cli-1.0.8-r0 do_compile: Function failed: do_compile (log file is located at /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/temp/log.do_compile.24782)
ERROR: Logfile of failure stored in: /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/temp/log.do_compile.24782
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target arm-poky-linux-gnueabi --release
|     Updating git repository `https://github.com/Azure/hyperlocal-windows`
|     Updating git repository `https://github.com/Azure/mio-uds-windows.git`
|     Updating git repository `https://github.com/Azure/tokio-uds-windows.git`
|    Compiling libc v0.2.42
|    Compiling cfg-if v0.1.2
|    Compiling cc v1.0.15
|    Compiling byteorder v1.3.1
|    Compiling futures v0.1.24
|    Compiling lazy_static v1.3.0
|    Compiling unicode-xid v0.1.0
|    Compiling serde v1.0.84

[....]
[....]
[....]

|    Compiling linked-hash-map v0.3.0
|      Running `rustc --crate-name linked_hash_map /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/linked-hash-map-0.3.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 --cfg 'feature="serde"' --cfg 'feature="serde_impl"' --cfg 'feature="serde_test"' -C metadata=9ddf1d1d6ce60bc4 -C extra-filename=-9ddf1d1d6ce60bc4 --out-dir /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-ccld -L dependency=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps -L dependency=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps --extern serde=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libserde-a93048c59156fa3c.rmeta --extern serde_test=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libserde_test-70db6facd39fab8b.rmeta --cap-lints allow -L /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0=/usr/src/debug/iotedge-cli/1.0.8-r0`
| error[E0713]: borrow may still be in use when destructor runs
|    --> /usr/src/debug/iotedge-cli/1.0.8-r0/cargo_home/bitbake/url-1.7.0/src/form_urlencoded.rs:261:40
|     |
| 259 | impl<'a> Target for ::UrlQuery<'a> {
|     |      -- lifetime `'a` defined here
| 260 |     fn as_mut_string(&mut self) -> &mut String { &mut self.url.serialization }
| 261 |     fn finish(self) -> &'a mut ::Url { self.url }
|     |                                        ^^^^^^^^ - here, drop of `self` needs exclusive access to `*self.url`, because the type `UrlQuery<'_>` implements the `Drop` trait
|     |                                        |
|     |                                        returning this value requires that `*self.url` is borrowed for `'a`
| 
|    Compiling tokio-threadpool v0.1.6
|      Running `rustc --crate-name tokio_threadpool /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/tokio-threadpool-0.1.6/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=7f2666776fd09d03 -C extra-filename=-7f2666776fd09d03 --out-dir /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-ccld -L dependency=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps -L dependency=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps --extern crossbeam_deque=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libcrossbeam_deque-682cde0e99f8f571.rmeta --extern crossbeam_utils=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libcrossbeam_utils-2d44eac8bd722b16.rmeta --extern futures=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libfutures-a015e1792da071bb.rmeta --extern log=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/liblog-b8905e89877938b7.rmeta --extern num_cpus=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libnum_cpus-faced388d885ee1b.rmeta --extern rand=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/librand-b97aedc7c95a3d1b.rmeta --extern tokio_executor=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libtokio_executor-f94ea6b3952cada8.rmeta --cap-lints allow -L /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0=/usr/src/debug/iotedge-cli/1.0.8-r0`
| error: aborting due to previous error
| 
| For more information about this error, try `rustc --explain E0713`.
| error: could not compile `url`.
| 
| Caused by:
|   process didn't exit successfully: `rustc --crate-name url /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/url-1.7.0/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=e798a36d88fd2d07 -C extra-filename=-e798a36d88fd2d07 --out-dir /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-ccld -L dependency=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps -L dependency=/mnt/seesv-distro/build-r [... cut ...]
meta-rust            = "HEAD:be88d857a6ba9134abb795b3a34d3a839196335f"
meta-virtualization  = "HEAD:7685c7d415e0002c448007960837ae8898cd57a5"
meta-iotedge         = "HEAD:ffd829d144eef4e255af9f8b4bd814567114a700"
meta-seesv           = "azure-iot:820235a7b6908dffb3458b3466fff0e57eb37eb4"
```

**Additional context**
It seems like a dependency error on `url`.
However, the documentation of `meta-iotedge` recommends HEAD revision of meta-rust.

Should `meta-rust` be pinned to some other compatible revision?

gvkalra, 2020-05-20:
With `meta-rust` pinned at 0f950f5e333a1c8999320bf18232144f3dd9c80e, `iotedge-cli` compiles. However, `iotedge-daemon` fails with:

```
[...]
Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "seesv"
DISTRO_VERSION       = "2.6.4"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
SDKMACHINE           = "x86_64"
meta                 
meta-poky            = "HEAD:51f6145f8f99a02df1dad937684f014b0172e72a"
meta-raspberrypi     = "HEAD:89bdf97beaf3b136c24c4d6f86c0bcbb47fe3043"
meta-oe              
meta-python          
meta-networking      
meta-initramfs       
meta-multimedia      
meta-filesystems     = "HEAD:446bd615fd7cb9bc7a159fe5c2019ed08d1a7a93"
meta-boot2qt         
meta-boot2qt-distro  
meta-raspberrypi-extras = "HEAD:6912408e3508d2dac2f9a26638b0db8fd3a0da74"
meta-mingw           = "HEAD:fa4ea76d93c87a2ad3ce88b052c282106b2761e4"
meta-qt5             = "HEAD:f7afa0a4e1348db01be88b9c36372842fb902136"
meta-mender-core     
meta-mender-raspberrypi = "HEAD:4014636a4684194d68e7d480b749211e7ab0c02a"
meta-rust            = "HEAD:0f950f5e333a1c8999320bf18232144f3dd9c80e"
meta-virtualization  = "HEAD:7685c7d415e0002c448007960837ae8898cd57a5"
meta-iotedge         = "HEAD:ffd829d144eef4e255af9f8b4bd814567114a700"
meta-seesv           = "azure-iot:fc4e41c4594373240607e7d9cf395ab0b73ff816"
[...]
[...]
ERROR: iotedge-daemon-1.0.8-r0 do_compile: Function failed: do_compile (log file is located at /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/temp/log.do_compile.27013)
ERROR: Logfile of failure stored in: /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/temp/log.do_compile.27013
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target arm-poky-linux-gnueabi --release
|     Updating git repository `https://github.com/Azure/hyperlocal-windows`
|     Updating git repository `https://github.com/Azure/mio-uds-windows.git`
|     Updating git repository `https://github.com/Azure/tokio-uds-windows.git`
|    Compiling libc v0.2.42
|    Compiling cfg-if v0.1.2
|    Compiling futures v0.1.24
|    Compiling cc v1.0.15
|    Compiling byteorder v1.3.1
|    Compiling lazy_static v1.3.0
|    Compiling slab v0.4.1
|    Compiling unicode-xid v0.1.0
[...]
[...]
[...]
|      Running `rustc --crate-name workload workload/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=b6c5e4d7a884afd5 -C extra-filename=-b6c5e4d7a884afd5 --out-dir /mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/wrapper/target-rust-ccld -L dependency=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps -L dependency=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps --extern base64=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libbase64-7718d933690cd576.rmeta --extern failure=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libfailure-b29c70362ceba458.rmeta --extern futures=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libfutures-9445e420a627eb7c.rmeta --extern hyper=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libhyper-677148a312ad3684.rmeta --extern serde=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libserde-ded32a7c0e559431.rmeta --extern serde_derive=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps/libserde_derive-de98041c38a7353f.so --extern serde_json=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/deps/libserde_json-bdc52b52ab08d3af.rmeta --extern serde_yaml=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet [... cut ...] map-path-prefix=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0=/usr/src/debug/iotedge-daemon/1.0.8-r0 -L native=/mnt/seesv-distro/build-raspberrypi3/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/iotedge-daemon/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-poky-linux-gnueabi/release/build/backtrace-sys-829f3094567b1d14/out/.libs` (exit code: 1)
| warning: build failed, waiting for other jobs to finish...
| error: build failed
```

kbumsik, 2020-05-22:
I got the same issue.
Even I switched to a quite old commit:

```
commit 969467c029422123aa5c445a1579ce23ff200795
Merge: 4c9068b 665213a
Author: Steven Walter <stevenrwalter@gmail.com>
Date:   Mon Dec 17 13:05:37 2018 -0500

    Merge pull request #222 from YoeDistro/yoe/mut

    libstd-rs: Depend on libunwind on musl builds
```

But I still get error:
```
DEBUG: Executing shell function do_compile
NOTE: cargo = /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/recipe-sysroot-native/usr/bin/cargo
NOTE: rustc = 
NOTE: cargo build -v --target arm-ostl-linux-gnueabi --release 
   Compiling proc-macro2 v0.4.19
   Compiling byteorder v1.3.1
   Compiling unicode-xid v0.1.0
   Compiling nodrop v0.1.12
   Compiling libc v0.2.42
   Compiling cfg-if v0.1.2
   Compiling cc v1.0.15
   Compiling memoffset v0.2.1
   Compiling crossbeam-utils v0.5.0
   Compiling lazy_static v1.3.0
   Compiling scopeguard v0.3.3
   Compiling lazycell v0.6.0
   Compiling typenum v1.10.0
   Compiling rand_core v0.2.1
   Compiling slab v0.4.1
   Compiling futures v0.1.24
   Compiling pkg-config v0.3.11
   Compiling version_check v0.1.3
   Compiling ryu v0.2.7
   Compiling failure_derive v0.1.2
   Compiling serde v1.0.84
   Compiling itoa v0.4.1
   Compiling fnv v1.0.6
   Compiling rustc-demangle v0.1.7
   Compiling openssl v0.10.12
   Compiling matches v0.1.6
   Compiling foreign-types-shared v0.1.1
   Compiling arrayref v0.3.4
   Compiling byte-tools v0.2.0
   Compiling bitflags v1.0.3
   Compiling native-tls v0.2.1
   Compiling unicode-normalization v0.1.5
     Running `rustc --crate-name build_script_build /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/proc-macro2-0.4.19/build.rs --color never --crate-type bin --emit=dep-info,link -C opt-level=3 --cfg 'feature="default"' --cfg 'feature="proc-macro"' -C metadata=63ae86250170909c -C extra-filename=-63ae86250170909c --out-dir /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/build/proc-macro2-63ae86250170909c -C linker=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/build-rust-ccld -L dependency=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps --cap-lints allow`
     Running `rustc --crate-name build_script_build /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/byteorder-1.3.1/build.rs --color never --crate-type bin --emit=dep-info,link -C opt-level=3 --cfg 'feature="default"' --cfg 'feature="std"' -C metadata=1952b976fe34ac33 -C extra-filename=-1952b976fe34ac33 --out-dir /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/build/byteorder-1952b976fe34ac33 -C linker=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/build-rust-ccld -L dependency=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps --cap-lints allow`
     Running `rustc --crate-name unicode_xid /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/unicode-xid-0.1.0/src/lib.rs --color never --crate-type lib --emit=dep-info,link -C opt-level=3 --cfg 'feature="default"' -C metadata=e884412bfa15afb7 -C extra-filename=-e884412bfa15afb7 --out-dir /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps -C linker=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/build-rust-ccld -L dependency=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps --cap-lints allow`
     Running `rustc --crate-name nodrop /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/cargo_home/bitbake/nodrop-0.1.12/src/lib.rs --color never --crate-type lib --emit=dep-info,link -C opt-level=3 -C metadata=8fb387bdc922aa82 -C extra-filename=-8fb387bdc922aa82 --out-dir /datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-ostl-linux-gnueabi/release/deps --target arm-ostl-linux-gnueabi -C linker=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/wrapper/target-rust-ccld -L dependency=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ost [... cut ...] /1.0.8-r0/wrapper/target-rust-ccld -L dependency=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/arm-ostl-linux-gnueabi/release/deps -L dependency=/datadrive/exodus_linux/build-openstlinuxweston-stm32mp1/tmp-glibc/work/cortexa7t2hf-neon-vfpv4-ostl-linux-gnueabi/iotedge-cli/1.0.8-r0/iotedge-1.0.8/edgelet/target/release/deps --extern block_buffer=/datadrive/exodus_linux/build-openst

kbumsik, 2020-05-22:
@gvkalra I successfully built with `9487b089ea4779c2b494b17b9254219226efa539`! Try this commit. I got this from here: https://kevinsaye.wordpress.com/2019/06/21/how-to-build-a-yocto-image-for-the-raspberry-pi-running-azure-iot-edge/

gvkalra, 2020-05-22:
@kbumsik thanks! worked for me as well.

geoffrey-vl, 2020-08-18:
Same as #33 but other branch

myagley, 2020-09-25:
With the 1.0.9 release of IoT Edge, we began pinning the version of the rust compiler used (via the rust-toolchain file). This should mitigate many of these issues going forward.

In order to support this in Yocto, we will need to start pinning the meta-rust version or you can use `PREFERRED_VERSION_*` in your distro config. For example, 1.0.9.4 of IoT Edge on Thud is pinned to meta-rust `d8d77be1292064a02adcb5e72e293604b704f69b`.

I'm open to other options to make this work other than pinning the meta-rust commit.

gvkalra, 2020-10-12:
Thanks, @myagley. Pinning is perfect.

Just for reference: Fixed by https://github.com/Azure/meta-iotedge/commit/f68eb87a9408d7a4d3d22cbb6106d353199c6c00

closed 2020-10-12 by gvkalra