Azure/meta-iotedge #52: Is the container engine Docker or moby-engine used?

SimonSimCity, 2020-11-20:
As I had some troubles getting Azure iotedge to run, I was looking through the platform support where I read (https://docs.microsoft.com/en-us/azure/iot-edge/support?view=iotedge-2018-06#container-engines) that:

> Microsoft provides a container engine, moby-engine, to fulfill this requirement. This container engine is based on the Moby open-source project. Docker CE and Docker EE are other popular container engines. They're also based on the Moby open-source project and are compatible with Azure IoT Edge. Microsoft provides best effort support for systems using those container engines; however, Microsoft can't ship fixes for issues in them. _For this reason, Microsoft recommends using moby-engine on production systems._

The current version of the code seems to include `Docker` as a dependency:
https://github.com/Azure/meta-iotedge/blob/736078713e4f81a59500580ef8811fee5cc07a23/recipes-core/iotedge-daemon/iotedge-daemon.inc#L1-L2

Isn't this against the recommendations? What's the reason here? Shouldn't this be changed?

emilm, 2020-12-08:
You can do this in your distro.conf or local.conf:

PREFERRED_RPROVIDER_docker = "docker-moby"
PREFERRED_RPROVIDER_docker = "docker-moby"

Lexmark-chad, 2021-01-08:
It's pointing to `Docker` because Microsoft does not yet share it's own fork of the moby-engine.  And, they won't until Yocto becomes a tier1 supported OS.  See this issue:

https://github.com/Azure/iotedge/issues/3921

Lexmark-chad, 2022-02-09:
@micahl, if you've eliminated any proprietary changes and what is left is just packaging & MSFT-specific versioning then why no go ahead and open source that so https://github.com/Azure/iotedge/issues/3921 can be addressed?

micahl, 2022-02-09:
@Lexmark-chad I agree we should fix that issue (which I don't think requires open sourcing the moby repo).

closed 2022-12-03 by micahl