Azure/meta-iotedge #57: Compatibility with meta-rust

rschwarz-sotec, 2020-12-08:
I am currently facing build issues with dunfell release of meta-iotedge

**To Reproduce**
Steps to reproduce the behavior:
1. checkout dunfell branch of meta-iotedge (6840cf2afb51b4f2c535a05d5635736652f664f7)
2. checkout master branch of meta-rust (f56fd4ec6835d68c4e9f8a9630963acd34d172e5)
3. bitbake iotedge-cli
4. get build error, looks like a linter error (I am no rust expert)

**Expected behavior**
build should succeed

**Logs**
```
|    Compiling edgelet-http v0.1.0 (/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/edgelet-http)
|      Running `rustc --crate-name edgelet_http --edition=2018 edgelet-http/src/lib.rs --error-format=json --json=diagnostic-rendered-ansi,artifacts --crate-type lib --emit=dep-info,metadata,link -C opt-level=3 -C metadata=3dc1b6a45037546b -C extra-filename=-3dc1b6a45037546b --out-dir /workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps --target arm-poky-linux-gnueabi -C linker=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/wrapper/target-rust-ccld -L dependency=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps -L dependency=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/release/deps --extern bytes=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libbytes-dd0f37e00c0eedd2.rmeta --extern chrono=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libchrono-c1146bc9c2b0ef1d.rmeta --extern edgelet_core=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libedgelet_core-ec8f9464c45610f5.rmeta --extern edgelet_hsm=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libedgelet_hsm-cfd83c5c74c00d82.rmeta --extern edgelet_utils=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libedgelet_utils-524cd83c44babc4c.rmeta --extern failure=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libfailure-ce364f9f077b5127.rmeta --extern futures=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libfutures-aca041b306de4689.rmeta --extern hyper=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libhyper-832dad6ded1dc24a.rmeta --extern hyper_proxy=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libhyper_proxy-6756a32e37251477.rmeta --extern hyper_tls=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libhyper_tls-f70b00fd733cbb38.rmeta --extern hyperlocal=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libhyperlocal-f5f402b7e22b34c5.rmeta --extern libc=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/liblibc-24c527897b21a507.rmeta --extern log=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/liblog-f30e0a7e441e45f8.rmeta --extern native_tls=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libnative_tls-31fb5225095bf2a0.rmeta --extern nix=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libnix-02bc8316c0682b9f.rmeta --extern openssl=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libopenssl-9a0263a7c33b49c5.rmeta --extern percent_encoding=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libpercent_encoding-b2c4ba31f464ee67.rmeta --extern regex=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libregex-cd3b150756cefa32.rmeta --extern scopeguard=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libscopeguard-a097fdb2204421a2.rmeta --extern serde=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libserde-c43e1045ee3a80ed.rmeta --extern serde_derive=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/release/deps/libserde_derive-dd6698e0ed9ba976.so --extern serde_json=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libserde_json-e1348c97bf0e7ea0.rmeta --extern systemd=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libsystemd-f6edc130f32e2fc7.rmeta --extern tokio=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libtokio-63a2a65349863574.rmeta --extern tokio_tls=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libtokio_tls-64c9757a0861c707.rmeta --extern tokio_uds=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libtokio_uds-34a29d066104e347.rmeta --extern typed_headers=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/libtyped_headers-d75b81805b379a41.rmeta --extern url=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/deps/liburl-afd13367c21046f8.rmeta -L /workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/recipe-sysroot/usr/lib/rust --remap-path-prefix=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0=/usr/src/debug/iotedge-cli/1.0.9.4-r0 -L native=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/target/arm-poky-linux-gnueabi/release/build/backtrace-sys-c8fa18eabcc4429a/out -L native=/workdir/build-cloudplugedge/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/recipe-sysroot/usr/lib`
| error: deny(rust_2018_idioms) is ignored unless specified at crate level
|  --> edgelet-http/src/authentication.rs:3:9
|   |
| 3 | #![deny(rust_2018_idioms, warnings)]
|   |         ^^^^^^^^^^^^^^^^
|   |
| note: the lint level is defined here
[... cut ...]
| 3 | #![deny(rust_2018_idioms, warnings)]
|   |         ^^^^^^^^^^^^^^^^
|
| error: deny(rust_2018_idioms) is ignored unless specified at crate level
|  --> edgelet-http/src/certificate_manager.rs:1:9
|   |
| 1 | #![deny(rust_2018_idioms, warnings)]
|   |         ^^^^^^^^^^^^^^^^
|
| error: aborting due to 4 previous errors
|
| error: could not compile `edgelet-http`.
|
```
**Build System:**
 - OS: Ubuntu 18.04 running in container (using crops)

**Target System:**
 - OS: Yocto dunfell linux for Arm based NXP imx7

mschwan-phytec, 2020-12-09:
The version of rust being shipped with meta-rust must match the version that iotedge is built against. You can check [builds/checkin.yaml](https://github.com/Azure/meta-iotedge/blob/dunfell/builds/checkin.yaml) for the tested meta-rust revision. Make sure to checkout the correct branch before looking at the meta-rust revision in this file. For dunfell this is currently `e4d25b98083bcecb94df6ee189a165d63ede7f3d`.

rschwarz-sotec, 2020-12-10:
Thanks a lot, that solved my problem!

closed 2020-12-10 by rschwarz-sotec