Azure/meta-iotedge #59: Yocto build fails with RUST > 1.41

ksaye, 2021-01-27:
**Describe the bug**
When building with the master branch of meta-rust, the build fails with "ERROR: iotedge-cli-1.0.9.4-r0 do_compile: Execution of '/mnt/rpi/rpi-build/tmp/work/corei7-64-poky-linux/iotedge-cli/1.0.9.4-r0/temp/run.do_compile.28917' failed with exit code 101:"

**To Reproduce**
Steps to reproduce the behavior:
1. Install Ubuntu 18.04
2. install the Ubuntu pre requisites
3. Add the following layers:

```
git clone -b dunfell https://github.com/Azure/meta-iotedge.git
git clone -b dunfell git://git.yoctoproject.org/meta-virtualization
git clone -b master git://github.com/meta-rust/meta-rust.git
git clone -b dunfell git://git.yoctoproject.org/poky
git clone -b dunfell git://git.openembedded.org/meta-openembedded
git clone -b dunfell git://git.yoctoproject.org/meta-intel
```

This works if I roll the meta-rust back to 1.41:
```
cd meta-rust
git checkout a012a1027defe28495f06ed522a7a82bdd59a610
```

**Expected behavior**
Build complete and not fail.

**Logs**
See attached log
[dunfellintel-corei7-64.log](https://github.com/Azure/meta-iotedge/files/5882755/dunfellintel-corei7-64.log)

**Build System (please complete the following information):**
 - OS: Ubuntu
 - Version 18.04

**Target System (please complete the following information):**
 - OS: intel-corei7-64
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

shr-project, 2021-02-07:
see https://github.com/Azure/meta-iotedge/pull/49

hurtadosanti, 2021-03-16:
In general, is best to use % or a specific tag from the layer. Specifying a version does not seem to be a good approach, anyway removing the bbappend is a better option if not needed.

closed 2021-07-08 by ksaye