Azure/meta-iotedge #63: are you planning to update the recipes for version 1.2 ?

GeorgeDimi, 2021-05-21:
I would like to know if you are planning to update the yocto recipes for version 1.2 ?

aziot-edge=1.2 
aziot-identity-service=1.2

willfrazer, 2021-05-24:
Much as I would like to be dissuaded, I'm beginning to think this repository is dead.

matthiasfeurer, 2021-05-25:
@myagley @micahl is this still maintained by Microsoft?

cskowronnek, 2021-06-03:
Maybe it helps to repost this question here: https://github.com/azure/iotedge
This project seems to be abandoned

micahl, 2021-06-03:
We're in the process of updating the Dunfell branch to use the latest LTS release of IoT Edge (v1.1.3).

cskowronnek, 2021-06-07:
@micahl Sounds awesome.

asergaz, 2021-06-15:
Related question in Stack Overflow: https://stackoverflow.com/questions/67615008/azure-iot-edge-yocto-project-support

micahl, 2021-07-09:
@GeorgeDimi yes, we plan to update the recipes for 1.2.  However, I don't have an exact timeline to share on when that will be done.  We're first looking at how we can streamline / automate things in our official IoT Edge release process to make it easier to keep this project up-to-date as we continue to have patch releases for our LTS (v1.1).  There are currently more manual steps than we'd like.

cskowronnek, 2021-09-23:
@micahl Is there an update on the timeline? We'd like to update our IoT devices with the latest iotedge runtime based on yocto

micahl, 2021-09-23:
@cskowronnek sorry, I don't have an update yet on 1.2.  Recently we've been needed to help out on a separate project and that temporarily limits our availability.  Is there a certain feature(s) in 1.2 that makes it the more interesting target to use instead of the LTS (v1.1)?  We recently updated Dunfell to 1.1.6 (#70).

rschwarz-sotec, 2021-09-27:
@micahl we are also waiting for a 1.2 release due to the support of hardware secure elements which is not available in versions < 1.2 as far as I see

SimonSimCity, 2021-12-10:
Is there anything the community could assist with? Is there e.g. something special why it needs much time, or is it rather that you couldn't yet allocate enough time to this project?

micahl, 2022-02-04:
I invite everyone to take a look at the proposal in #77 and leave any comments/feedback.

micahl, 2022-09-19:
For those who missed it, we're updating the Dunfell recipes to target our latest long-term servicing release (1.4).  You can see the (draft) PR here and we'd appreciate any early feedback / validation: https://github.com/Azure/meta-iotedge/pull/90. 

The dunfell-1.1 branch has been created for the existing 1.1.x recipes if anyone still needs access to those.

micahl, 2022-12-03:
Closing this as the dunfell branch now has recipes for the latest 1.4 LTS.

closed 2022-12-03 by micahl