Azure/meta-iotedge #64: Build is failing for Yocto Zeus

sathyendranv, 2021-05-28:
Hi, 

Am trying to include iotedge in my yocto os. 
Following steps I followed to get started. 

git clone -b zeus https://github.com/openembedded/openembedded-core.git
git clone -b zeus https://github.com/Azure/meta-iotedge.git
git clone https://github.com/meta-rust/meta-rust.git
git clone -b zeus https://git.yoctoproject.org/git/meta-virtualization

Added these to BBLayer conf

${BSPDIR}/sources/meta-rust \
  ${BSPDIR}/sources/poky/meta \
  ${BSPDIR}/sources/poky/meta-poky \
  ${BSPDIR}/sources/poky/meta-yocto-bsp \
  ${BSPDIR}/sources/meta-freescale \
  ${BSPDIR}/sources/meta-freescale-3rdparty \
  ${BSPDIR}/sources/meta-freescale-distro \
  ${BSPDIR}/sources/meta-openembedded/meta-oe \
  ${BSPDIR}/sources/meta-openembedded/meta-python \
  ${BSPDIR}/sources/meta-openembedded/meta-multimedia \
  ${BSPDIR}/sources/meta-openembedded/meta-gnome \
  ${BSPDIR}/sources/meta-openembedded/meta-networking \
  ${BSPDIR}/sources/meta-ts \
  ${BSPDIR}/sources/meta-qt5 \
  ${BSPDIR}/sources/meta-custom \
  ${BSPDIR}/sources/meta-openembedded/meta-filesystems \
  ${BSPDIR}/sources/meta-virtualization \
   ${BSPDIR}/sources/meta-iotedge \

Added the below in local.conf
CORE_IMAGE_EXTRA_INSTALL += "libunwind icu libcurl openssl libgssapi-krb5"
IMAGE_INSTALL_append += "  docker connman connman-client "  
IMAGE_INSTALL_append += " iotedge-cli iotedge-daemon libiothsm-std"
DISTRO_FEATURES_append += " virtualization"

**_Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "tsimx6"
DISTRO               = "poky"
DISTRO_VERSION       = "3.0.4"
TUNE_FEATURES        = "arm vfp cortexa9 neon thumb callconvention-hard"
TARGET_FPU           = "hard"
meta-rust            = "master:494d879a1d337cda5b3a19406cabe46fcd3aee76"
meta
meta-poky
meta-yocto-bsp       = "HEAD:d88d62c20d7d8da85f02edb170dae0280624ad7e"
meta-freescale       = "HEAD:14f1a630a47375432f93c556927b879b51d84c4e"
meta-freescale-3rdparty = "HEAD:81d336c6c4a78916b06eb4d2190e39bc23f22336"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-oe
meta-python
meta-multimedia
meta-gnome
meta-networking      = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-ts              = "HEAD:a40463beb840dd3788cafe2d1527df451a4cc4f7"
meta-qt5             = "HEAD:91e6e3b1f5aebd17f895caac3dee32a14630da99"
meta-custom          = "<unknown>:<unknown>"
meta-filesystems     = "HEAD:2b5dd1eb81cd08bc065bc76125f2856e9383e98b"
meta-virtualization  = "zeus:7e5219669ff6f8e9c8c33ffd230e95a6b2b025f4"
meta-iotedge         = "zeus:1e926424395abcb683b045452dfd7fde4b2679f4"_**

when i run bitbake core-image-minimal,  am getting the below error

_**Parsing recipes: 100% |#############################################################################################################| Time: 0:00:01
Parsing of 2793 .bb files complete (2790 cached, 3 parsed). 3989 targets, 260 skipped, 0 masked, 0 errors.
ERROR: No recipes available for:
  /data/ts-bsp/sources/meta-iotedge/recipes-devtools/cargo/cargo_1.40.0.bbappend**_

I tried removing cargo_1.40.0.bbappend from meta-iotedge and tried building again, that gives an another error 

**_| NOTE: rustc =
| NOTE: cargo build -v --target arm-poky-linux-gnueabi --release --manifest-path=/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/iotedge/iotedge/Cargo.toml
| error: manifest path `/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/iotedge/iotedge/Cargo.toml` does not exist
| WARNING: exit code 101 from a shell command.
| ERROR: Execution of '/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/temp/run.do_compile.15765' failed with exit code 101:
| error: manifest path `/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-cli/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/iotedge/iotedge/Cargo.toml` does not exist
| WARNING: exit code 101 from a shell command.
|
ERROR: Task (/data/ts-bsp/sources/meta-iotedge/recipes-core/iotedge-cli/iotedge-cli_1.0.9.4.bb:do_compile) failed with exit code '1'
ERROR: iotedge-daemon-1.0.9.4-r0 do_compile: Execution of '/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/temp/run.do_compile.15783' failed with exit code 101:
error: manifest path `/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/iotedged/iotedged/Cargo.toml` does not exist
WARNING: exit code 101 from a shell command.

ERROR: Logfile of failure stored in: /data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/temp/log.do_compile.15783
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target arm-poky-linux-gnueabi --release --manifest-path=/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/iotedged/iotedged/Cargo.toml
| error: manifest path `/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/iotedged/iotedged/Cargo.toml` does not exist
| WARNING: exit code 101 from a shell command.
| ERROR: Execution of '/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/temp/run.do_compile.15783' failed with exit code 101:
| error: manifest path `/data/ts-bsp/build/tmp/work/cortexa9t2hf-neon-poky-linux-gnueabi/iotedge-daemon/1.0.9.4-r0/iotedge-1.0.9.4/edgelet/iotedged/iotedged/Cargo.toml` does not exist
| WARNING: exit code 101 from a shell command.
|
ERROR: Task (/data/ts-bsp/sources/meta-iotedge/recipes-core/iotedge-daemon/iotedge-daemon_1.0.9.4.bb:do_compile) failed with exit code '1'_**

Please let me know what am doing it wrong

darobs, 2021-06-03:
Hi @sathyendranv,

I just merged a fix to build 1.1.1 in Zeus (#61)- can you try it again with this change?

sathyendranv, 2021-06-04:
Hi @darobs

I pulled the latest and run a fresh build. Still seeing the same issue. 

This is git log
[image]

ksaye, 2021-07-07:
@darobs and @sathyendranv , I think the issue is an incompatible version of RUST.  If I revert the version back to 1.49, it will start to compile:

```
git clone -b master git://github.com/meta-rust/meta-rust.git
# reverting to a known good version 
cd meta-rust
git checkout 0ee328aa26e6615d0546c05e987d2f9b19a30e3d
```

It will start to compile, but fails eventually with:
```
| install: cannot stat '/mnt/data/yocto/yocto/tmp/work/corei7-64-poky-linux/iotedge-daemon/1.1.1-r0/iotedge-1.1.1/edgelet/target/x86_64-poky-linux/release/iotedged': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Execution of '/mnt/data/yocto/yocto/tmp/work/corei7-64-poky-linux/iotedge-daemon/1.1.1-r0/temp/run.do_install.34315' failed with exit code 1:
| install: cannot stat '/mnt/data/yocto/yocto/tmp/work/corei7-64-poky-linux/iotedge-daemon/1.1.1-r0/iotedge-1.1.1/edgelet/target/x86_64-poky-linux/release/iotedged': No such file or directory
| WARNING: exit code 1 from a shell command.
```

There appears to not be a directory target, so is there a pathing issue?

```
ksaye@androidubuild:/mnt/data/yocto$ ls -all /mnt/data/yocto/yocto/tmp/work/corei7-64-poky-linux/iotedge-cli/1.1.1-r0/iotedge-1.1.1/edgelet/
total 260
drwxr-xr-x 39 ksaye ksaye  4096 Jul  7 21:03 .
drwxr-xr-x 22 ksaye ksaye  4096 Jul  7 21:03 ..
drwxr-xr-x  2 ksaye ksaye  4096 Jul  7 21:03 api
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 build
-rwxr-xr-x  1 ksaye ksaye 65880 Jul  7 21:03 Cargo.lock
-rw-r--r--  1 ksaye ksaye   683 Jul  7 21:03 Cargo.toml
drwxr-xr-x  9 ksaye ksaye  4096 Jul  7 21:03 contrib
-rw-r--r--  1 ksaye ksaye  1018 Jul  7 21:03 Cross.toml
drwxr-xr-x  2 ksaye ksaye  4096 Jul  7 21:03 doc
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 docker-rs
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 dps
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 edgelet-core
drwxr-xr-x  6 ksaye ksaye  4096 Jul  7 21:03 edgelet-docker
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 edgelet-hsm
drwxr-xr-x  5 ksaye ksaye  4096 Jul  7 21:03 edgelet-http
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 edgelet-http-external-provisioning
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 edgelet-http-mgmt
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 edgelet-http-workload
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 edgelet-iothub
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 edgelet-kube
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 edgelet-test-utils
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 edgelet-utils
drwxr-xr-x  5 ksaye ksaye  4096 Jul  7 21:03 external-provisioning
-rw-r--r--  1 ksaye ksaye    55 Jul  7 21:03 .gitignore
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 hsm-rs
drwxr-xr-x  5 ksaye ksaye  4096 Jul  7 21:03 hsm-sys
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 hyper-named-pipe
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 iotedge
drwxr-xr-x  5 ksaye ksaye  4096 Jul  7 21:03 iotedged
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 iotedged-eventlog-messages
drwxr-xr-x  6 ksaye ksaye  4096 Jul  7 21:03 iotedge-proxy
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 iothubservice
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 kube-client
lrwxrwxrwx  1 ksaye ksaye    10 Jul  7 21:03 LICENSE -> ../LICENSE
-rw-r--r--  1 ksaye ksaye  8599 Jul  7 21:03 Makefile
drwxr-xr-x  5 ksaye ksaye  4096 Jul  7 21:03 management
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 mini-sntp
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 proto
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 provisioning
-rw-r--r--  1 ksaye ksaye   368 Jul  7 21:03 README.md
-rw-r--r--  1 ksaye ksaye     6 Jul  7 21:03 rust-toolchain
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 signal-future
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 support-bundle
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 systemd
lrwxrwxrwx  1 ksaye ksaye    20 Jul  7 21:03 THIRDPARTYNOTICES -> ../THIRDPARTYNOTICES
drwxr-xr-x  4 ksaye ksaye  4096 Jul  7 21:03 tokio-named-pipe
-rw-r--r--  1 ksaye ksaye     5 Jul  7 21:03 version.txt
drwxr-xr-x  3 ksaye ksaye  4096 Jul  7 21:03 win-logger
drwxr-xr-x  5 ksaye ksaye  4096 Jul  7 21:03 workload
ksaye@androidubuild:/mnt/data/yocto$ cat /mnt/data/yocto/yocto/tmp/work/corei7-64-poky-linux/iotedge-cli/1.1.1-r0/iotedge-1.1.1/edgelet/version.txt
1.1.1
```

Full log file attached: 
[20210707205018.log](https://github.com/Azure/meta-iotedge/files/6780435/20210707205018.log)

darobs, 2021-07-07:
Hello @ksaye, @sathyendranv

The 1.1 branch is tied to a specific version of Rust (1.47.0) - this is why we use that specific commit of meta-rust (METARUST_REV: '7ff669d8cedd83a2d3efb73073a63b0a7efffddc').  

Need to figure out what is going on with that pathing issue - these paths should be OK - `CARGO_TARGET_SUBDIR` should be set per project.

ksaye, 2021-07-08:
@sathyendranv , I think you are still dealing with a RUST version issue.  I saw the same thing.

When I switched (back) to RUST 1.47, it worked.  You can change it by using the following command:

```
git clone -b master git://github.com/meta-rust/meta-rust.git
# reverting to a known good version 
cd meta-rust
git checkout 7ff669d8cedd83a2d3efb73073a63b0a7efffddc
```

I just compiled and booted Yocto for ``` MACHINE              = "intel-corei7-64" ``` with success.

sathyendranv, 2021-07-08:
@darobs @ksaye  Thank you for the input. That worked :)

closed 2021-07-08 by sathyendranv

ksaye, 2021-07-08:
Yep, we need better information on this.  :)