Azure/meta-iotedge #69: version and branch questions

harryliou, 2021-08-12:
There two questions:
1. Is there a plan to update 1.1.4 version at dunfell branch?
2. Is there a plan to support yocto Gatesgarth(3.2), Hardknott(3.3) or newer ?

thank you.

darobs, 2021-09-10:
Hello @harryliou 

No plans for 1.1.4, but I had my reasons. :) 1.1.4 didn't have any functional changes from 1.1.3 so it wasn't strictly necessary.
1.1.6 was recently released, and I just put out a PR for that: #70.  

Right now, @micahl and I are talking about how we want to organize this repo, for both 1.2+ support and best Yocto version practices.

darobs, 2021-09-10:
Other customer have asked the question, "Is it OK to run 1.1.X iotedged with later versions of the system modules (edgeAgent/edgeHub)?"   For example, we released edgeAgent/Hub versions 1.1.4, but I left Yocto at 1.1.3.

The answer is generally "yes." Unless there's a specifically documented reason, the iotedged service's API has not changed and any 1.1.X version of iotedged is compatible with any 1.1.X release of edgeAgent.  The main reason to keep all versions up to date are security and bugfix updates.

micahl, 2022-12-03:
Closing.  The latest 1.4 recipes are in the Dunfell branch.  Newer releases of yocto will require we target openssl 3.0.  Similar work would be required for us to support off-the-shelf distros like Ubuntu 22.04 and RHEL 9 as they also use openssl 3.0.  We'd tackle those first before we look at something newer like Kirkstone.

For the general plan on version / branching see https://github.com/Azure/meta-iotedge/issues/77

closed 2022-12-03 by micahl