Azure/meta-iotedge #72: Build error creating an image weston for STM32MP157C-DK2

EnricoBeltramo, 2021-09-18:
**Describe the bug**
Build error using mateioedge error on weston for STM32MP157C-DK2

**To Reproduce**
Steps to reproduce the behavior:
1. create a base image for weston openstlinux-5.10-dunfell-mp1-21-03-31
2. added iotedge layer
3. run build
4. See error

**Expected behavior**
All compile complete

**Logs**
If applicable, add logs to help explain your problem.

| error: attributes are not yet allowed on `if` expressions

|  --> /usr/src/debug/iotedge-daemon/1.1.6-r0/iotedge-1.1.6/vendor/sysinfo/src/utils.rs:27:5

|  |

| 27 |   #allow(clippy::identity_conversion)

|  |   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

| 

| error: aborting due to previous error

| 

| error: could not compile `sysinfo`.

**Build System (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04

**Target System (please complete the following information):**
 - OS: weston openstlinux-5.10-dunfell-mp1-21-03-31

**Additional context**
Add any other context about the problem here.
Looking a similar issues, it seem a problem of version of Rust: https://github.com/Frederick888/git-credential-keepassxc/issues/12

asergaz, 2021-09-28:
Relates with: https://github.com/MicrosoftDocs/azure-docs/issues/81428

darobs, 2021-12-10:
Azure IoT Edge 1.1 is tied to Rust 1.47.0 - right now that means we pin the meta-rust repo at this commit:

METARUST_REV: '7ff669d8cedd83a2d3efb73073a63b0a7efffddc'

I'm looking into a way to improve that experience.  If I find a way, I will issue a PR and update this issue.

micahl, 2023-01-20:
Closing this as the recipes have been rewritten for 1.4 and the product is pinned to a more recent version of rust which (AFAIK) doesn't complain.

closed 2023-01-20 by micahl