Azure/meta-iotedge #77: Proposal: Branching strategy for meta-iotedge

micahl, 2022-02-04:
# Proposal
**Please reply with any feedback / questions before February 25th, 2022.**

Track only the most recent stable and LTS releases of yocto.
Track the last stable of IoT Edge and all active LTS releases (simultaneous max is 2).

| Active branches | Yocto | IoT Edge | Example |
| :- | :- | :- | :- |
| main | Latest _stable_ | Latest _stable_ (e.g. v1.2.x) | _main_ |
| _{yocto LTS}_ | Latest LTS | Latest _stable_ (e.g. 1.2.x) | _dunfell_ |
| _{yocto LTS}-{IoT Edge LTS}_ | Latest LTS | Latest LTS (i.e. 1.1.x) | _dunfell-1.1_ |

## Examples
The table below is how the branching would appear if: 1) the above strategy was already in use then a new yocto LTS released (Kirkstone) and then subsequently there was a new IoT Edge LTS (hypothetically a v1.3).

<table>
<thead>
<th>Today</th>
<th>new yocto LTS (i.e. Kirkstone)</th>
<th>new IoT Edge LTS (e.g. 1.3)</th>
</thead>
<tr valign="top">
<td>
main
<li>Yocto target: Honistor</li>
<li>IoT Edge recipes: v1.2</li>
<br>dunfell
<li>IoT Edge recipes: v1.2</li>
<br>dunfell-1.1
<li>IoT Edge recipes: v1.1</li>
</td>
<td>
main
<li>Yocto target: <b>Kirkstone</b></li>
<li>IoT Edge recipes: v1.2</li>
<br><strike>dunfell</strike>
<li>still available, not maintained</li>
<br><strike>dunfell-1.1</strike>
<li>still available, not maintained</li>
<br>kirkstone
<li>branched from main</li>
<li>IoT Edge recipes: v1.2</li>
<br>kirkstone-1.1
<li>branched from kirkstone</li>
<li>IoT Edge recipes: v1.1</li>
</td>
<td>
main
<li>Yocto target: Kirkstone</li>
<li>IoT Edge recipes: <b>v1.3</b></li>
<br><strike>dunfell</strike>
<li>still available, not maintained</li>
<br><strike>dunfell-1.1</strike>
<li>still available, not maintained</li>
<br>kirkstone
<li>branched from main</li>
<li>IoT Edge recipes: <b>v1.3</b></li>
<br>kirkstone-1.1
<li><b>active until 1.1 EOL (Dec '22)</b></li>
<li>branched from kirkstone</li>
<li>IoT Edge recipes: v1.1</li>
<br><b>kirkstone-1.3</b>
<li>branched from kirkstone</li>
<li>IoT Edge recipes: <b>v1.3</b></li>
</td>
</tr>

darobs, 2022-02-07:
Commenting so I get notifications on this issue. We are interested to hear public feedback.

micahl, 2022-02-10:
@sindelar-fr thanks for asking.  Yes, only applies to the LTS.  So at most there might be 3 in total (two LTS releases and the one current feature release) until the older LTS hits EOL and then just the two remain (one LTS and the latest feature release).

vish74, 2022-03-03:
Is there any update on above proposal?

micahl, 2022-03-04:
@vish74 given reactions my read on the proposal is that there aren't major concerns.  Next step would be for us to snap to that structure.  

Of course, creating a branch is easy.  But what I don't know to inform a timeline is how big the delta would be to move the main branch to the most recent Yocto release and what would be needed for Kirkstone once it releases.  

What I do know at the moment is that @darobs is juggling multiple things.  :)  So I'll defer to him on when he expects to have cycles considering his other priorities.  And, if there are easy opportunities for others to pitch in.

vish74, 2022-03-23:
Thanks @micahl 
Hi @darobs are you planning 1.2.x anytime soon? One of our project and development depends on integration with 1.2 IoTEdge. I noticed 1.2.8 of IoTEdge is also released, which captures good amount of bug fixes.

Also, thank you for all the work :) Looking forward to an update in repo.

sindelar-fr, 2022-07-07:
Hi! What's the status on this? Is there some time frame for this change to be in effect?

rschwarz-sotec, 2022-09-12:
Hi,

there is a lot of progress on https://github.com/Azure/azure-iotedge/releases, new features and new releases. Are there any plans to update to the latest lts version 1.4?

micahl, 2022-09-19:
@rschwarz-sotec and @sindelar-fr see https://github.com/Azure/meta-iotedge/pull/90.  Would very much appreciate any early feedback and having folks try out the change once we move it out of a draft PR state.

micahl, 2022-11-28:
The [dunfell branch](https://github.com/azure/meta-iotedge/tree/dunfell) now has recipes for the 1.4.x release.  If anyone still needs the 1.1 recipes you can find them in the [dunfell-1.1 branch](https://github.com/azure/meta-iotedge/tree/dunfell-1.1).

micahl, 2022-12-08:
Migrated the branching strategy info to the repo's top-level README.  Closing this out here.

closed 2022-12-08 by micahl