Azure/meta-iotedge #78: 'master->main' branch renaming broke the recipes

chli, 2022-02-22:
If any project out there still uses meta-iotedge please note that you will have to edit the recipes for iotedge-cli and iotedge-daemon to fetch the "main" branches for the following repositories:

git://github.com/Azure/hyperlocal-windows;protocol=https;nobranch=1;name=hyperlocal-windows;destsuffix=hyperlocal-windows \
git://github.com/Azure/mio-uds-windows.git;protocol=https;nobranch=1;name=mio-uds-windows;destsuffix=mio-uds-windows \
git://github.com/Azure/tokio-uds-windows.git;protocol=https;nobranch=1;name=tokio-uds-windows;destsuffix=tokio-uds-windows \

Must be changed to :

git://github.com/Azure/hyperlocal-windows;protocol=https;branch=main;name=hyperlocal-windows;destsuffix=hyperlocal-windows \
git://github.com/Azure/mio-uds-windows.git;protocol=https;branch=main;name=mio-uds-windows;destsuffix=mio-uds-windows \
git://github.com/Azure/tokio-uds-windows.git;protocol=https;branch=main;name=tokio-uds-windows;destsuffix=tokio-uds-windows \

Otherwise the automatic AUTOINC resolve fail since the 'master' branch got deleted (silently) on the 3 repositories above.

donofan0, 2022-02-22:
I made a Pull Request to fix this #79

emilm, 2022-02-25:
Not good practice to just silently rename the main branch like that.
Seems like I can't just override it in my own layer either, bitbake insists on parsing and process meta-iotedge files.

darobs, 2022-03-01:
This was an unanticipated side-effect of changing the source repo's main branch. @donofan0's PR looks good, but the build failed due the agent running out of disk space. I cleaned up some disk space and re-ran the build. On success, I'll merge the PR.

I apologize for the problem.

darobs, 2022-03-01:
PR #79 is merged, please try it now and let us know.

emilm, 2022-03-02:
It builds!

darobs, 2022-03-15:
Closing duplicates of this issue.  There is still some discussion on #81.

closed 2022-03-15 by darobs