Azure/meta-iotedge #80: Both iotedge-daemon_1.1.8.bb and iotedge-cli_1.1.8.bb changes are blocking the builds

Deepthi-263, 2022-02-23:
Changes from master to main causing an issue in my work area hence ending up with an errors.
Modify meta-iotedge/recipes-core/iotedge-daemon/iotedge-daemon_1.1.8.bb & meta-iotedge/recipes-core/iotedge-cli/iotedge-cli_1.1.8.bb as below:
SRC_URI += " \
    git://github.com/Azure/hyperlocal-windows;protocol=https;branch=main;name=hyperlocal-windows;destsuffix=hyperlocal-windows \
    git://github.com/Azure/mio-uds-windows.git;protocol=https;branch=main;name=mio-uds-windows;destsuffix=mio-uds-windows \
    git://github.com/Azure/tokio-uds-windows.git;protocol=https;branch=main;name=tokio-uds-windows;destsuffix=tokio-uds-windows \
"

Please help to merge the changes soon if its working as expected.

darobs, 2022-03-01:
Merged PR #79 to address this issue, please try the latest in dunfell branch.

darobs, 2022-03-15:
Closing this as fixed, but there is some discussion on #81, so linking that here.