Azure/meta-iotedge #84: Query : Do you have any plans to get compatible with krikstone

Vishwasrao1, 2022-06-15:
As yocto is now having krikstone as LTS branch, do you plan to be compatible with the same?

aurelihein, 2022-06-20:
I am also very interested in the upgrade to support last [Yocto Kirkstone 4.0](https://docs.yoctoproject.org/dev/migration-guides/migration-4.0.html) !
It might go with [iot-identity-service compatible with Openssl 3.0](https://github.com/Azure/iot-identity-service/issues/413)

Vishwasrao1, 2022-06-23:
Looks it will need lot of changes to the core, I believe this will take time. However, it will be interesting to see the developments  @aurelihein . I keep this issue open for others to give input.

rschwarz-sotec, 2022-12-06:
Hi,

anyone already working with kirkstone? Or any plans of "official" support?

rschwarz-sotec, 2023-01-12:
Hi,

I tried to do the kirkstone update by myself, but I run into very early build issues. As I am no expert in Rust, I want to ask first if my plan is doable or if there any other ideas.

What I tried
- checking out meta-iotedge on dunfell release
- add all dependencies as documented on their kirkstone release
```$ git submodule status
 013b5eeb580bc13b156aaf77d0c1175f75b89671 meta-clang (remotes/origin/kirkstone)
+37015b5f43cbcd49aef673d9697df14ed3bb96ae meta-iotedge (heads/dunfell)
+458ab61befa4fb5c192cd970ed0db2225f0c5030 meta-rust (remotes/origin/fix-ci-6-g458ab61)
+c79262a30bd385f5dbb009ef8704a1a01644528e meta-security (heads/kirkstone)
+927537108bcf2b98859512ce3eae59a73439994d meta-virtualization (heads/kirkstone)
 2e79b199114b25d81bfaa029ccfb17676946d20d poky (yocto-4.0.5)
```
- add kirkstone to LAYERSERIES_COMPAT in meta-rust/conf/layer.conf
- add kirkstone to LAYERSERIES_COMPAT in meta-iotedge/conf/layer.conf
- run convertion scripts from poky/scripts/contrib against recipes in meta-iotedge
- changes in meta-iotedge:
<details><summary>CLICK ME</summary>
```
diff --git a/conf/layer.conf b/conf/layer.conf
index 02b0396..aa31239 100644
--- a/conf/layer.conf
+++ b/conf/layer.conf
@@ -9,6 +9,6 @@ BBFILE_COLLECTIONS += "meta-iotedge"
 BBFILE_PATTERN_meta-iotedge = "^${LAYERDIR}/"
 BBFILE_PRIORITY_meta-iotedge = "10"
 
-LAYERDEPENDS_meta-iotedge = "core rust-layer virtualization-layer"
-LAYERSERIES_COMPAT_meta-iotedge = "dunfell"
+LAYERDEPENDS_meta-iotedge = "core virtualization-layer"
+LAYERSERIES_COMPAT_meta-iotedge = "dunfell kirkstone"
 
diff --git a/conf/local.conf.sample b/conf/local.conf.sample
index c489f90..302be76 100644
--- a/conf/local.conf.sample
+++ b/conf/local.conf.sample
@@ -234,8 +234,8 @@ SSTATE_MIRRORS = "\
 # seen. The two lines below enable the SDL backend too. By default libsdl-native will
 # be built, if you want to use your host's libSDL instead of the minimal libsdl built
 # by libsdl-native then uncomment the ASSUME_PROVIDED line below.
-PACKAGECONFIG_append_pn-qemu-native = " sdl"
-PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
+PACKAGECONFIG:append:pn-qemu-native = " sdl"
+PACKAGECONFIG:append:pn-nativesdk-qemu = " sdl"
 #ASSUME_PROVIDED += "libsdl-native"
 
 # CONF_VERSION is increased each time build/conf/ changes incompatibly and is used to
@@ -243,6 +243,6 @@ PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
 # this doesn't mean anything to you.
 CONF_VERSION = "1"
 
-DISTRO_FEATURES_append = " systemd virtualization"
+DISTRO_FEATURES:append = " systemd virtualization"
 VIRTUAL-RUNTIME_init_manager = "systemd"
 VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
diff --git a/recipes-core/aziot-edged/aziot-edged.inc b/recipes-core/aziot-edged/aziot-edged.inc
index 26d7786..7844acc 100644
--- a/recipes-core/aziot-edged/aziot-edged.inc
+++ b/recipes-core/aziot-edged/aziot-edged.inc
@@ -1,10 +1,10 @@
 DEPENDS += "openssl virtual/docker iotedge aziotd systemd"
-RDEPENDS_${PN} += "docker iotedge aziotd aziot-keys systemd"
+RDEPENDS:${PN} += "docker iotedge aziotd aziot-keys systemd"
 
 inherit systemd
 
-SYSTEMD_AUTO_ENABLE_${PN} = "enable"
-SYSTEMD_SERVICE_${PN} = "aziot-edged.service"
[... 366 lines cut ...]
| 
| Caused by:
|   namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
| WARNING: /workdir/odroid-c4/build/tmp/work/cortexa55-poky-linux/aziotctl/1.4.1.AUTOINC+f349b64d3b-r0/temp/run.do_compile.128:176 exit 101 from '"cargo" build -v --target aarch64-poky-linux --release --manifest-path=/workdir/odroid-c4/build/tmp/work/cortexa55-poky-linux/aziotctl/1.4.1.AUTOINC+f349b64d3b-r0/git/aziotctl/Cargo.toml "$@"'
| WARNING: Backtrace (BB generated script):
|       #1: oe_cargo_build, /workdir/odroid-c4/build/tmp/work/cortexa55-poky-linux/aziotctl/1.4.1.AUTOINC+f349b64d3b-r0/temp/run.do_compile.128, line 176
|       #2: cargo_do_compile, /workdir/odroid-c4/build/tmp/work/cortexa55-poky-linux/aziotctl/1.4.1.AUTOINC+f349b64d3b-r0/temp/run.do_compile.128, line 166
|       #3: do_compile, /workdir/odroid-c4/build/tmp/work/cortexa55-poky-linux/aziotctl/1.4.1.AUTOINC+f349b64d3b-r0/temp/run.do_compile.128, line 160
|       #4: main, /workdir/odroid-c4/build/tmp/work/cortexa55-poky-linux/aziotctl/1.4.1.AUTOINC+f349b64d3b-r0/temp/run.do_compile.128, line 208
ERROR: Task (/workdir/meta-iotedge/recipes-core/aziotctl/aziotctl_1.4.1.bb:do_compile) failed with exit code '1'
```

</p>
</details>
```

My questions:
- Is this the right way of doing the migration?
- I read something about rust support in yocto [link](https://github.com/meta-rust/meta-rust/issues/380). Is meta-rust still needed on kirkstone?
- Is anybody else working on this? Are there any plans when kirkstone support will be releases?

I would be happy if I can support development for kirkstone! It would be great to get some advice how to get started.

Cheers
Ruben

MZDN, 2023-01-15:
Is there any update on this ?

rschwarz-sotec, 2023-01-16:
Not from my side, I'm still waiting for any response.

micahl, 2023-01-18:
@timesys-nathan contributed a PR (https://github.com/Azure/meta-iotedge/pull/107) that we will review (thanks!).  Related to Kirkstone (and other distros like the latest Ubuntu or RHEL) is working with openssl 3.0 and we're investigating why one of our e2e tests fails.

theobarkerh, 2023-02-15:
We definitely need Kirkstone support ASAP. 
Longer term: does MS plan to support Yocto as a Tier 1 environment? Seems that it would be wise considering how many IoT devices use it.

micahl, 2023-03-03:
Closing as now complete (see https://github.com/Azure/meta-iotedge/commit/63785dfcf446a3f48ca9a7a26ca121109746b64e).  Thanks @timesys-nathan for the contribution and @theobarkerh and @rschwarz-sotec for help in validating!

closed 2023-03-03 by micahl