Azure/meta-iotedge #104: aziot-edged fails to start in Yocto Built OS with Azure IoT Edge Runtime

acrophobia007, 2022-12-19:
**Describe the bug**
After completing a yocto build containing meta-iotedge layer (dunfell) , it looks iotedge 1.4.4 was succesfully build into the OS.  Then I was trying to connect the IoT Edge device to the IoT hub using connection string. However, after configuring the Azure IoT Edge, I can't get aziot-edged running. The log files indicated aziot-edged always failed to load settings (No such file or directory (os error 2)), but I found 00-super.toml file in /etc/aziot/edged/config.d/ folder. Please see the following log files for more information. I'm not sure if I configured the IoT Edge in a correct way, so could you please help take look at the following logs and see what may go wrong? 

**To Reproduce**
 1. Configured the Azure IoT Edge using connection string:
     1) sudo iotedge config mp --connection-string 'My primary connection string'.  It looks IoT Edge was configured successfully and I 
         got the following messages:
         Azure IoT Edge has been configured successfully!
         The configuration has been written to /etc/aziot/config.toml
         To apply the new configuration to services, run:
    sudo iotedge config apply -c '/etc/aziot/config.toml'
     2) Ran  sudo iotedge config apply -c '/etc/aziot/config.toml'
         I got the following messages: 
          Note: Symmetric key will be written to /var/secrets/aziot/keyd/device-id
          Azure IoT Edge has been configured successfully!
          
          Restarting service for configuration to take effect...
          Stopping aziot-edged.service...Stopped!
          Stopping aziot-identityd.service...Stopped!
          Stopping aziot-keyd.service...Stopped!
          Stopping aziot-certd.service...Stopped!
          Stopping aziot-tpmd.service...Stopped!
          Starting aziot-edged.mgmt.socket...Started!
          Starting aziot-edged.workload.socket...Started!
          Starting aziot-identityd.socket...Started!
          Starting aziot-keyd.socket...Started!
          Starting aziot-certd.socket...Started!
          Starting aziot-tpmd.socket...Started!
          Starting aziot-edged.service...Started!
          Done.
However, when I ran "sudo iotedge list", I got the following error messages: 
        A module runtime error occurred
                       caused by: A module runtime error occurred
                       caused by: connection error: Connection reset by peer (os error 104)
                       caused by: Connection reset by peer (os error 104)
Then I ran "sudo iotedge system status", and got the following error messages: 
System services:
    aziot-edged             Down - failed
    aziot-identityd         Ready
    aziot-keyd               Ready
    aziot-certd              Ready
    aziot-tpmd              Ready

aziot-edged is in a bad state because:
aziot-edged.service: Down - failed : Printing the last 10 log lines.
-- Logs begin at Mon 2022-12-19 19:28:09 UTC, end at Mon 2022-12-19 20:31:33 UTC. --
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: aziot-edged.service: Failed with result 'exit-code'.
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: Started Azure IoT Edge daemon.
Dec 19 20:24:59 colibri-imx8x-06995462 aziot-edged[2911]: 2022-12-19T20:24:59Z [INFO] - Starting Azure IoT Edge Daemon
Dec 19 20:24:59 colibri-imx8x-06995462 aziot-edged[2911]: 2022-12-19T20:24:59Z [INFO] - Version - 1.4.4
Dec 19 20:24:59 colibri-imx8x-06995462 aziot-edged[2911]: 2022-12-19T20:24:59Z [ERR!] - Failed to load settings: No such file or directory (os error 2)
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: aziot-edged.service: Main process exited, code=exited, status=153/n/a
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: aziot-edged.service: Failed with result 'exit-code'.
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: aziot-edged.service: Start request repeated too quickly.
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: aziot-edged.service: Failed with result 'exit-code'.
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: Failed to start Azure IoT Edge daemon.

aziot-edged.mgmt.socket: Down - failed : Printing the last 10 log lines.
-- Logs begin at Mon 2022-12-19 19:28:09 UTC, end at Mon 2022-12-19 20:31:33 UTC. --
Dec 19 19:56:24 colibri-imx8x-06995462 systemd[1]: Starting Azure IoT Edge daemon management socket.
Dec 19 19:56:24 colibri-imx8x-06995462 systemd[1]: Listening on Azure IoT Edge daemon management socket.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: aziot-edged.mgmt.socket: Succeeded.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: Closed Azure IoT Edge daemon management socket.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: Starting Azure IoT Edge daemon management socket.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: Listening on Azure IoT Edge daemon management socket.
Dec 19 20:05:47 colibri-imx8x-06995462 systemd[1]: aziot-edged.mgmt.socket: Failed with result 'service-start-limit-hit'.
Dec 19 20:18:35 colibri-imx8x-06995462 systemd[1]: Starting Azure IoT Edge daemon management socket.
Dec 19 20:18:35 colibri-imx8x-06995462 systemd[1]: Listening on Azure IoT Edge daemon management socket.
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: aziot-edged.mgmt.socket: Failed with result 'service-start-limit-hit'.

aziot-edged.workload.socket: Down - failed : Printing the last 10 log lines.
-- Logs begin at Mon 2022-12-19 19:28:09 UTC, end at Mon 2022-12-19 20:31:33 UTC. --
Dec 19 19:56:24 colibri-imx8x-06995462 systemd[1]: Starting Azure IoT Edge daemon workload socket.
Dec 19 19:56:24 colibri-imx8x-06995462 systemd[1]: Listening on Azure IoT Edge daemon workload socket.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: aziot-edged.workload.socket: Succeeded.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: Closed Azure IoT Edge daemon workload socket.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: Starting Azure IoT Edge daemon workload socket.
Dec 19 20:04:43 colibri-imx8x-06995462 systemd[1]: Listening on Azure IoT Edge daemon workload socket.
Dec 19 20:05:47 colibri-imx8x-06995462 systemd[1]: aziot-edged.workload.socket: Failed with result 'service-start-limit-hit'.
Dec 19 20:18:35 colibri-imx8x-06995462 systemd[1]: Starting Azure IoT Edge daemon workload socket.
Dec 19 20:18:35 colibri-imx8x-06995462 systemd[1]: Listening on Azure IoT Edge daemon workload socket.
Dec 19 20:24:59 colibri-imx8x-06995462 systemd[1]: aziot-edged.workload.socket: Failed with result 'service-start-limit-hit'.

Use 'iotedge system logs' to check for non-fatal errors.
Use 'iotedge check' to diagnose connectivity and configuration issues.

After that, I used 'iotedge check" to diagnose connectivity and configuration issues. I found the following 2 configuration errors:
1. × aziot-edged configuration is well-formed - Error
    Error running check: The IoT Edge daemon's configuration file is not well-formed.
2. × read all preloaded certificates from the Certificates Service - Error
    could not load cert with ID "aziot-edged-trust-bundle"
    Caused by:
        parameter "id" has an invalid value
        caused by: not found

**Expected behavior**

After running "sudo iotedge system status", I should get the following results: 
System services:
    aziot-edged             Ready
    aziot-identityd         Ready
    aziot-keyd               Ready
    aziot-certd              Ready
    aziot-tpmd              Ready
And after running "sudo iotedge list", I should see the EdgeAgent and EdgeHub running. 

**Logs**
If applicable, add logs to help explain your problem.

**Build System (please complete the following information):**
 - OS:  Ubuntu 
 - Version: 22.04

**Target System (please complete the following information):**
 - OS: Dunfell 
 - Version: TorizonCore 5.7.1

**Additional context**
I ran "aziot-edged" in the command line, and it printed the following messages: 
<6>2022-12-19T20:41:52Z [INFO] - Starting Azure IoT Edge Daemon
<6>2022-12-19T20:41:52Z [INFO] - Version - 1.4.4
<3>2022-12-19T20:41:52Z [ERR!] - Failed to load settings: could not read config from /etc/aziot/edged/config.d/00-super.toml
I changed the file permissions, but it still doesn't work...

micahl, 2023-01-04:
Hmm... might be related to #105.  The `iotedge config apply` creates a 00-super.toml file for the respective services and assumes they're running under an expected userid (e.g., aziotid, aziotcs, azioks, aziottpm).  @ggjjj 

What's the output of the following: 
* `id iotedge`
* `sudo ls -lahR /etc/aziot` 
* `sudo ls -la /var/lib/aziot`

acrophobia007, 2023-01-05:
> Hmm... might be related to #105. The `iotedge config apply` creates a 00-super.toml file for the respective services and assumes they're running under an expected userid (e.g., aziotid, aziotcs, [...]

Hi @micahl. Thanks for talking time to look into my issue. Actually I arbitrarily assigned the userids for new users in my Yocto build environment. I was trying to build the iotedge runtime into TorizonCore with meta-iotedge and its dependency layers added. But TorizonCore doesn't automatically assign the userids for the new users so I had to create the following userids on my own in order to complete the Yocto build. 

iotedge:x:1001:1001::/bin:/bin/sh
aziotcs:x:1002:1002::/bin:/bin/sh
aziotks:x:1003:1003::/bin:/bin/sh
aziotid:x:1004:1004::/bin:/bin/sh
aziottpm:x:1005:1005::/bin:/bin/sh
tss:x:1006:1006::/bin:/bin/sh
edgeagentuser:x:13622:65534::/bin:/bin/sh
edgehubuser:x:13623:65534::/bin:/bin/sh

But honestly I'm not sure if these userids were correctly defined. Maybe this caused the aziot-edged configuration issues in the built OS? Also, we found /var/lib/aziot/edged doesn't exist in the built OS. 

Here's the output for the commands:
1. id iotedge -> uid=1001(iotedge) gid=1001(iotedge) groups=1001(iotedge),990(docker),988(systemd                           -journal),1002(aziotcs),1003(aziotks),1004(aziotid),1005(aziottpm)
2. sudo ls -lahR /etc/aziot ->
/etc/aziot:
total 44K
drwxr-xr-x  7 root root 4.0K Jan  4 16:40 .
drwxr-xr-x 51 root root 4.0K Jan  4 16:35 ..
drwxr-xr-x  3 root root 4.0K Dec 16 23:47 certd
-rw-------  1 root root  854 Jan  4 16:40 config.toml
-rw-r--r--  1 root root 9.7K Dec 16 23:47 config.toml.template
drwxr-xr-x  3 root root 4.0K Dec 16 23:47 edged
drwxr-xr-x  3 root root 4.0K Dec 16 23:47 identityd
drwxr-xr-x  3 root root 4.0K Dec 16 23:47 keyd
drwxr-xr-x  3 root root 4.0K Dec 16 23:47 tpmd

/etc/aziot/certd:
total 16K
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 .
drwxr-xr-x 7 root    root    4.0K Jan  4 16:40 ..
drwx------ 2 aziotcs aziotcs 4.0K Jan  4 16:40 config.d
-rw-r--r-- 1 root    root      78 Dec 16 23:47 config.toml.default

/etc/aziot/certd/config.d:
total 12K
drwx------ 2 aziotcs aziotcs 4.0K Jan  4 16:40 .
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 ..
-rw------- 1 aziotcs aziotcs  489 Jan  4 16:40 00-super.toml

/etc/aziot/edged:
total 12K
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 .
drwxr-xr-x 7 root    root    4.0K Jan  4 16:40 ..
drwx------ 2 iotedge iotedge 4.0K Jan  4 16:40 config.d

/etc/aziot/edged/config.d:
total 12K
drwx------ 2 iotedge iotedge 4.0K Jan  4 16:40 .
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 ..
-rw------- 1 iotedge iotedge  960 Jan  4 16:40 00-super.toml

/etc/aziot/identityd:
total 16K
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 .
drwxr-xr-x 7 root    root    4.0K Jan  4 16:40 ..
drwx------ 2 aziotid aziotid 4.0K Jan  4 16:40 config.d
-rw-r--r-- 1 root    root    1.7K Dec 16 23:47 config.toml.default

/etc/aziot/identityd/config.d:
total 12K
drwx------ 2 aziotid aziotid 4.0K Jan  4 16:40 .
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 ..
-rw------- 1 aziotid aziotid  447 Jan  4 16:40 00-super.toml

/etc/aziot/keyd:
total 16K
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 .
drwxr-xr-x 7 root    root    4.0K Jan  4 16:40 ..
drwx------ 2 aziotks aziotks 4.0K Jan  4 16:40 config.d
-rw-r--r-- 1 root    root     123 Dec 16 23:47 config.toml.default

/etc/aziot/keyd/config.d:
total 12K
drwx------ 2 aziotks aziotks 4.0K Jan  4 16:40 .
drwxr-xr-x 3 root    root    4.0K Dec 16 23:47 ..
-rw------- 1 aziotks aziotks  519 Jan  4 16:40 00-super.toml

/etc/aziot/tpmd:
total 16K
drwxr-xr-x 3 root     root     4.0K Dec 16 23:47 .
drwxr-xr-x 7 root     root     4.0K Jan  4 16:40 ..
drwx------ 2 aziottpm aziottpm 4.0K Jan  4 16:40 config.d
-rw-r--r-- 1 root     root       86 Dec 16 23:47 config.toml.default

/etc/aziot/tpmd/config.d:
total 12K
drwx------ 2 aziottpm aziottpm 4.0K Jan  4 16:40 .
drwxr-xr-x 3 root     root     4.0K Dec 16 23:47 ..
-rw------- 1 aziottpm aziottpm  138 Jan  4 16:40 00-super.toml

3. sudo ls -la /var/lib/aziot -> 
total 20
drwxr-xr-x  5 root root 4096 Jan  4 16:41 .
drwxr-xr-x 13 root root 4096 Jan  4 16:41 ..
drwxr-xr-x  3 root root 4096 Jan  4 16:41 certd
drwxr-xr-x  3 root root 4096 Jan  4 16:41 identityd
drwxr-xr-x  3 root root 4096 Jan  4 16:41 keyd

micahl, 2023-01-07:
> Also, we found /var/lib/aziot/edged doesn't exist in the built OS

Hmm... I would have expected that to be [created by the recipe](https://github.com/Azure/meta-iotedge/blob/083c7baa2aa3fd79777f464d2afaf5bedf750fc8/recipes-core/aziot-edged/aziot-edged.inc#L33).  Which recipes did you include?

xiao-zhu-007, 2023-01-10:
> > Also, we found /var/lib/aziot/edged doesn't exist in the built OS
> 
> Hmm... I would have expected that to be [created by the recipe](https://github.com/Azure/meta-iotedge/blob/083c7baa2aa3fd79777f464d2afaf5bedf750fc8/recipes-core/aziot-edged/aziot-edged.inc#L33). Which recipes did you include?

I included aziot-edged, aziotctl, and tpm2-tss in /build/conf/local.conf file as custom recipes. Based on my understanding, "aziot-edged" and "aziotctl" are the two core recipes for the yocto build. Their dependencies will be automatically called and then added. Do I need to include any custom recipes for Yocto build?

ggjjj, 2023-01-11:
Hi @xiao-zhu-007 You need to add 'aziot-edged' and 'aziotd' recipes.  'aziotctl' is a dependency for 'aziotd'

xiao-zhu-007, 2023-01-11:
> Hi @xiao-zhu-007 You need to add 'aziot-edged' and 'iotedge' recipes. 'aziotctl' is a dependency for 'iotedge'
> 
> you can see our build script here
> 
> https://github.com/Azure/meta-iotedge/blob/083c7baa2aa3fd79777f464d2afaf5bedf750fc8/scripts/build.sh#L7-L10

Hi @ggjjj Thanks for your reply.  I added "iotedge" recipe and ran another Yocto build (with iotedge, aziot-edged and aziotctl recipes included)... However, after loading the image to the dev board, I got the exact the same issues... /var/lib/aziot/edged is still missing in the OS... Something  goofy is going on. I doublechecked all the dependency layers and they all look OK...

xiao-zhu-007, 2023-01-11:
I'm going to clean up the all the tmp and cached files, include all the core recipes and run another Yocto build from scratch. Let's see how it goes this time...

xiao-zhu-007, 2023-01-12:
> Hi @ggjjj Thanks for your reply. I added "iotedge" recipe and ran another Yocto build (with iotedge, aziot-edged and aziotctl recipes included)... However, after loading the image to the dev board, [...]

@ggjjj I set up a new build directory and included the following recipes for the Yocto build. 

iotedge, aziot-edged, aziotctl, aziotd, aziot-keys, systemd

However, I still got the same issues and same errors showed up when I ran "sudo iotedge check" . After configuring the IoT Edge device, /var/lib/aziot/edged still wasn't created...

ggjjj, 2023-01-17:
having no access to your platform configuration, I don't have an answer to resolve this. I have added a bug fix in the latest PR. may be you can try that recipe.

ggjjj, 2023-01-17:
@maskimthedog has tried out the current recipe and it does work. Is there any reason why would the files not show up? thoughts @maskimthedog

maskimthedog, 2023-01-17:
> @maskimthedog has tried out the current recipe and it does work. Is there any reason why would the files not show up? thoughts

@xiao-zhu-007 we did find out recently that the services were not being started up under the correct users and have remedied that.  As far as the missing /var/lib directories, the directories below get installed by the recipe into /var/lib.  Mirrors what is done in the .deb install files for Ubuntu/Debian.

    # Data dir
    install -d ${D}${localstatedir}/lib
    install -d ${D}${localstatedir}/lib/aziot
    install -d -m 700 -o aziotks -g aziotks ${D}${localstatedir}/lib/aziot/keyd
    install -d -m 700 -o aziotcs -g aziotcs ${D}${localstatedir}/lib/aziot/certd
    install -d -m 700 -o aziotid -g aziotid ${D}${localstatedir}/lib/aziot/identityd
    install -d -m 700 -o aziottpm -g aziottpm ${D}${localstatedir}/lib/aziot/tpmd

xiao-zhu-007, 2023-01-17:
> Contributor

> having no access to your platform configuration, I don't have an answer to resolve this. I have added a bug fix in the latest PR. may be you can try that recipe.

Thanks for the response. I pulled the latest changes from 

> having no access to your platform configuration, I don't have an answer to resolve this. I have added a bug fix in the latest PR. may be you can try that recipe.

Thanks for the response. I pulled the latest changes from your new branch, ran another build and confirmed aziot-edged was recompiled during the build. However, after loading the new image to the dev board, I still got the same issues. I don't know if there's any compatible issues between iotedge and the platform I was using. I'll contact the platform support and see if I could find anything that might go wrong on their end.

xiao-zhu-007, 2023-01-17:
> @xiao-zhu-007 we did find out recently that the services were not being started up under the correct users and have remedied that. As far as the missing /var/lib directories, the directories below [...]

Thanks for chiming in and looking into this issue for me! In the latest built OS, I have 3 directories available in /var/lib/aziot/:
1. /var/lib/aziot/certd
2. /var/lib/aziot/identityd
3. /var/lib/aziot/keyd
And /var/lib/aziot/edged is the only directory missing,  which caused aziot-edged failure. But I'm not sure if this issue is related to those users you listed above. I ran another build using the latest recipe with the bugfix, but I still got the same issue... I really can't figure out what's going on here...

maskimthedog, 2023-01-18:
@xiao-zhu-007 @ggjjj I created a new build using the latest meta-iotedge a couple of days ago and below is the contents of the /var/lib/aziot directory.  We do not explicitly create the edged directory in the recipes nor do the deb files for the Ubuntu/Debian installs.  Maybe it is created by the service which is not starting on your platform IRT IoT Edge?

root@bobcat:~# ls -l /var/lib/aziot/
total 20
drwx------ 3 aziotcs  aziotcs  4096 Jan 16 01:43 certd
drwxr-xr-x 5 iotedge  iotedge  4096 Jan 16 01:42 edged
drwx------ 3 aziotid  aziotid  4096 Jan 16 01:43 identityd
drwx------ 3 aziotks  aziotks  4096 Jan 16 01:42 keyd
drwx------ 2 aziottpm aziottpm 4096 Mar  9  2018 tpmd
root@bobcat:~#

xiao-zhu-007, 2023-01-18:
> @xiao-zhu-007 @ggjjj I created a new build using the latest meta-iotedge a couple of days ago and below is the contents of the /var/lib/aziot directory. We do not explicitly create the edged [...]

Thanks for the info. It looks I have another directory missing (tpmd) in /var/lib/aziot/, which kind of explains why I got the following output when I ran "sudo iotedge system status": 

aziot-edged Down - failed
aziot-identityd Running 
aziot-keyd Running 
aziot-certd Running 
aziot-tpmd Ready

As @micahl mentioned in this comment: https://github.com/Azure/meta-iotedge/issues/104#issuecomment-1374425581, /var/lib/aziot/edged directory should be created by the recipe while the other four directories (certd, identityd, keyd and tpmd) might be created by the service I think?

maskimthedog, 2023-01-18:
I think the reverse is true.

maskimthedog, 2023-01-18:
edged seems to be created by a service.  Other four are created by the recipe which is a verbatim copy of the Ubuntu/Debian deb install files.

maskimthedog, 2023-01-18:
It seems as though aziot-edged service is failing for some reason and not able to create /var/lib/aziot/edged directory.

maskimthedog, 2023-01-18:
Maybe @ggjjj or @micahl can provide more insight as to what "settings" are not getting loaded by the service.

Dec 19 20:24:59 colibri-imx8x-06995462 aziot-edged[2911]: 2022-12-19T20:24:59Z [ERR!] - Failed to load settings: No such file or directory (os error 2)

maskimthedog, 2023-01-18:
I would suspect that maybe the toml config file is missing.  I think either @ggjjj or @micahl may have previously mentioned that.

xiao-zhu-007, 2023-01-18:
> I would suspect that maybe the toml config file is missing. I think either @ggjjj or @micahl may have previously mentioned that.

@maskimthedog The config.toml has been in place since I set up the connection string and applied the configuration. Here's what it looks like in /etc/aziot directory: 

torizon@colibri-imx8x-06995462:/etc/aziot$ ls -l
total 36
drwxr-xr-x 3 root root 4096 Jan 17 22:51 certd
-rw------- 1 root root  854 Jan 12 22:56 config.toml
-rw-r--r-- 1 root root 9932 Jan 17 22:51 config.toml.template
drwxr-xr-x 3 root root 4096 Jan 17 22:51 edged
drwxr-xr-x 3 root root 4096 Jan 17 22:51 identityd
drwxr-xr-x 3 root root 4096 Jan 17 22:51 keyd
drwxr-xr-x 3 root root 4096 Jan 17 22:51 tpmd

maskimthedog, 2023-01-18:
If you want, you can pull from my fork...it has fixes not yet accepted into main...

https://github.com/maskimthedog/meta-iotedge

xiao-zhu-007, 2023-01-18:
> When you apply the toml configuration, there are other toml config files created and placed in appropriate directories.

Good point. I didn't really pay attention to those sub folders. So I double-checked all the sub folders in /etc/aziot/ and I found a "config.toml.default" file in each sub-folder except "/etc/aziot/edged/". Please see the following information. But I really don't know why only /edged has this toml configuration file missing... 

In /etc/aziot/edged/
total 4
drwx------ 2 iotedge iotedge 4096 Jan 17 22:52 config.d

In /etc/aziot/certd/
total 8
drwx------ 2 aziotcs aziotcs 4096 Jan 17 22:52 config.d
-rw-r--r-- 1 root    root      78 Jan 17 22:51 config.toml.default

In /etc/aziot/identityd/
total 8
drwx------ 2 aziotid aziotid 4096 Jan 17 22:52 config.d
-rw-r--r-- 1 root    root    1661 Jan 17 22:51 config.toml.default

In /etc/aziot/keyd/
total 8
drwx------ 2 aziotks aziotks 4096 Jan 17 22:52 config.d
-rw-r--r-- 1 root    root     123 Jan 17 22:51 config.toml.default

In /etc/aziot/tpmd/
total 8
drwx------ 2 aziottpm aziottpm 4096 Jan 17 22:52 config.d
-rw-r--r-- 1 root     root       86 Jan 17 22:51 config.toml.default

maskimthedog, 2023-01-18:
Try building an image with my meta-ioedge fork and see what shakes out.

xiao-zhu-007, 2023-01-18:
> Try building an image with my meta-ioedge fork and see what shakes out.

That's what exactly I'm planning to do. Thanks again for sharing your meta-iotedge fork, and I'll keep you updated.

xiao-zhu-007, 2023-01-18:
@maskimthedog I used your meta-ioedge fork layer and built another image. However, after loading the new image to my dev board, I'm still getting the same issues.  There's still no config.toml.default file in /etc/aziot/edged/. But I do notice one interesting thing:
Before I configured IoT Edge using connection string and applied the config file, I noticed there was actually no "/var/lib/aziot/" folder created. And /var/lib/edged/ and its 3 sub folders were created after the edge was configured and their service started running.

xiao-zhu-007, 2023-01-18:
And that makes me wonder if this issue is actually related to the TorizonCore platform I'm using...

maskimthedog, 2023-01-18:
I can only say we replicated what was done in the Ubuntu/Debian deb files.  I am working on a NXP imx8mm based platform with Yocto Hardknott and am having no issues.

maskimthedog, 2023-01-18:
There should be a /etc/aziot/edged/config.d owned by iotedge:iotedge.

maskimthedog, 2023-01-18:
Also, the recipes do create /var/lib/aziot/edged with owner iotedge:iotedge.

   install -d -m 755 -o iotedge -g iotedge ${D}${localstatedir}/lib/aziot/edged

maskimthedog, 2023-01-18:
The /var directory is not symlinked is it?

xiao-zhu-007, 2023-01-18:
> There should be a /etc/aziot/edged/config.d owned by iotedge:iotedge.

At least this looks good on my end:

 sudo ls -l /etc/aziot/edged/
total 4
drwx------ 2 iotedge iotedge 4096 Jan 18 22:07 config.d

xiao-zhu-007, 2023-01-18:
> Also, the recipes do create /var/lib/aziot/edged with owner iotedge:iotedge.
> 
> `install -d -m 755 -o iotedge -g iotedge ${D}${localstatedir}/lib/aziot/edged`

OK, this is consistent with what @micahl commented. In my new image, /var/lib/aziot/edged was missing... So it looks something went wrong during the build and /var/lib/aziot/edged wasn't able to be created or got erased...

maskimthedog, 2023-01-18:
> OK, this is consistent with what @micahl commented. In my new image, /var/lib/aziot/edged was missing... So it looks something went wrong during the build and /var/lib/aziot/edged wasn't able to be [...]

Sure seems that way.

xiao-zhu-007, 2023-01-18:
I've been using the Toradex board with NXP imx8x and trying to build the iotedge into their TorizonCore OS. I've reached out to their tech support and I'll see if I could find anything unusual on their end. Thanks again for your help today!

xiao-zhu-007, 2023-01-23:
@maskimthedog @micahl @ggjjj I got some response from Toradex regarding the TorizonCore build environment I've been using. According to Toradex, TorizonCore uses an OSTree based filesystem to allow for easy updates. Now in a OSTree filesystem certain directories are treated specially. The /var/lib directory is one of these. In short during the build when the initial filesystem is transformed into an OSTree filesystem the /var/lib directory actually gets cleared out. I imagine what’s happening is that the iotedge recipe does properly install these files in /var/lib, but then the OSTree recipe clears this directory.  

So I think the root cause of this issue the home directly iotedge recipe created in /var/lib/aziot was later deleted by the OSTree recipe before generating the new image. So it shouldn't be an iotedge issue. In other words, the TorizonCore build environment doesn't seem to be compatible with iotedge Yocto layer and it won't work unless we modify the iotedge recipe...

maskimthedog, 2023-01-23:
@xiao-zhu-007 I feel for you.  Goodness.  I am glad you found the issue.  You may want to ask Torizon then how you go about getting IoT Edge integrated into their ecosystem.

closed 2023-01-23 by ggjjj