Azure/meta-iotedge #105: Four systemd services are not started up as correct user.

maskimthedog, 2022-12-22:
I noticed that the system .service files below which I unarchived from the current IoT Edge Debian 11 Arm64 deb files in the Yocto GitHub repos do not specify User.  They need to or x509 with PKs stored in TPM will not work.

**rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-certd.service**
[Unit]
Description=Azure IoT Certificates Service
Requires=aziot-certd.socket
After=aziot-certd.socket

[Service]
Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service"
ExecStart=/usr/libexec/aziot-identity-service/aziot-certd
KillMode=process
Restart=on-failure
RestartSec=5s
**_User=aziotcs_**

[Install]
WantedBy=multi-user.target

**rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-identityd.service**
[Unit]
Description=Azure IoT Identity Service
Requires=aziot-identityd.socket
After=aziot-identityd.socket

[Service]
Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service"
ExecStart=/usr/libexec/aziot-identity-service/aziot-identityd
KillMode=process
Restart=on-failure
RestartSec=5s
**_User=aziotid_**

[Install]
WantedBy=multi-user.target

**rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-keyd.service**
[Unit]
Description=Azure IoT Keys Service
Requires=aziot-keyd.socket
After=aziot-keyd.socket

[Service]
Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service"
ExecStart=/usr/libexec/aziot-identity-service/aziot-keyd
KillMode=process
Restart=on-failure
RestartSec=5s
**_User=aziotks_**

[Install]
WantedBy=multi-user.target

**rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-tpmd.service**
[Unit]
Description=Azure IoT TPM Service
Requires=aziot-tpmd.socket
After=aziot-tpmd.socket

[Service]
Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service"
ExecStart=/usr/libexec/aziot-identity-service/aziot-tpmd
KillMode=process
Restart=on-failure
RestartSec=5s
**_User=aziottpm_**

[Install]
WantedBy=multi-user.target
rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$

maskimthedog, 2022-12-23:
> I noticed that the systemd **.service** files below which I unarchived from the current IoT Edge Debian 11 Arm64 deb files specify **_User_** while the ones in the Yocto GitHub repos do not specify **_User_**. Not sure if this is important for Yocto or not. Just an observation at the moment.
> 
> **rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-certd.service** [Unit] Description=Azure IoT Certificates Service Requires=aziot-certd.socket After=aziot-certd.socket
> 
> [Service] Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service" ExecStart=/usr/libexec/aziot-identity-service/aziot-certd KillMode=process Restart=on-failure RestartSec=5s **_User=aziotcs_**
> 
> [Install] WantedBy=multi-user.target
> 
> **rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-identityd.service** [Unit] Description=Azure IoT Identity Service Requires=aziot-identityd.socket After=aziot-identityd.socket
> 
> [Service] Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service" ExecStart=/usr/libexec/aziot-identity-service/aziot-identityd KillMode=process Restart=on-failure RestartSec=5s **_User=aziotid_**
> 
> [Install] WantedBy=multi-user.target
> 
> **rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-keyd.service** [Unit] Description=Azure IoT Keys Service Requires=aziot-keyd.socket After=aziot-keyd.socket
> 
> [Service] Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service" ExecStart=/usr/libexec/aziot-identity-service/aziot-keyd KillMode=process Restart=on-failure RestartSec=5s **_User=aziotks_**
> 
> [Install] WantedBy=multi-user.target
> 
> **rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$ cat aziot-tpmd.service** [Unit] Description=Azure IoT TPM Service Requires=aziot-tpmd.socket After=aziot-tpmd.socket
> 
> [Service] Environment="LD_LIBRARY_PATH=/usr/lib/aziot-identity-service" ExecStart=/usr/libexec/aziot-identity-service/aziot-tpmd KillMode=process Restart=on-failure RestartSec=5s **_User=aziottpm_**
> 
> [Install] WantedBy=multi-user.target rkrakora@opbuild-7920:~/Downloads/lib/systemd/system$

Looks like the services mentioned are not getting started up as the correct users.  Also the aziot_keys.so does not seem to be getting installed in /usr/lib/aarch64-linux-gnu/engines-1.1.

ggjjj, 2023-01-11:
@maskimthedog  we do have the users added here. https://github.com/Azure/meta-iotedge/blob/083c7baa2aa3fd79777f464d2afaf5bedf750fc8/recipes-core/aziotd/aziotd.inc#L69-L74 

does it happen only on ARM device?

maskimthedog, 2023-01-11:
We have the users added, but we are not running the services as the appropriate user.  Also, there is a post installation step needed where all ADMIN and SUDO users need to be added to iotedge group and a few directories with wrong user ownership and permissions.  Take a look at my fork<https://github.com/maskimthedog/meta-iotedge>.  We are testing today.

Rob Krakora, Senior Embedded Engineer
[https://ci6.googleusercontent.com/proxy/bsn1VfuRFOg6BhNBKA-YGjT7_69kzftLHSVXIdWa6izu7Vb587WqtW_lQlGZFsCwUG4qOSFsZfce3-adwnpnySDz2JXF7ROmMzFLefTQveJ5n10=s0-d-e1-ft#https://meshsystems.blob.core.windows.net/helpimages/mesh.png]
p:
317.661.<tel:(317)%20661-4800>4800
f:
317.661.4801<tel:(317)%20661-4801>
w:
mesh-systems.com<http://mesh-systems.com/>
[https://ci6.googleusercontent.com/proxy/Sh-8YmBCXXgMOdOoPgPtgPnKQjxnBXcEwel_xmwS_cHPKRztClUPBPiJEHPdtG3Hei5hccBS7ONl-N_UCmjOHpHCwBzTomOq7BWdwmxz1KhrziiqbYJgeX4FOXOOh5DEqF8y=s0-d-e1-ft#http://cdn2.hubspot.net/hubfs/184235/dev_images/signature_app/twitter_sig.png]<https://twitter.com/mesh_systems>  [https://ci5.googleusercontent.com/proxy/hzQ50OystqOxlKMduZD32nQoNlmY4n_0V4lxPUWy5nU5cr1gbQR-03vxhbJL7qRILyM-QnaRbLIvZIoxOof999Ep266X82hDAOK2K4jC9rBUhUVgcslhZQ1jOUHJwKu2s6_4sA=s0-d-e1-ft#http://cdn2.hubspot.net/hubfs/184235/dev_images/signature_app/linkedin_sig.png] <https://www.linkedin.com/company/mesh-systems-llc>
Watch Our Corporate Overview<https://youtu.be/UGs-HR7KQQ8>

ggjjj, 2023-01-11:
Thanks for letting us know. I added the users for the services while making the PR not sure how it got missed. thanks for noticing it. I will look into the post installation step for fixing wrong user groups

maskimthedog, 2023-01-13:
[image]

maskimthedog, 2023-01-13:
@ggjjj What I have on my fork IRT changes made to support x509 with PK in fTPM seems to be working quite well.

maskimthedog, 2023-01-13:
Would be good to have these changes and then update to latest 1.4.x LTS mainline.

maskimthedog, 2023-02-01:
Fixed in main.

maskimthedog, 2023-02-01:
Fixed in main.

closed 2023-02-01 by maskimthedog

volzer-lincoln, 2024-01-04:
> Looks like the services mentioned are not getting started up as the correct users. Also the aziot_keys.so does not seem to be getting installed in /usr/lib/aarch64-linux-gnu/engines-1.1.

I've noticed that aziot_keys.so is also missing from /usr/lib/aarch64-linux-gnu/engines-1.1 in my builds and I could be overlooking it, but I didn't see any changes that addressed this. Is there something I'm missing or does anyone have some suggestions on where I can look to dig deeper into the issue?