Azure/meta-iotedge #116: aziot-edged 1.4.8 doesn't build on dunfell due to missing dependencies

corollaries, 2023-02-08:
Last week an update was made to the dunfell branch of meta-iotedge to include aziot-edged 1.4.8. (Nice to see the new versions of iotedge being supported fast by the way!)

Unfortunately this is leading to some build errors that is related to work being done on the release/1.4 branch of oit-identity-service repository. After 1.4.2 was released of the identity service, a change was made that added new dependencies to http-common in this PR: https://github.com/Azure/iot-identity-service/commit/a6aacdaaadde4052f02eb828bd7f6ef583a550fd#diff-1f6b02f28b1d007276753d58e716161aba092eb5a789705bfd1a42ba06fedf82. 

Part of build error:

```
error: no matching package found
| searched package name: `rand`
| perhaps you meant:      want, fnv, idna, ...
| location searched: registry `crates-io`
| required by package `http-common v0.1.0 (/work/build/tmp-glibc/work/x86_64-oe-linux/aziot-edged/1.4.8-r0/aziot-cert-client-async)`
|     ... which satisfies git dependency `http-common` (locked to 0.1.0) of package `aziot-edged v0.1.0 (/work/build/tmp-glibc/work/x86_64-oe-linux/aziot-edged/1.4.8-r0/git/edgelet/aziot-edged)`
| WARNING: exit code 101 from a shell command.
```

Cargo is now obviously complaining that the `rand` dependency is missing and doesn't continue the build using the latest commit of the dunfell branch. This currently applies only to the `iotedge_1.4.8.bb` and `aziot-edged_1.4.8` files as far as I can see. I think there are two solutions to this:

1. Add the `rand` dependency for the recipes. (Will also require sub dependencies for `rand` again as I played around a little bit with this). 
2. Use the tag `1.4.2` instead of the branch `release/1.4` for all the `SRCREV_*` definitions in the recipe files.

I think going for the second solution would be best as we could run into the same problem in the future when more dependencies are added on the `release/x` branch, plus the current definitions kind of break reproducible builds.

rschwarz-sotec, 2023-02-10:
I can confirm this build error:

```
| error: no matching package found
| searched package name: `rand`
| perhaps you meant:      want, fnv, idna, ...
| location searched: registry `crates-io`
| required by package `http-common v0.1.0 (/home/yocto/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge/1.4.8-r0/aziot-cert-client-async)`
|     ... which satisfies git dependency `http-common` (locked to 0.1.0) of package `iotedge v0.1.0 (/home/yocto/build/tmp/work/armv7vet2hf-neon-poky-linux-gnueabi/iotedge/1.4.8-r0/git/edgelet/iotedge)`
| WARNING: exit code 101 from a shell command.
```

maskimthedog, 2023-02-14:
> Last week an update was made to the dunfell branch of meta-iotedge to include aziot-edged 1.4.8. (Nice to see the new versions of iotedge being supported fast by the way!)
> 
> Unfortunately this is leading to some build errors that is related to work being done on the release/1.4 branch of iot-identity-service repository. After 1.4.2 was released of the identity service, a change was made that added new dependencies to http-common in this PR: [Azure/iot-identity-service@a6aacda#diff-1f6b02f28b1d007276753d58e716161aba092eb5a789705bfd1a42ba06fedf82](https://github.com/Azure/iot-identity-service/commit/a6aacdaaadde4052f02eb828bd7f6ef583a550fd#diff-1f6b02f28b1d007276753d58e716161aba092eb5a789705bfd1a42ba06fedf82).
> 
> Part of build error:
> 
> ```
> error: no matching package found
> | searched package name: `rand`
> | perhaps you meant:      want, fnv, idna, ...
> | location searched: registry `crates-io`
> | required by package `http-common v0.1.0 (/work/build/tmp-glibc/work/x86_64-oe-linux/aziot-edged/1.4.8-r0/aziot-cert-client-async)`
> |     ... which satisfies git dependency `http-common` (locked to 0.1.0) of package `aziot-edged v0.1.0 (/work/build/tmp-glibc/work/x86_64-oe-linux/aziot-edged/1.4.8-r0/git/edgelet/aziot-edged)`
> | WARNING: exit code 101 from a shell command.
> ```
> 
> Cargo is now obviously complaining that the `rand` dependency is missing and doesn't continue the build using the latest commit of the dunfell branch. This currently applies only to the `iotedge_1.4.8.bb` and `aziot-edged_1.4.8` files as far as I can see. I think there are two solutions to this:
> 
> 1. Add the `rand` dependency for the recipes. (Will also require a bunch of sub dependencies for `rand`).
> 2. Use the tag `1.4.2` instead of the branch `release/1.4` for all the `SRCREV_*` definitions in the recipe files.
> 
> I think going for the second solution would be best as we could run into the same problem in the future when more dependencies are added on the `release/x` branch, plus the current definitions kind of break reproducible builds.

I just built 1.4.8 early last week and it built fine but now appears broken when I rerun my build on the server from scratch.

maskimthedog, 2023-02-14:
@micahl  I guess SRCREV needs to reference a tag rather than release/1.4 in order to prevent this issue from arising in the future.  cargo-bitbake needs to be run on the crates and update the dependencies in the recipes in the short term.

micahl, 2023-02-14:
@ggjjj is investigating

maskimthedog, 2023-02-16:
@micahl I forked and made the change suggested by @corollaries and tested on one of our customers targets which utilizes x509 with keys stored in TPM and it worked great.  iotedge --version reported 1.4.8.  Maybe put this fix in place until @ggjjj is able to come up with a better solution?  That way the tip is not broken and you don't have everyone forking and pointing their manifests at their fork like I did.  :)

micahl, 2023-02-23:
@ggjjj is this now fixed with https://github.com/Azure/meta-iotedge/commit/72fb9186cba3928335373eef938b618c23a4ffff as part of bumping the Dunfell branch to the latest patch release 1.4.9?

ggjjj, 2023-02-23:
yes also the rust version is bumped up to 1.65 to resolve the dependency issue

maskimthedog, 2023-02-24:
@ggjjj I can confirm that this issue is fixed.

corollaries, 2023-02-24:
@ggjjj Thanks for creating the new patch release!

I've had a look at the current source but I noticed the `SRCREV_.*` definitions are still set to `release/1.4` for the recipes. This will work, for now, but given the continued work in that branch on the `aziot-identity-service` repository this might end up failing in the near future again upon some breaking change.

I'm also worried about the reproducibility of the builds. A nightly build on one day might be different from the other if we keep the `SRCREV_.*` definitions as they currently are. This can result in unwanted behavior. 

Is there a particular reason for staying on the `release/1.4` branch instead of using a tag release for the `aziot-identity-service` repository?

ggjjj, 2023-03-08:
Currently main branch has them removed. I have plans to do the same for dunfell

ggjjj, 2023-03-09:
Fix merged. Please re-open if needed

closed 2023-03-09 by ggjjj