Azure/meta-iotedge #121: azuit-edged 1.4.9 doesn't want to build with read-only root

sindelar-fr, 2023-02-27:
I'm still investigating this, but build with ro root started failing for me (maybe it's "broken" for longer than since 1.4.9).

bitbake ends with
`
do_rootfs: The following packages could not be configured offline and rootfs is read-only: ['aziot-edged']
`
log:
`
NOTE: If an image is being built, the postinstalls for the following packages will be postponed for first boot: aziot-edged
NOTE: Executing write_package_manifest ...
DEBUG: Executing python function write_package_manifest
DEBUG: Python function write_package_manifest finished
NOTE: Executing license_create_manifest ...
DEBUG: Executing python function license_create_manifest
DEBUG: Python function license_create_manifest finished
NOTE: Executing zap_empty_root_password ...
DEBUG: Executing shell function zap_empty_root_password
DEBUG: Shell function zap_empty_root_password finished
NOTE: Executing rootfs_update_timestamp ...
DEBUG: Executing shell function rootfs_update_timestamp
NOTE: rootfs_update_timestamp: set /etc/timestamp to 20180309123456
DEBUG: Shell function rootfs_update_timestamp finished
NOTE: Executing read_only_rootfs_hook ...
DEBUG: Executing shell function read_only_rootfs_hook
DEBUG: Shell function read_only_rootfs_hook finished
NOTE: Executing write_image_test_data ...
DEBUG: Executing python function write_image_test_data
DEBUG: Python function write_image_test_data finished
NOTE: Executing set_systemd_default_target ...
DEBUG: Executing shell function set_systemd_default_target
DEBUG: Shell function set_systemd_default_target finished
NOTE: Executing systemd_create_users ...
DEBUG: Executing shell function systemd_create_users
groupadd: group 'systemd-journal' already exists
useradd: user 'systemd-network' already exists
useradd: user 'systemd-resolve' already exists
useradd: user 'systemd-timesync' already exists
DEBUG: Shell function systemd_create_users finished
NOTE: Executing empty_var_volatile ...
DEBUG: Executing shell function empty_var_volatile
DEBUG: Shell function empty_var_volatile finished
NOTE: Executing deploy_pkgstatus ...
DEBUG: Executing shell function deploy_pkgstatus
DEBUG: Shell function deploy_pkgstatus finished
NOTE: Executing sort_passwd ...
DEBUG: Executing python function sort_passwd
DEBUG: Python function sort_passwd finished
NOTE: Executing rootfs_reproducible ...
DEBUG: Executing shell function rootfs_reproducible
NOTE: rootfs_reproducible: set /etc/version to 20180309123456
DEBUG: Shell function rootfs_reproducible finished
ERROR: The following packages could not be configured offline and rootfs is read-only: ['aziot-edged']
DEBUG: Python function do_rootfs finished
`

sindelar-fr, 2023-02-27:
It is that. Hmm being in .inc file it's not that easy to override ...

maskimthedog, 2023-03-08:
@sindelar-fr we added a post install script to add admin and sudo users to the iotedge group so the services could be start/stop/resume from an admin/sudo user directly.  On a read-only rootfs you need to add the following to your local.conf.  Just following what was done in the Debian/Ubuntu installers for IoT Edge.  @timesys-nathan noticed this during the Kirkstone Upgrade efforts.  Maybe needs to be refactored to omit as this seems only useful to desktop environments possibly?

IMAGE_FEATURES_append = " read-only-rootfs-delayed-postinsts"

maskimthedog, 2023-03-08:
@sindelar-fr and @ggjjj, on second thought, the below section in aziot-edged.inc needs to be omitted.  It's really only useful in a desktop environment.

```
pkg_postinst_ontarget_${PN}() {
    #!/bin/sh

    # Add each admin user to the iotedge group
    for u in $(getent group admin | sed -e "s/^.*://" -e "s/,/ /g"); do
        usermod -aG iotedge "$u" >/dev/null || true
    done

    # Add each sudo user to the iotedge group
    for u in $(getent group sudo | sed -e "s/^.*://" -e "s/,/ /g"); do
        usermod -aG iotedge "$u" >/dev/null || true
    done
}
```

sindelar-fr, 2023-03-08:
> @sindelar-fr and @ggjjj, on second thought, the below section in aziot-edged.inc needs to be omitted. It's really only useful in a desktop environment.
>
> ```
> pkg_postinst_ontarget_${PN}() { [...]

Yes, that's what I've done locally to temporarily unblock us - I've just removed this and it did build fine. But since it's included in the bb file, it's not that simple to get rid of in bbappend (I've seen something with using python to remove a variable of the function thus effectively removing it from what's already included, but that seems a bit more like a hack than a sollution and I didn't really have time to investigate it further yet).

maskimthedog, 2023-03-08:
@sindelar-fr it needs to be gone.  I debated putting it in their initially.  (I was not working with read only rootfs).  I then started working with a read only rootfs a couple of weeks ago and regretted my decision.

ggjjj, 2023-03-09:
Fix merged. Please re-open if needed

closed 2023-03-09 by ggjjj