Azure/meta-iotedge #126: Cannot reprovision with 1.4.9 due to 401 error

maskimthedog, 2023-03-09:
After 1.4.9 was released, I did a build and was able to re-provision successfully (iotedge system reprovision).  I did a clean build just a few days ago and now reprovisioning is failing.

[image]

ggjjj, 2023-03-09:
dunfell branch?

maskimthedog, 2023-03-09:
dunfell, yes.

maskimthedog, 2023-03-09:
> dunfell branch?

Yes.

maskimthedog, 2023-03-09:
Odd, because we built 1.4.9 when it was first release two weeks ago and that image works fine IRT reprovisioning.  We did a fresh build with 1.4.9 and ran into this reprovisioning issue.

maskimthedog, 2023-03-11:
We have been working on a Yocto recipe for [azure-osconfig](https://github.com/Azure/azure-osconfig) and it generated osconfig.toml (contents below) and placed it in the /etc/aziot/identityd/config.d directory.  I deleted it, rebooted, ran "iotedge config apply" and now "iotedge system reprovision" no longer fails with 401.  It works just find.

[[principal]]
name="osconfig"
idtype=["module"]
uid=0

ggjjj, 2023-03-15:
Thanks for sharing. I do not have much info on why this would cause provisioning to fail. Will keep this issue open so anyone with the same issue can use your fix

gdoffe, 2023-09-01:
Hi. I have a similar problem (maybe unrelated so just tell me and I create a new issue).

I'm on Kirkstone using last commit on main branch: https://github.com/Azure/meta-iotedge/commit/cba54fdae09d8c51c73b11e2b4879331006d1f2d

```
Sep 01 01:21:39 mydevice aziot-identityd[18161]: 2023-08-31T23:21:39Z [INFO] - !!! authorization error
Sep 01 01:21:39 mydevice aziot-identityd[18161]: 2023-08-31T23:21:39Z [INFO] - --> 401 {"content-type": "application/json"}
Sep 01 01:21:39 mydevice aziot-edged[18136]: 2023-08-31T23:21:39Z [WARN] - Failed to reprovision: invalid HTTP status code
Sep 01 01:21:44 mydevice aziot-edged[18136]: 2023-08-31T23:21:44Z [INFO] - Obtaining Edge device provisioning data...
Sep 01 01:21:44 mydevice aziot-edged[18136]: 2023-08-31T23:21:44Z [ERR!] - Failed to obtain device identity: authorization error
Sep 01 01:21:44 mydevice aziot-edged[18136]: 2023-08-31T23:21:44Z [INFO] - Requesting device reprovision
Sep 01 01:21:44 mydevice aziot-edged[18136]: 2023-08-31T23:21:44Z [WARN] - Failed to clear provisioning cache before reprovision: No such file or directory (os error 2)
Sep 01 01:21:44 mydevice aziot-identityd[18161]: 2023-08-31T23:21:44Z [INFO] - <-- POST /identities/device?api-version=2020-09-01 {"content-type": "application/json", "host": "identityd.sock", "content-length": "16"}
Sep 01 01:21:44 mydevice aziot-identityd[18161]: 2023-08-31T23:21:44Z [INFO] - !!! authorization error
Sep 01 01:21:44 mydevice aziot-identityd[18161]: 2023-08-31T23:21:44Z [INFO] - --> 401 {"content-type": "application/json"}
Sep 01 01:21:44 mydevice aziot-identityd[18161]: 2023-08-31T23:21:44Z [INFO] - <-- POST /identities/device/reprovision?api-version=2020-09-01 {"content-type": "application/json", "host": "identityd.sock", "content-length": "16"}
Sep 01 01:21:44 mydevice aziot-identityd[18161]: 2023-08-31T23:21:44Z [INFO] - !!! authorization error
Sep 01 01:21:44 mydevice aziot-identityd[18161]: 2023-08-31T23:21:44Z [INFO] - --> 401 {"content-type": "application/json"}
Sep 01 01:21:44 mydevice aziot-edged[18136]: 2023-08-31T23:21:44Z [WARN] - Failed to reprovision: invalid HTTP status code
Sep 01 01:21:49 mydevice aziot-edged[18136]: 2023-08-31T23:21:49Z [INFO] - Obtaining Edge device provisioning data...
Sep 01 01:21:49 mydevice aziot-edged[18136]: 2023-08-31T23:21:49Z [ERR!] - Failed to obtain device identity: authorization error
Sep 01 01:21:49 mydevice aziot-edged[18136]: 2023-08-31T23:21:49Z [INFO] - Requesting device reprovision
Sep 01 01:21:49 mydevice aziot-edged[18136]: 2023-08-31T23:21:49Z [WARN] - Failed to clear provisioning cache before reprovision: No such file or directory (os error 2)
Sep 01 01:21:49 mydevice aziot-identityd[18161]: 2023-08-31T23:21:49Z [INFO] - <-- POST /identities/device?api-version=2020-09-01 {"content-type": "application/json", "host": "identityd.sock", "content-length": "16"}
Sep 01 01:21:49 mydevice aziot-identityd[18161]: 2023-08-31T23:21:49Z [INFO] - !!! authorization error
Sep 01 01:21:49 mydevice aziot-identityd[18161]: 2023-08-31T23:21:49Z [INFO] - --> 401 {"content-type": "application/json"}
Sep 01 01:21:49 mydevice aziot-identityd[18161]: 2023-08-31T23:21:49Z [INFO] - <-- POST /identities/device/reprovision?api-version=2020-09-01 {"content-type": "application/json", "host": "identityd.sock", "content-length": "16"}
Sep 01 01:21:49 mydevice aziot-identityd[18161]: 2023-08-31T23:21:49Z [INFO] - !!! authorization error
Sep 01 01:21:49 mydevice aziot-identityd[18161]: 2023-08-31T23:21:49Z [INFO] - --> 401 {"content-type": "application/json"}
```

Using `curl` request on `identity.sock` as `iotedge` user I have the following result:
```shell
# su - iotedge -s /bin/bash -c "curl --unix-socket '/run/aziot/identityd.sock' 'http://identityd.sock/identities/identity?api-version=2020-09-01'" && echo
{"message":"authorization error"}
```
But as root it works fine:
```
# curl --unix-socket '/run/aziot/identityd.sock' 'http://identityd.sock/identities/identity?api-version=2020-09-01' && echo
{"type":"aziot","spec":{"hubName":"blablabla-dev.azure-devices.net","gatewayHost":"blablabla.azure-devices.net","deviceId":"XXXX","auth":{"type":"tpm"}}}
```

So I had to modify all systemd aziot services files to run aziot daemons as root and then it works fine.
But for evident security concerns I would like to make user services working.

Do I have special configuration to do in some config.toml to make this work ?

Thanks for help. ;)

gdoffe, 2023-09-06:
ok, fixed, due to bad `[[principal]]` configurations. Fixed by using `iotedge config apply` and diff with my actual configuration.

But do you think that such principal configuration could not be applied by yocto recipes:
```
# cat /etc/aziot/identityd/config.d/aziot-edged.toml 
[[principal]]
uid = 987
name = "aziot-edge"
```

Does that make sense ?

ggjjj, 2024-07-22:
As dunfell went out of support, closing this issue

closed 2024-07-22 by ggjjj