Azure/meta-iotedge #129: build fail at kirkstone branch

harryliou, 2023-03-16:
Error log:

ERROR: aziotctl-1.4.3.AUTOINC+cfa39b58ec-r0 do_compile: ExecutionError('/source/build_imx8mq-DEBUG/tmp/work/armv8a-poky-linux/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/temp/run.do_compile.23117', 101, None, None)
ERROR: Logfile of failure stored in: /source/build_imx8mq-DEBUG/tmp/work/armv8a-poky-linux/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/temp/log.do_compile.23117
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /source/build_imx8mq-DEBUG/tmp/work/armv8a-poky-linux/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target aarch64-poky-linux --release --manifest-path=/source/build_imx8mq-DEBUG/tmp/work/armv8a-poky-linux/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/git/aziotctl/Cargo.toml
| error: failed to get `anyhow` as a dependency of package `aziotctl v1.4.3 (/source/build_imx8mq-DEBUG/tmp/work/armv8a-poky-linux/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/git/aziotctl)`
| 
| Caused by:
|   failed to load source for dependency `anyhow`
| 
| Caused by:
|   Unable to update registry `crates-io`
| 
| Caused by:
|   failed to update replaced source registry `crates-io`
| 
| Caused by:
|   failed to parse manifest at `/source/build_imx8mq-DEBUG/tmp/work/armv8a-poky-linux/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/cargo_home/bitbake/toml-0.7.2/Cargo.toml`
| 
| Caused by:
|   namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
| WARNING: exit code 101 from a shell command.
NOTE: recipe aziotctl-1.4.3.AUTOINC+cfa39b58ec-r0: task do_compile: Failed
ERROR: Task (/source/sources/meta-iotedge/recipes-core/aziotctl/aziotctl_1.4.3.bb:do_compile) failed with exit code '1'
NOTE: recipe linux-harry-imx-local-5.15.52+gitAUTOINC+4019891b43-r0: task do_unpack: Succeeded
NOTE: Tasks Summary: Attempted 1964 tasks of which 1952 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /source/sources/meta-iotedge/recipes-core/aziotctl/aziotctl_1.4.3.bb:do_compile
Summary: There were 96 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

meta-rust:
RUST_VERSION ?= "1.66.0"

local.conf:
PREFERRED_VERSION_rust ?= "1.65%"
DISTRO_FEATURES:append = "systemd virtualization"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"

harryliou, 2023-03-27:
After referring to conf/local.conf.sample, fix the error.
Thanks :)

closed 2023-03-27 by harryliou

manojrampur, 2023-11-16:
Hello @harryliou,

I'm encountering a similar issue. Could you share the steps you took to resolve the problem in the local.conf file? 

Additionally, Can you please provide the meta-iotedge layer and its dependent layers, along with their respective commit IDs?

Your assistance in achieving this would be highly appreciated.

Thanks in advance!!