Azure/meta-iotedge #144: How to solve the aziotctl cargo error while building Yocto Image?

manojrampur, 2023-11-17:
I'm building the Yocto Image using the Kirkstone branch using meta-iotedge layers and its dependency layers. While building the image. I have provided the layer details along with the commit IDs and the error as well below. Please have a look into it.

**Layers with commit IDs:**
meta-virtualization  = "HEAD:478a91800c4a95c7410dc8ef52e229a81be24b4e"
meta-iotedge         = "HEAD:1276528e537c2cc5a7342eecf4643e686d1c5978"
meta-rust            = "HEAD:083f6103b914c2f162b3f966822d9aeaf8cb6b99"
meta-intel           = "HEAD:3f0115e9e59f94af54b74838f4f33339ea7584ae"
meta-clang           = "HEAD:79169d9be565b7a87310ca280d3a21aaf608ce33"
meta-security
meta-tpm             = "HEAD:1a3e42cedbd94ca73be45800d0e902fec35d0f0f"

**Error:**
ERROR: aziotctl-1.4.3.AUTOINC+cfa39b58ec-r0 do_compile: ExecutionError('/home/azureuser/poky-kirkstone/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/temp/run.do_compile.3017269', 101, None, None)
ERROR: Logfile of failure stored in: /home/azureuser/poky-kirkstone/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/temp/log.do_compile.3017269
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /home/azureuser/poky-kirkstone/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target arm-poky-linux-gnueabi --release --manifest-path=/home/azureuser/poky-kirkstone/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/git/aziotctl/Cargo.toml
| error: failed to get `anyhow` as a dependency of package `aziotctl v1.4.3 (/home/azureuser/poky-kirkstone/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/git/aziotctl)`
|
| Caused by:
|   failed to load source for dependency `anyhow`
|
| Caused by:
|   Unable to update registry `crates-io`
|
| Caused by:
|   failed to update replaced source registry `crates-io`
|
| Caused by:
|   failed to parse manifest at `/home/azureuser/poky-kirkstone/build/tmp/work/cortexa7t2hf-neon-vfpv4-poky-linux-gnueabi/aziotctl/1.4.3.AUTOINC+cfa39b58ec-r0/cargo_home/bitbake/toml-0.7.2/Cargo.toml`
|
| Caused by:
|   namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
| WARNING: exit code 101 from a shell command.
ERROR: Task (/home/azureuser/poky-kirkstone/meta-iotedge/recipes-core/aziotctl/aziotctl_1.4.3.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 5684 tasks of which 5666 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/azureuser/poky-kirkstone/meta-iotedge/recipes-core/aziotctl/aziotctl_1.4.3.bb:do_compile
Summary: There were 2 WARNING messages.
Summary: There was 1 ERROR message, returning a non-zero exit code.

Please respond asap to resolve the issue.

Thanks in Advance!!

rschwarz-sotec, 2023-11-23:
Hi @manojrampur ,

did you add the BBMASK lines as mentioned in the local.conf.sample to your local.conf?

https://github.com/Azure/meta-iotedge/blob/d1f09b5bfe29db57c75447b73912c47fc623f442/conf/local.conf.sample#L245C1-L247C34

hurtadosanti, 2023-12-07:
Hi, I have the same issue I , here is my configuration:

```
...
CONF_VERSION = "2"

BBMASK += "poky/meta/recipes-devtools/rust"
BBMASK += "poky/meta/recipes-devtools/cargo"
PREFERRED_VERSION_rust ?= "1.65%"
DISTRO_FEATURES:append = " systemd virtualization"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"

```
My build configuration, I try with the latest from meta-rust and also the issue is there.

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "x86_64-poky-linux"
MACHINE              = "qemux86-64"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.14"
TUNE_FEATURES        = "m64 core2"
TARGET_FPU           = ""
meta-iotedge         = "main:d1f09b5bfe29db57c75447b73912c47fc623f442"
meta                 
meta-poky            = "kirkstone:3e73216a32a2b01916eb8c555a41579bd69a47aa"
meta-rust            = "default:5656b5ad104bbbbe1b711e3bbe35c88f0d9c9a05"
meta-yocto-bsp       = "kirkstone:3e73216a32a2b01916eb8c555a41579bd69a47aa"
meta-filesystems     
meta-oe              
meta-networking      
meta-python          = "kirkstone:8a75c61cce2aa1d6e5a3597ab8fc5a7e6aeae1e4"
meta-virtualization  = "kirkstone:67b0ef4256a484851f3556d281ca4caa88374388"
meta-tpm             = "kirkstone:1a3e42cedbd94ca73be45800d0e902fec35d0f0f"
meta-clang           = "kirkstone:79169d9be565b7a87310ca280d3a21aaf608ce33"
```
And the error: 
```
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /data/iotedge/build/tmp/work/core2-64-poky-linux/aziotctl/1.4.6.AUTOINC+e4b08714e1-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target x86_64-poky-linux --release --manifest-path=/data/iotedge/build/tmp/work/core2-64-poky-linux/aziotctl/1.4.6.AUTOINC+e4b08714e1-r0/git/aziotctl/Cargo.toml
| error: failed to get `anyhow` as a dependency of package `aziotctl v1.4.6 (/data/iotedge/build/tmp/work/core2-64-poky-linux/aziotctl/1.4.6.AUTOINC+e4b08714e1-r0/git/aziotctl)`
| 
| Caused by:
|   failed to load source for dependency `anyhow`
| 
| Caused by:
|   Unable to update registry `crates-io`
| 
| Caused by:
|   failed to update replaced source registry `crates-io`
| 
| Caused by:
|   failed to parse manifest at `/data/iotedge/build/tmp/work/core2-64-poky-linux/aziotctl/1.4.6.AUTOINC+e4b08714e1-r0/cargo_home/bitbake/time-0.3.22/Cargo.toml`
| 
| Caused by:
|   namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
| WARNING: exit code 101 from a shell command.
```

hurtadosanti, 2024-01-03:
Hi @mile-mwebert based on your answer, I try another approach using kas, I also remove completely the .cargo folder from my machine, not sure if both are related but now is working: 

This config:

```
BBMASK += "poky/meta/recipes-devtools/rust"
BBMASK += "poky/meta/recipes-devtools/cargo"
PREFERRED_VERSION_rust ?= "1.65%"
DISTRO_FEATURES:append = " systemd virtualization"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
```

With this lock is now working:
```
header:
    version: 1
overrides:
    repos:
        meta-clang:
            commit: 312ff1c39b1bf5d35c0321e873417eb013cea477
        meta-iotedge:
            commit: 6518416d7a3ee5260f5fb879216df35ee3115a4d
        meta-openembedded:
            commit: 402affcc073db39f782c1ebfd718edd5f11eed4c
        meta-rust:
            commit: f6a80656953678f17de7d41343107112d99492d9
        meta-security:
            commit: 1a3e42cedbd94ca73be45800d0e902fec35d0f0f
        meta-virtualization:
            commit: 88327090d26955a678c6f8bd2585aad4d802f6c4
        poky:
            commit: f754b5d45b504e7486364ea9cfcb5f96086084bc
```

emilm, 2024-02-06:
@hurtadosanti Nice! I don't use kas but do you really need to pin all those commits?

manojrampur, 2024-03-25:
The solution is:

1. Delete the **rust** and **cargo** folder from the **meta/recipes-devtools** directory
2. Copy the **rust** and **cargo** folder from the **meta-rust** directory
3. Paste those folders in the **meta/recipes-devtools** directory
4. Start the build

**meta-rust commit ID: 5656b5ad104bbbbe1b711e3bbe35c88f0d9c9a05**

The issue will be resolved.

closed 2024-03-25 by manojrampur