Azure/meta-iotedge #153: Yocto support until April 2024?

hurtadosanti, 2024-03-07:
According to the official website Yocto support as tier 2 will be stopped end of April 2024. Is this correct?

rschwarz-sotec, 2024-03-07:
I guess this is related to the current yocto kirkstone LTS release which also ends in April 2024. Hopefully there will be an upgrade to next yocto LTS release nanbield?!?

moto-timo, 2024-04-09:
The `dunfell` (Yocto Project 4.0 LTS) release ends support in April 2024. The latest LTS is in the final process of being released `scarthgap` (Yocto Project 5.0 LTS) with support ending in April 2028. Hopefully a scarthgap branch will be created soon. I started working on it: https://github.com/moto-timo/meta-iotedge/tree/scarthgap

Whether the maintainers of `meta-iotedge` decide to go in a new direction (dropping meta-rust) is outside of my knowledge or control, but I am happy to share my architecture decisions and direction. I would also like to see a `packagegroup-iotedge` added, since we always need the same applications to make `iotedge check` work.

ggjjj, 2024-07-22:
We will release Kirkstone recipes for 1.4 LTS and 1.5 LTS and continue to support it till April 2024. 
We are also planning Scarthgap release in the coming months.

hurtadosanti, 2024-09-24:
Thanks now we can use it and see the long term goal of the project.

closed 2024-09-24 by hurtadosanti