Azure/meta-iotedge #164: Support for IoT Edge 1.5.0 and Kirkstone

marcovinci, 2024-06-10:
Hello,

I would like to ask if there are any plans to support IoT Edge 1.5.0 on Kirkstone in the next months. I see on Microsoft website that Kirkstone should be supported up to April 2026.

https://learn.microsoft.com/en-us/azure/iot-edge/support?view=iotedge-1.5

Thank You

maskimthedog, 2024-06-28:
@ggjjj any news on this?  We need v1.5 with .NET8 support.

jolting, 2024-07-09:
Kirkstone ships with a Rust version 1.59 which I think is too old. Scarthgap ships with Rust 1.75 , which seems to work fine.
When I tried it based on #166 I got the following error.

```
Initialising tasks: 100% |#######################################################################################################################| Time: 0:00:01
Checking sstate mirror object availability: 100% |###############################################################################################| Time: 0:00:03
Sstate summary: Wanted 138 Local 0 Mirrors 50 Missed 88 Current 549 (36% match, 87% complete)
NOTE: Executing Tasks
ERROR: aziotctl-1.5.0-r0 do_compile: ExecutionError('/var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/temp/run.do_compile.110', 101, None, None)
ERROR: Logfile of failure stored in: /var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/temp/log.do_compile.110
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target x86_64-poky-linux --release --manifest-path=/var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/git/aziotctl/Cargo.toml
| error: failed to get `anyhow` as a dependency of package `aziotctl v1.5.0 (/var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/git/aziotctl)`
| 
| Caused by:
|   failed to load source for dependency `anyhow`
| 
| Caused by:
|   Unable to update registry `crates-io`
| 
| Caused by:
|   failed to update replaced source registry `crates-io`
| 
| Caused by:
|   failed to parse manifest at `/var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/cargo_home/bitbake/toml-0.7.5/Cargo.toml`
| 
| Caused by:
|   namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
| WARNING: /var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/temp/run.do_compile.110:199 exit 101 from '"cargo" build -v --target x86_64-poky-linux --release --manifest-path=/var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/git/aziotctl/Cargo.toml "$@"'
| WARNING: Backtrace (BB generated script):
| 	#1: oe_cargo_build, /var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/temp/run.do_compile.110, line 199
| 	#2: cargo_do_compile, /var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/temp/run.do_compile.110, line 170
| 	#3: do_compile, /var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/temp/run.do_compile.110, line 165
| 	#4: main, /var/build/build/tmp/work/core2-64-poky-linux/aziotctl/1.5.0-r0/temp/run.do_compile.110, line 212
ERROR: Task (/var/build/poky/meta-iotedge/recipes-core/aziotctl/aziotctl_1.5.0.bb:do_compile) failed with exit code '1'
ERROR: aziot-keys-1.5.0-r0 do_compile: ExecutionError('/var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/temp/run.do_compile.113', 101, None, None)
ERROR: Logfile of failure stored in: /var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/temp/log.do_compile.113
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target x86_64-poky-linux --release --manifest-path=/var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/git/key/aziot-keys/Cargo.toml
| error: failed to get `anyhow` as a dependency of package `aziotctl v1.5.0 (/var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/git/aziotctl)`
| 
| Caused by:
|   failed to load source for dependency `anyhow`
| 
| Caused by:
|   Unable to update registry `crates-io`
| 
| Caused by:
|   failed to update replaced source registry `crates-io`
| 
| Caused by:
|   failed to parse manifest at `/var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/cargo_home/bitbake/toml-0.7.5/Cargo.toml`
| 
| Caused by:
|   namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
| WARNING: /var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/temp/run.do_compile.113:193 exit 101 from '"cargo" build -v --target x86_64-poky-linux --release --manifest-path=/var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/git/key/aziot-keys/Cargo.toml "$@"'
| WARNING: Backtrace (BB generated script):
| 	#1: oe_cargo_build, /var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/temp/run.do_compile.113, line 193
| 	#2: cargo_do_compile, /var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/temp/run.do_compile.113, line 164
| 	#3: do_compile, /var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/temp/run.do_compile.113, line 159
| 	#4: main, /var/build/build/tmp/work/core2-64-poky-linux/aziot-keys/1.5.0-r0/temp/run.do_compile.113, line 206
ERROR: Task (/var/build/poky/meta-iotedge/recipes-core/aziot-keys/aziot-keys_1.5.0.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2392 tasks of which 2386 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /var/build/poky/meta-iotedge/recipes-core/aziotctl/aziotctl_1.5.0.bb:do_compile
  /var/build/poky/meta-iotedge/recipes-core/aziot-keys/aziot-keys_1.5.0.bb:do_compile
Summary: There was 1 WARNING message.
Summary: There were 2 ERROR messages, returning a non-zero exit code.
failed to build
```

sindelar-fr, 2024-07-22:
> Kirkstone ships with a Rust version 1.59 which I think is too old. Scarthgap ships with Rust 1.75 , which seems to work fine. When I tried it based on #166 I got the following error.
>
> ``` [...]

Can't we keep meta-rust for kirkstone?

ggjjj, 2024-07-22:
I am also facing the issue reported by @jolting. This is blocking the release of 1.4 and 1.5 versions of iotedge for kirkstone.

ggjjj, 2024-07-22:
> Can't we keep meta-rust for kirkstone?

as oe-core already has rust components, it makes sense to remove meta-rust but it does causes these issues which need solutions

jolting, 2024-07-26:
meta-rust doesn't support Scarthgap and I'm not sure if there are plans to support it in the future, so for now Scarthgap needs to use oe-core. Kirkstone can use meta-rust or oe-core, but within the kirkstone release oe-core ships with a version that is too old. 

I'll leave it up to you guys to get meta-rust working on Kirkstone.

There are some incompatibilities between meta-rust and oe-core rust, which I believe would prevent supporting both on the same branch. I recommend bifurcating Scarthgap and Kirkstone support on two separate branches. Kirkstone is supported until April 2026.

For some unrelated reasons(mostly driver related), I cannot use Kirkstone and needed to upgrade to Scarthgap.

closed 2026-02-05 by damonbarry