Azure/meta-iotedge #176: Building iotedge in Yocto Kirkstone on a TI AM68A platform

surajgampa95, 2024-08-29:
**Describe the bug**
Trouble building the iotedge package in Yocto project Kirkstone release using bitbake.  
**To Reproduce**
Steps to reproduce the behavior:
1. On a host Ubuntu 22.04 machine, download yocto-build and generate build directory using documentation at [Processor SDK for AM68A - Building SDK with Yocto](https://software-dl.ti.com/jacinto7/esd/processor-sdk-linux-am68a/09_02_00/exports/docs/linux/Overview_Building_the_SDK.html)
2. Download meta-iotedge layer in the sources directory and add the path to bbconf file in build directory
3. Make changes to localconf file in build dir. Added following lines:
`BBMASK += "${OEBASE}/sources/oe-core/meta/recipes-devtools/rust"
BBMASK += "${OEBASE}/sources/oe-core/meta/recipes-devtools/cargo"
PREFERRED_VERSION_rust ?= "1.66%"
DISTRO_FEATURES:append = " systemd virtualization"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"`
4. Used the following command to build iotedge in the build dir:
`bitbake iotedge`
**Logs**
ERROR: aziotctl-1.4.7.AUTOINC+91e058880c-r0 do_compile: ExecutionError('/media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/temp/run.do_compile.250136', 101, None, None)
ERROR: Logfile of failure stored in: /media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/temp/log.do_compile.250136
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/recipe-sysroot-native/usr/bin/cargo
| NOTE: rustc =
| NOTE: cargo build -v --target aarch64-oe-linux --release --manifest-path=/media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/git/aziotctl/Cargo.toml
| error: failed to get `anyhow` as a dependency of package `aziotctl v1.4.7 (/media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/git/aziotctl)`
| 
| Caused by:
|   failed to load source for dependency `anyhow`
| 
| Caused by:
|   Unable to update registry `crates-io`
| 
| Caused by:
|   failed to update replaced source registry `crates-io`
| 
| Caused by:
|   failed to parse manifest at `/media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/cargo_home/bitbake/byte-unit-4.0.19/Cargo.toml`
| 
| Caused by:
|   namespaced features with the `dep:` prefix are only allowed on the nightly channel and requires the `-Z namespaced-features` flag on the command-line
| WARNING: /media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/temp/run.do_compile.250136:193 exit 101 from '"cargo" build -v --target aarch64-oe-linux --release --manifest-path=/media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/git/aziotctl/Cargo.toml "$@"'
| WARNING: Backtrace (BB generated script):
| 	#1: oe_cargo_build, /media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/temp/run.do_compile.250136, line 193
| 	#2: cargo_do_compile, /media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/temp/run.do_compile.250136, line 164
| 	#3: do_compile, /media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/temp/run.do_compile.250136, line 159
| 	#4: main, /media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/build/arago-tmp-default-glibc/work/aarch64-oe-linux/aziotctl/1.4.7.AUTOINC+91e058880c-r0/temp/run.do_compile.250136, line 206
ERROR: Task (/media/rei-dev/d3e07922-2074-4f99-88d1-18778349e6a9/yocto-build/sources/meta-iotedge/recipes-core/aziotctl/aziotctl_1.4.7.bb:do_compile) failed with exit code '1'

**Build System (please complete the following information):**
 - OS: Ubuntu
 - Version: 22.04 

**Target System (please complete the following information):**
 - OS: Arago 
 - Version: 2023
 - Machine: TI AM68A, j721s2-evm

**Additional context**
The Yocto project provided by TI has Rust version 1.59 by default

surajgampa95, 2024-08-29:
When I update the Rust version to 1.78 using meta-rust, I get the following error (detailed log in attached file)
```
error: field `0` is never read
|   --> iotedge/src/config/import/old_config/provisioning.rs:21:14
|    |
| 21 |     External(External),
|    |     -------- ^^^^^^^^
|    |     |
|    |     field in this variant
|    |
| note: the lint level is defined here
|   --> iotedge/src/lib.rs:3:27
|    |
| 3  | #![deny(rust_2018_idioms, warnings)]
|    |                           ^^^^^^^^
|    = note: `#[deny(dead_code)]` implied by `#[deny(warnings)]`
| help: consider changing the field to be of unit type to suppress this warning while preserving the field numbering, or remove the field
|    |
| 21 |     External(()),
|    |              ~~
| 
| warning: `iotedge` (lib) generated 3 warnings (3 duplicates)
| error: could not compile `iotedge` (lib) due to 1 previous error; 3 warnings emitted
| 
```
[error-logs-meta-iotedge.txt](https://github.com/user-attachments/files/16802837/error-logs-meta-iotedge.txt)

ggjjj, 2024-09-05:
which kirkstone recipe are you building? This error usually comes when not using meta-rust layer but instead relying on the oe-core rust. It is at 1.59 and creates crate problems of versions used from nightly build

surajgampa95, 2024-09-06:
After the aforementioned issue, I updated my TI yocto project to Scarthgap so now I don't have issues with rust version. I pulled the main branch from meta-iotedge project, since it has updates for Scarthgap, and added it using bitbake using following commands -
`bitbake-layers add-layer meta-iotedge`
`MACHINE="am68-sk" bitbake iotedge`
`MACHINE="am68-sk" bitbake -c cleanall tisdk-edgeai-image`
`MACHINE="am68-sk" bitbake -k tisdk-edgeai-image`
 I am able to compile and build the iotedge and OS image. But when I inspect the root directory of the OS after booting off of this new image, I don't find any files related to iotedge. 
Any guidance as to how I can handle this issue?

surajgampa95, 2024-09-10:
I was able to install azure iot edge on the target platform. However, I am having trouble setting up edgeHub. Here is the logs:

`root@am68a-sk:/# sudo iotedge config apply -c '/etc/aziot/config.toml'
Warning: the previous identity config file is unreadable
Note: Symmetric key will be written to /var/secrets/aziot/keyd/device-id
Azure IoT Edge has been configured successfully!

Restarting service for configuration to take effect...
Stopping aziot-edged.service...Stopped!
Stopping aziot-identityd.service...Stopped!
Stopping aziot-keyd.service...Stopped!
Stopping aziot-certd.service...Stopped!
Stopping aziot-tpmd.service...Stopped!
Starting aziot-edged.mgmt.socket...Started!
Starting aziot-edged.workload.socket...Started!
Starting aziot-identityd.socket...Started!
Starting aziot-keyd.socket...Started!
Starting aziot-certd.socket...Started!
Starting aziot-tpmd.socket...Started!
Starting aziot-edged.service...Started!
Done.
root@am68a-sk:/# sudo iotedge check

Configuration checks (aziot-identity-service)
---------------------------------------------
√ keyd configuration is well-formed - OK
√ certd configuration is well-formed - OK
√ tpmd configuration is well-formed - OK
√ identityd configuration is well-formed - OK
√ daemon configurations up-to-date with config.toml - OK
√ identityd config toml file specifies a valid hostname - OK
‼ aziot-identity-service package is up-to-date - Warning
    Installed aziot-identity-service package has version 1.5.0 but 1.5.3 is the latest stable version available.
    Please see https://aka.ms/aziot-update-runtime for update instructions.
√ host time is close to reference time - OK
√ preloaded certificates are valid - OK
√ keyd is running - OK
√ certd is running - OK
√ identityd is running - OK
√ read all preloaded certificates from the Certificates Service - OK
√ read all preloaded key pairs from the Keys Service - OK
√ check all EST server URLs utilize HTTPS - OK
√ ensure all preloaded certificates match preloaded private keys with the same ID - OK

Connectivity checks (aziot-identity-service)
--------------------------------------------
√ host can connect to and perform TLS handshake with iothub AMQP port - OK
√ host can connect to and perform TLS handshake with iothub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with iothub MQTT port - OK

Configuration checks
--------------------
√ aziot-edged configuration is well-formed - OK
√ configuration up-to-date with config.toml - OK
√ container engine is installed and functional - OK
× configuration has correct URIs for daemon mgmt endpoint - Error
    Unable to find image 'mcr.microsoft.com/azureiotedge-diagnostics:1.5.5' locally
    docker: Error response from daemon: manifest for mcr.microsoft.com/azureiotedge-diagnostics:1.5.5 not found: manifest unknown: manifest tagged by "1.5.5" is not found.
    See 'docker run --help'.
‼ aziot-edge package is up-to-date - Warning
    Installed IoT Edge daemon has version 1.5.5 but 1.5.3 is the latest stable version available.
    Please see https://aka.ms/iotedge-update-runtime for update instructions.
× container time is close to host time - Error
    Could not query local time inside container
‼ DNS server - Warning
    Container engine is not configured with DNS server setting, which may impact connectivity to IoT Hub.
    Please see https://aka.ms/iotedge-prod-checklist-dns for best practices.
    You can ignore this warning if you are setting DNS server per module in the Edge deployment.
‼ production readiness: logs policy - Warning
    Container engine is not configured to rotate module logs which may cause it run out of disk space.
[... 15 lines cut ...]
× container on the default network can connect to upstream HTTPS / WebSockets port - Error
    Container on the default network could not connect to reijetsons-iothub1.azure-devices.net:443
× container on the IoT Edge module network can connect to upstream AMQP port - Error
    Container on the azure-iot-edge network could not connect to reijetsons-iothub1.azure-devices.net:5671
× container on the IoT Edge module network can connect to upstream HTTPS / WebSockets port - Error
    Container on the azure-iot-edge network could not connect to reijetsons-iothub1.azure-devices.net:443
23 check(s) succeeded.
5 check(s) raised warnings. Re-run with --verbose for more details.
7 check(s) raised errors. Re-run with --verbose for more details.
2 check(s) were skipped due to errors from other checks. Re-run with --verbose for more details.

root@am68a-sk:/# sudo systemctl restart aziot-edged
root@am68a-sk:/# sudo iotedge system status 
System services:
    aziot-edged             Running
    aziot-identityd         Running
    aziot-keyd              Running
    aziot-certd             Running
    aziot-tpmd              Ready

Use 'iotedge system logs' to check for non-fatal errors.
Use 'iotedge check' to diagnose connectivity and configuration issues.
root@am68a-sk:/# sudo iotedge list
NAME             STATUS           DESCRIPTION      Config
edgeAgent        running          Up 27 seconds    mcr.microsoft.com/azureiotedge-agent:1.5`

surajgampa95, 2024-09-10:
Just to add to my previous comment, the main issue is as follows - 
`configuration has correct URIs for daemon mgmt endpoint - Error
    Unable to find image 'mcr.microsoft.com/azureiotedge-diagnostics:1.5.5' locally
    docker: Error response from daemon: manifest for mcr.microsoft.com/azureiotedge-diagnostics:1.5.5 not found: manifest unknown: manifest tagged by "1.5.5" is no`
    
There is no version 1.5.5 for azureiotedge-diagnostics. I tried changing the azureiotedge-agent version to 1.5.9. However, this change did not resolve this issue. Can you kindly help me resolve this issue?

surajgampa95, 2024-09-24:
> Just to add to my previous comment, the main issue is as follows - `configuration has correct URIs for daemon mgmt endpoint - Error Unable to find image 'mcr.microsoft.com/azureiotedge-diagnostics:1.5.5' locally docker: Error response from daemon: manifest for mcr.microsoft.com/azureiotedge-diagnostics:1.5.5 not found: manifest unknown: manifest tagged by "1.5.5" is not found.`
> 
> There is no version 1.5.5 for azureiotedge-diagnostics. I tried changing the azureiotedge-agent version to 1.5.9. However, this change did not resolve this issue. Can you kindly help me resolve this issue?

This issue was resolved in #178

closed 2024-09-24 by surajgampa95