Azure/meta-iotedge #178: Error: manifest for mcr.microsoft.com/azureiotedge-diagnostics:1.5.5 not found - Scarthgap release

surajgampa95, 2024-09-11:
**Describe the bug**
I am using meta-iotedge Scarthgap release 1.5 in my Yocto project. I am able to run IoT Edge Agent (reflects in Azure portal IoT hub) but **not IoT Edge Hub**. 
**To Reproduce**
Steps to reproduce the behavior:
1. After system boot up, set the connection string command 
`sudo iotedge config mp --connection-string 'YOUR_DEVICE_CONNECTION_STRING'`
2. Apply changes using command
`sudo iotedge config apply`
3. Restart edged service using command
`sudo systemctl restart aziot-edged`
4. Perform checks using command 
`sudo iotedge check`

**Expected behavior**
Both iotedge agent and iotedge hub working 
**Logs**
Configuration checks (aziot-identity-service)
---------------------------------------------
√ keyd configuration is well-formed - OK
√ certd configuration is well-formed - OK
√ tpmd configuration is well-formed - OK
√ identityd configuration is well-formed - OK
√ daemon configurations up-to-date with config.toml - OK
√ identityd config toml file specifies a valid hostname - OK
‼ aziot-identity-service package is up-to-date - Warning
    Installed aziot-identity-service package has version 1.5.0 but 1.5.3 is the latest stable version available.
    Please see https://aka.ms/aziot-update-runtime for update instructions.
√ host time is close to reference time - OK
√ preloaded certificates are valid - OK
√ keyd is running - OK
√ certd is running - OK
√ identityd is running - OK
√ read all preloaded certificates from the Certificates Service - OK
√ read all preloaded key pairs from the Keys Service - OK
√ check all EST server URLs utilize HTTPS - OK
√ ensure all preloaded certificates match preloaded private keys with the same ID - OK

Connectivity checks (aziot-identity-service)
--------------------------------------------
√ host can connect to and perform TLS handshake with iothub AMQP port - OK
√ host can connect to and perform TLS handshake with iothub HTTPS / WebSockets port - OK
√ host can connect to and perform TLS handshake with iothub MQTT port - OK

Configuration checks
--------------------
√ aziot-edged configuration is well-formed - OK
√ configuration up-to-date with config.toml - OK
√ container engine is installed and functional - OK
× configuration has correct URIs for daemon mgmt endpoint - Error
    Unable to find image 'mcr.microsoft.com/azureiotedge-diagnostics:1.5.5' locally
    docker: Error response from daemon: manifest for mcr.microsoft.com/azureiotedge-diagnostics:1.5.5 not found: manifest unknown: manifest tagged by "1.5.5" is not found.
    See 'docker run --help'.
‼ aziot-edge package is up-to-date - Warning
    Installed IoT Edge daemon has version 1.5.5 but 1.5.3 is the latest stable version available.
    Please see https://aka.ms/iotedge-update-runtime for update instructions.
× container time is close to host time - Error
    Could not query local time inside container
‼ DNS server - Warning
    Container engine is not configured with DNS server setting, which may impact connectivity to IoT Hub.
    Please see https://aka.ms/iotedge-prod-checklist-dns for best practices.
    You can ignore this warning if you are setting DNS server per module in the Edge deployment.
‼ production readiness: logs policy - Warning
    Container engine is not configured to rotate module logs which may cause it run out of disk space.
    Please see https://aka.ms/iotedge-prod-checklist-logs for best practices.
    You can ignore this warning if you are setting log policy per module in the Edge deployment.
‼ production readiness: Edge Agent's storage directory is persisted on the host filesystem - Warning
    The edgeAgent module is not configured to persist its /tmp/edgeAgent directory on the host filesystem.
    Data might be lost if the module is deleted or updated.
    Please see https://aka.ms/iotedge-storage-host for best practices.
× production readiness: Edge Hub's storage directory is persisted on the host filesystem - Error
    Could not check current state of edgeHub container
√ Agent image is valid and can be pulled from upstream - OK
√ proxy settings are consistent in aziot-edged, aziot-identityd, moby daemon and config.toml - OK

Connectivity checks
-------------------
× container on the default network can connect to upstream AMQP port - Error
    Container on the default network could not connect to reijetsons-iothub1.azure-devices.net:5671
× container on the default network can connect to upstream HTTPS / WebSockets port - Error
    Container on the default network could not connect to reijetsons-iothub1.azure-devices.net:443
× container on the IoT Edge module network can connect to upstream AMQP port - Error
    Container on the azure-iot-edge network could not connect to reijetsons-iothub1.azure-devices.net:5671
× container on the IoT Edge module network can connect to upstream HTTPS / WebSockets port - Error
    Container on the azure-iot-edge network could not connect to reijetsons-iothub1.azure-devices.net:443
23 check(s) succeeded.
5 check(s) raised warnings. Re-run with --verbose for more details.
7 check(s) raised errors. Re-run with --verbose for more details.
2 check(s) were skipped due to errors from other checks. Re-run with --verbose for more details.

**Build System (please complete the following information):**
 - Ubuntu 22.04

**Target System (please complete the following information):**
 - Arago 
 - TI AM68A

**Additional context**
I tried upgrading the image in [agent.config] config.toml file to mcr.microsoft.com/azureiotedge-agent:1.5.9. Although the edge agent got upgraded to 1.5.9 (verified on target machine and azure portal iot hub), iotedge version still remained at 1.5.5 and I still get the same error on iotedge check.

surajgampa95, 2024-09-16:
@ggjjj Any insight would be helpful. Thanks.

hurtadosanti, 2024-09-20:
[qemux86-64.txt](https://github.com/user-attachments/files/17079809/qemux86-64.txt)
I see the same, I include my kas file

hurtadosanti, 2024-09-23:
Hi @surajgampa95  this should fix the issue: `sudo iotedge check --diagnostics-image-name mcr.microsoft.com/azureiotedge-diagnostics:1.5.10` make sure you add the extra space on Yocto to store the containers.

surajgampa95, 2024-09-23:
> Hi @surajgampa95 this should fix the issue: `sudo iotedge check --diagnostics-image-name mcr.microsoft.com/azureiotedge-diagnostics:1.5.10` make sure you add the extra space on Yocto to store the containers.

This worked for me. Thanks.

closed 2024-09-23 by surajgampa95