Azure/meta-iotedge #182: Systemd services fail to start due to missing libaziot_keys.so in runtime path after LD_LIBRARY_PATH removal

cramirez-actia, 2025-04-29:
**Describe the bug**
When starting azure systemd services (aziot-certd, aziot-identityd, aziot-keyd and aziot-tpmd), the dynamic linker is unable to load the libaziot_keys.so dynamic lib.

On my system the shared library are placed under /usr/lib/rust/libaziot_keys.so.
But when we look at the aziod recipe, we can see we build our application by telling to the compiler the shared libraries are located into /usr/lib/<RUST_HOST_SYS>/lib.
> RUSTFLAGS += "-Clink-arg=-Wl,-rpath,${libdir}/rustlib/${RUST_HOST_SYS}/lib"

I saw this RUSTFLAGS addition has been done at the same time LD_LIBRARY_PATH environment variables have been removed from the aziot-certd, aziot-identityd, aziot-keyd and aziot-tpmd systemd services. This is coming from the last commit on the kirkstone branch (d144445c441428fbc5f7e9910be36931bab3de32)

I tested to re-add LD_LIBRARY_PATH  variables to services and they started successfully, so the issue seems to be linked with that.

Do you have any idea of why I'm observing this issue, and do you think there is something wrong with the shared library installation ?

**To Reproduce**
Steps to reproduce the behavior:
1. Build azureiot using the kirkstone branch of meta-iotedge
2. Deploy the image on the target
3. Observe systemd logs

**Expected behavior**
The systemd service should start without any shared library errors

**Logs**
> Apr 29 09:48:35 <name> systemd[1]: Stopped Azure IoT Certificates Service.
> Apr 29 09:48:35 <name> systemd[1]: aziot-certd.socket: Deactivated successfully.
> Apr 29 09:48:35 <name> systemd[1]: Closed Azure IoT Certificates Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Stopping Azure IoT Certificates Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Starting Azure IoT Certificates Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Stopped Azure IoT Identity Service.
> Apr 29 09:48:35 <name> systemd[1]: aziot-identityd.socket: Deactivated successfully.
> Apr 29 09:48:35 <name> systemd[1]: Closed Azure IoT Identity Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Stopping Azure IoT Identity Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Starting Azure IoT Identity Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Stopped Azure IoT Keys Service.
> Apr 29 09:48:35 <name> systemd[1]: aziot-keyd.socket: Deactivated successfully.
> Apr 29 09:48:35 <name> systemd[1]: Closed Azure IoT Keys Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Stopping Azure IoT Keys Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Starting Azure IoT Keys Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Stopped Azure IoT TPM Service.
> Apr 29 09:48:35 <name> systemd[1]: aziot-tpmd.socket: Deactivated successfully.
> Apr 29 09:48:35 <name> systemd[1]: Closed Azure IoT TPM Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Stopping Azure IoT TPM Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Starting Azure IoT TPM Service API socket...
> Apr 29 09:48:35 <name> systemd[1]: Listening on Azure IoT Certificates Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Started Azure IoT Certificates Service.
> Apr 29 09:48:35 <name> systemd[1]: Listening on Azure IoT Identity Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Started Azure IoT Identity Service.
> Apr 29 09:48:35 <name> systemd[1]: Listening on Azure IoT Keys Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Started Azure IoT Keys Service.
> Apr 29 09:48:35 <name> systemd[1]: Listening on Azure IoT TPM Service API socket.
> Apr 29 09:48:35 <name> systemd[1]: Started Azure IoT TPM Service.
> Apr 29 09:48:35 <name> aziot-certd[5141]: /usr/bin/aziot-certd: error while loading shared libraries: libaziot_keys.so: cannot open shared object file: No such file or directory
> Apr 29 09:48:35 <name> aziot-identityd[5142]: /usr/bin/aziot-identityd: error while loading shared libraries: libaziot_keys.so: cannot open shared object file: No such file or directory
> Apr 29 09:48:35 <name> systemd[1]: aziot-certd.service: Main process exited, code=exited, status=127/n/a
> Apr 29 09:48:35 <name> systemd[1]: aziot-certd.service: Failed with result 'exit-code'.
> Apr 29 09:48:35 <name> systemd[1]: aziot-identityd.service: Main process exited, code=exited, status=127/n/a
> Apr 29 09:48:35 <name> systemd[1]: aziot-identityd.service: Failed with result 'exit-code'.
> Apr 29 09:48:35 <name> aziot-keyd[5143]: /usr/bin/aziot-keyd: error while loading shared libraries: libaziot_keys.so: cannot open shared object file: No such file or directory
> Apr 29 09:48:35 <name> systemd[1]: aziot-keyd.service: Main process exited, code=exited, status=127/n/a
> Apr 29 09:48:35 <name> systemd[1]: aziot-keyd.service: Failed with result 'exit-code'.
> Apr 29 09:48:35 <name> aziot-tpmd[5144]: /usr/bin/aziot-tpmd: error while loading shared libraries: libaziot_keys.so: cannot open shared object file: No such file or directory
> Apr 29 09:48:35 <name> systemd[1]: aziot-tpmd.service: Main process exited, code=exited, status=127/n/a
> Apr 29 09:48:35 <name> systemd[1]: aziot-tpmd.service: Failed with result 'exit-code'.

**Build System (please complete the following information):**
 - OS: Debian 12

**Target System (please complete the following information):**
 - Yocto kirkstone release based system

cramirez-actia, 2025-04-30:
I have tried to edit the RUSTFLAGS to change the path of dynamic libs :

> RUSTFLAGS += "-Clink-arg=-Wl,-rpath,${libdir}/rust"

After a successful build the services can now locate and load the shared libs.
We still can see errors on the below logs but this is normal as i didn't created config.toml file.

> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Starting Azure IoT Certificates Service API socket...
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Starting Azure IoT Edge daemon management socket...
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Starting Azure IoT Edge daemon workload socket...
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Starting Azure IoT Identity Service API socket...
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Starting Azure IoT Keys Service API socket...
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Starting Azure IoT TPM Service API socket...
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Listening on Azure IoT Edge daemon management socket.
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Listening on Azure IoT Certificates Service API socket.
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Listening on Azure IoT Identity Service API socket.
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Listening on Azure IoT Edge daemon workload socket.
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Listening on Azure IoT Keys Service API socket.
> Apr 23 07:34:42 <name>-c2-mmc systemd[1]: Listening on Azure IoT TPM Service API socket.
> Apr 23 07:34:43 <name>-c2-mmc systemd[1]: Started Azure IoT Certificates Service.
> Apr 23 07:34:43 <name>-c2-mmc systemd[1]: Started Azure IoT Identity Service.
> Apr 23 07:34:43 <name>-c2-mmc systemd[1]: Started Azure IoT Keys Service.
> Apr 23 07:34:43 <name>-c2-mmc systemd[1]: Started Azure IoT TPM Service.
> Apr 23 07:34:43 <name>-c2-mmc systemd[1]: Starting aziot setup service...
> Apr 23 07:34:43 <name>-c2-mmc systemd[1]: Finished aziot setup service.
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: WARNING:tcti:../tpm2-tss-3.2.3/src/tss2-tcti/tcti-device.c:443:Tss2_Tcti_Device_Init() Failed to open default TCTI device file /dev/tpmrm0: No such file or directory
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: WARNING:tcti:../tpm2-tss-3.2.3/src/tss2-tcti/tcti-device.c:443:Tss2_Tcti_Device_Init() Failed to open default TCTI device file /dev/tpm0: No such file or directory
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: ERROR:tcti:../tpm2-tss-3.2.3/src/tss2-tcti/tcti-device.c:448:Tss2_Tcti_Device_Init() Could not open any default TCTI device file
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: ERROR:tcti:../tpm2-tss-3.2.3/src/tss2-tcti/tctildr-dl.c:154:tcti_from_file() Could not initialize TCTI file: device
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: ERROR:tcti:../tpm2-tss-3.2.3/src/tss2-tcti/tctildr.c:428:Tss2_TctiLdr_Initialize_Ex() Failed to instantiate TCTI
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: 2025-04-23T07:34:44Z [ERR!] - service encountered an error
> Apr 23 07:34:44 <name>-c2-mmc aziot-identityd[364]: 2025-04-23T07:34:44Z [ERR!] - could not read config
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]: 2025-04-23T07:34:44Z [ERR!] - could not read config
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]: 2025-04-23T07:34:44Z [ERR!] - caused by: could not read config
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]: 2025-04-23T07:34:44Z [ERR!] - caused by: missing field `homedir_path`
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]: 2025-04-23T07:34:44Z [ERR!] -    0: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    1: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    2: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    3: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    4: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    5: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    6: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    7: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    8: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:    9: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:   10: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:   11: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-certd[363]:   12: __libc_start_main
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: 2025-04-23T07:34:44Z [ERR!] - caused by: internal error
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: 2025-04-23T07:34:44Z [ERR!] - caused by: could not initialize TPM
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: 2025-04-23T07:34:44Z [ERR!] - caused by: tcti:IO failure
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]: 2025-04-23T07:34:44Z [ERR!] -    0: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]:    1: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]:    2: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]:    3: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]:    4: <unknown>
> Apr 23 07:34:44 <name>-c2-mmc aziot-tpmd[366]:    5: <unknown>

jlian, 2026-02-06:
Fixed in PR #190 (merged to main). The rpath fix (`RUSTFLAGS += "-Clink-arg=-Wl,-rpath,${libdir}/rust"`) is now included in `aziotd.inc`. Available in release [1.5.35](https://github.com/Azure/meta-iotedge/releases/tag/1.5.35).